terjeio/ioSender #3: initial questions and feedback

jschoch, 2020-01-23:
Overall it looks good!  I ran it with an esp32 running bart's port but not connected to anything else.

I see a tool selection, how do tool offsets work?

I also see "program limits", seems to be the boundaries of the current g-code file loaded.

wondering if you support extended firmware settings e.g bart's stuff.  those don't seem to be editable assuming that is something on the backlog.

window resizing doesn't seem to work, assuming this is due to alpha stage.  nothing moves on maximize so nothing is "broken" it just stays the fixed sizing.

any plans for a "terminal view".  It is sometimes handy to see the responses in list out.  Also it is nice to see the pin states when troubleshooting.  If your stated update quickly it may not be an issue. UGS in particular seems to have a lot of lag updating the states. I really like having terminal view i most modes so it would be great to have a sizable full width row at the bottom you could use to toggle on/off in each tab.  At a bear minimum it would be nice to see the MDI input on both the grbl and offset tabs.

regarding MDI, it would be great to be able to add "macros" and or console type wizards.  For instance i'd like to make some canned routines for facing, probing etc.  I was looking at mecode which seems like an interesting attempt to make gcode a bit higher level.

I don't see where the gcode viewer is supposed to render.

The camera needs to be plugged in before you start the app, restart with camera plugged in and it worked good with my C270.  My pcb mill has a endoscope camera and bCNC has a camera offset function.  I see a move offset button but it isnt' clear what it does.  I'm not sure if it uses one of the work coordinate systems, but it allows you to align your gcode for double sided pcb milling which is a very nice feature!  Camera zoom seems to be a good feature also when trying to center on .9mm holes.  Docking the camera window would be a big win.  I'm not a big fan of multiple windows.  Centroid's software does this and it is one of the biggest bothers.

plans for jog buttons in the UI?

i ran some gcode that creates a error 33 and the sender seemed to ignore it.  not sure if this is intended.  bCNC halted with the same gcode.

you seem to need to hit stop to load a new gcode file.

i ran other gcode file that had a M8 command.  It halted at that line.  UGS's behavior is that it goes into "HOLD" and if you resume it will skip the offending line.  This is great for things like M8 that you may forget to comment out if you forget to disable cooling in fusion.  I also noticed the flood and mist toggles and wondered why grbl doesnt' support m8...

terjeio, 2020-01-23:
Tool selection: bug, you should not see it unless a tool table is enabled in grbl. Offsets should work automatically, they are bound to tool number.

Extended firmware settings are supported as long as they follow the $<n>= convention. The sender uses a tab-separated text file for UI-parameters, automatically selected between standard grbl or grblHAL versions. The file to be used may also be  specified on the command line by the 
`-configmapping` parameter. Format specs should be added to the wiki...
Definitions for extended settings has to be added to the file in use to show up correctly, `setting_codes_en_US.txt` for non grblHAL firmware. The `[ESPxxx]` setting commands in bart's port... To be honest I do not like them, but I guess a separate tab for them could be made.

Window resizing: not yet supported - should be disabled until then.

Terminal view: yet another control to be added I guess.

Macros: again, I would go for a separate control for that.

GCode viewer: will render in a separate tab for now. Disabled by default, enable in `Settings: App` tab. Should be enable by default, my bad.
 
Camera: offset can be set in Settings: App tab. I'll disable the `Move offset` button if zero. Regarding zoom, do you mean software based? Docking will have to wait until user configurable UI is available? I do not want to resereve UI real estate for controls that are not always used.

Jog buttons: I do not like them, I prefer keyboard jogging. Can make a control for them, I guess you mean a bunch of buttons for different jog distances. IMO that isn't really jogging but rather predefined relative moves, this since I do not see the need issue the jog cancel command. Or do you mean buttons you can click and hold similar in function as keyboard keys? 

Stop on error: when I test here sending stops. Can you upload the offending file?

`Stop` to reload: bug? If a file run to completion the Stop button should be automatically disabled, didn't that happen? If not it would again be helpful if you upload the file.

M8: Bart's port does not support M8 and returns an error? I wonder if it is a generally good idea to continue after any error returned from the controller. Per default grblHAL firmware will reissue the same error code for subsequent lines, it has to be cleared before continuation is possible. I am open for input on this, but from a safety perspective I do not like beeing able to just continue as nothing happened. Perhaps a good solution well be to add a configurable filter to remove unsupported commands from the file?

jschoch, 2020-01-23:
want to break this stuff up into individual issues where it makes sense or just keep it in here?

here is the error 33 code, i believe this is an issue with the arc size and the issue stems from the default grbl post processor.

```

(1001)
G90 G94
G17
G21

(Face1)
S5000 M3
G54
G0 X87.9 Y1.896
Z6
Z2
G1 Z-0.5 F800
G18 G3 X87.3 Z-0.5 I-0.6 K0
G1 X84
X0
G17 G2 X0 Y6.736 I0 J2.42
G1 X84
G3 X84 Y11.575 I0 J2.42
G1 X0
G2 X0 Y16.414 I0 J2.42
G1 X84
G3 X84 Y21.254 I0 J2.42
G1 X0
G2 X0 Y26.093 I0 J2.42
G1 X84
G18 G2 X84.6 Z0.1 I0 K0.6
G17
M5
M30
```

terjeio, 2020-01-23:
Perhaps best to have separate issues for bugs at least, possibly also for discussion about functionality.

As for the G33 error execution halts when using grblHAL firmware because that refuses to continue after an error. However, when testing with vanilla grbl on a Mega it continues - not good. From the code I see I have commented out a line that flushes the buffers in the controller and stops sending, not good either I assume.

Using the Grbl Panel sender it stops after the line after the error is executed, and it is possible to restart (perhaps it sends a feed hold command? I will check). Is that how your other senders behaves too? I am not so sure I am comfortable about that.

So what should be the correct behaviour? In my opinion no commands after an error is encountered  should be executed, but I believe that is impossible to achieve unless a reset command is sent. If reset is sent then it is also impossible (AFAIK) to continue as position may be lost and the controller might no longer be in the correct state. IIRC i think I've seen it mentioned somewhere that this is something Sonny is not entirely happy about too.

Clearly I have to fix this, any ideas for the best way forward?

jschoch, 2020-01-23:
you may not have to fix the error 33 right away since I only ever saw this error last week and it seems uncommon.  The other senders seem to behave differently.

for the M8 or M6 I do like the UGS behavior.  A prompt to skip the offending line would let the operator decide assuming the parser catches it and position is not lost.

terjeio, 2020-01-23:
You will have to fix the G33 error yourself, as the controller fails it ;-) I see that the gcode changes between planes, seems a bit strange to me but what do I know. The 3D view looks a bit odd too...

I believe I have found a solution for stopping at or near the offendig error, this by only sending line by line waiting for reply and not stuffing the receive buffer. It looks like that is what the Grbl Panel does. Since I know if grblHAL is the controller I can use this approach for other controllers only.

As for catching unsupported commands would you like to do that on load or on run? Perhaps on load is the best?

terjeio, 2020-01-24:
Ï have added configurable command stripping for M6, M7 and M8:

[image]

Check is done on load, and for `Prompt` mode you will be asked for every instance of the command in a file. Is this how you want it? Or is it better to ask for the first instance and keep the answer? Or perhaps add a `Prompt once` alternative?

jschoch, 2020-01-24:
looks great and seems simple!

closed 2020-04-17 by terjeio