terjeio/ioSender #5: command history for mdi

jschoch, 2020-01-24:
up/down arrow to navigate command history in MDI window would be very nice.

terjeio, 2020-01-24:
Have you used Mach3? In Mach3 a history list pops up when the MDI is given focus. I kind of like that since it is hidden when not needed. Perhaps try to emulate that?

jschoch, 2020-01-24:
i have  not used mach3.

I don't think it needs to be a window with a list, but it would really be great to run small blocks of gcode and have them as single items in the history that can run whole. so something like:

```
--- 1
g00 X0, Y0
--- 2
g00 Z1
--- 3
g01 F100 
g91 X10
g91 y-3
g91 x-10
--- *current*
m5
```

so item "--- 3" would run all 4 lines.

terjeio, 2020-03-16:
Simple command history added to latest alpha release along with a number of other improvements.