terjeio/ioSender #6: newb friendly factor

jschoch, 2020-01-24:
wondering how newbie friendly you want to be.  I doubt most newbs can get work positions set without an MPG/pendant or buttons in the UI, they likely don't know the gcode move commands to put in the MDI.

Also assuming many folks will want to use standard keyboard keymapping to jog,

terjeio, 2020-01-24:
Good question, hard to answer. UI design is tricky - I am of the opinion that at least three versions are needed in a perfect world. This far I have had focus on my own needs, publising a binary and getting users is a different ballgame. Kind of risky decision, I know... I hope it will be a positive experience for both me and users.

terjeio, 2020-01-24:
A suggestion for a button based jog panel added to latest release, accessible from the right upper corner of the main window. Layout is similar Grbl Panel.
Is this an ok way to do it? I put it together rather quickly as a test, may need some refinement. Perhaps keyboard jogging should be disabled for non-grblHAL controllers?

jschoch, 2020-01-24:
It wasn't' clear how the keys were mapped, did I miss that in the settings?  If there is a way to adjust the keymapping then I think you can keep it for non-grblHAL.

terjeio, 2020-01-24:
Jogging is mapped to the cursor and Page up/Page Down keys, currently with no option for change. `<Ctrl>` (step) and `<Shift>` (fast) to modify mode.

A few other keys has been assigned as well, I have to add these to the [wiki page](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Usage-tips):

`<Space bar>` for feed hold.
`<Alt>`+`<R>` for cycle start.
`<Alt>`+`<S>` for stop.