terjeio/ioSender #8: building in VS 2019

jschoch, 2020-01-31:
Well, I installed VS and my computer has been BSOD'ing for a week.  I think it has finally stabalized with some ram settings.

I tried to build the project and my newbness with both VS and c# are glaring.  I will try to read some tutorials on c#.  Any tips you might the build would be helpful.

here are the errors:

[image]

jschoch, 2020-01-31:
i forgot to do a clean pull between VS19 and my attempts to get VS Code working.

also it seems that you have to nuget install the packages for each dependency not just in the main proj thingy.

Trying to work these out now.

[image]

jschoch, 2020-01-31:
deleting the "CNC Core/obj" dir seems to have removed the issues for building core.

jschoch, 2020-01-31:
i had to delete the obj dirs for all the dependencies and it built!!! 

I'll close this out now.

closed 2020-01-31 by jschoch

terjeio, 2020-01-31:
See [wiki](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Building) for dependencies.

Just saw your latest messsage, great you got it working.

I uploaded a new version yesterday, it does not look like you have downloaded that.