terjeio/ioSender #11: alpha 7

phil-barrett, 2020-02-21:
Tried Alpha 7.  Still hangs on first run against newly rebooted teensy4.

Would be nice to be able to change port and toggle from the app.  deleting app.config gets old fast.

terjeio, 2020-02-21:
Which Windows version?  I am running Win7. Driver:

[image]

For [selecting port on startup](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Setup-and-configuration): create a shortcut to the exe-file and add -selectport to the command line (target).

phil-barrett, 2020-02-21:
Win 10. Teensy install says I don't need an update.
[image]

terjeio, 2020-02-21:
Then I guess it must be the driver causing this problem. I will investigate.

terjeio, 2020-02-21:
Toggling DTR twice seems to have solved the issue with the Windows driver on Win10.

I have just started a 10-12 hours job on my Win10 machine that I'll run overnight to see if I can reproduce your communication glitch.

Too bad you fried your Teensy - you did that while trying too hook up the UART <> serial breakout?

phil-barrett, 2020-02-22:
OK, looking forward to trying it out.  

Yup, not sure what I did.  The USB serial device was working fine looped back (tx to rx) but when I hooked it up the teensy, my PC barfed on it.  Oh well, shit happens.

closed 2020-05-20 by terjeio