terjeio/ioSender #13: Thoughs on Grbl-GCode-Sender

phil-barrett, 2020-02-27:
After using GGS for a few days, I am quite impressed with the quality of it. Very well done. Program limits are a useful feature. I could have used this a couple of times in the past. Even with the arc issue, it's a great feature. 

I usually use GRBLPanel but given that the owner has stopped supporting it, I've been looking for a replacement for GRBLPanel.  I think this is it.

Some thoughts and questions:
- Com port, baud rate should be settable from the Settings: App panel.  Would rather not have to restart with a command line option. Especially when using multiple boards.
- Visualization doesn't seem to work at all for me. Am I missing something?
- It would be nice to have jog and macro panels that don't obscure the main panel (i.e. allow them to be made permanent)
- Macro definition is not intuitive. You have to create the button and then edit it again to add the actual gcode. It would be more clear if there were separate new and edit buttons. New would essentially be the edit screen with the ability to fill in a name. As it stands, it seems like you are changing an existing macro when you create a new one. Also, there doesn't seem to be a way to delete a macro.
- How would it support an ethernet interface?
- On the grbl settings panel, I think there should be 2 modes - basic and advanced.  Basic would show only the standard GRBL functions. Advanced, shows all.  If basic was the default, grbl tutorials would be consistent with the initial view.
- It would be great if the MDI input and status showed up in the list of commands accepted by the grbl board.
- It needs a snazzier name than grbl-gcode-sender. It is an outstanding program and should have a outstanding name.

OK, one that is a huge ask and probably not happening anytime soon but I would love to see realtime GCode execution feedback.  So 2 statuses: ok (received) and completed. I know it's a complex area and needs GRBL changes so it's hard(er).

Anyway, hope that's not too many things at once. GGS is already a fantastic program so please excuse my enthusiasm for making it even better.

terjeio, 2020-02-27:
Thanks for positive feedback. The sender is still in alpha so early days...

* Settings: My idea was to have separate config files and use shortcuts if the sender is used for more than one machine. Path to the config file to use is a [command line option](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Setup-and-configuration).

* Visualization: What is not working? If the `3D View` tab shows up it should work. If not it needs to be enabled in the App settings. Note that the rendering is currently a bit rough.

* Jog and macros: I am experimenting with this, be aware that [keyboard jogging](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Usage-tips) is available - so I do not understand why this panel/control is needed at all...

* Macro definitions: was put together in a hurry, to delete a macro delete the gcode and commit. I use a button with a pop-up menu for commands in my custom laser sender (add, save, delete), perhaps I should try a similar same approach for this control. 

* Ethernet interface: supports simple telnet and websockets communication. Currently connection data needs to be entered on the [command line](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Setup-and-configuration) of a shortcut or by editing the config file.

* Settings: One idea is to categorize the settings and use a grouped view. Mode can be implemented, the mode should then be put in the settings definition file (setting_codes_en_US.txt) as the UI is completely driven by that (in anticipation that settings available may evolve differently for ports - hardcoded for me is a no-no). Again, IMO a new project should be set up for handling grbl specs, this to avoid ports becoming incompatible. Preferably moderated by a third party not involved in porting or sender development... Kind of grbl standards body.

* MDI and status input: on my todo list. I have already made a response view control...

* Snazzier name: any ideas?

A lot can be done to improve the sender - and I have many ideas/plans. The UI could be made user configurable, drop in and arrange the controls you want from a palette. The 3D renderer needs attention, is currently too slow and should be viewable from the main tab. Camera view is too simple. Etc, etc...

phil-barrett, 2020-02-27:
Some of this is about different expectations for applications. I'm in the visual/in the app control school of thought. Want to see buttons/controls/etc. Command lines and key presses require remembering or looking up syntax. So, of course you will hear stuff like that from me.  

I saw how to do ethernet in the wiki after I sent the comments. sorry.

I didn't realize that Visualization needs an app restart to activate.  That's a bit confusing. That will be a source of confusion by users. At the least, a note to restart the app would help there. Of course, immediately taking effect would be better.

I like the idea of a configurable UI. Especially if you have a way to save multiple and have them quickly available.

Name? Hmmm been thinking about it. GRBLizer?   GSender? 

I agree visualization needs to be sped up.  I loaded one of my 10 hour tests (the big arc one) and it takes about 5 minutes to render but much shorter programs seem to take a lot of time too. I think arcs are part of the problem. Part of the issue is that you can't use the app while it's rendering. Maybe tie the rendering to opening the 3D tab rather than opening the GCode file as a stop-gap. I think some visualization programs allow viewing while running the actual rendering code in a separate thread. I will play with it a bit more, seeing some issues.

terjeio, 2020-02-27:
> Some of this is about different expectations for applications. I'm in the visual/in the app control school of thought. Want to see buttons/controls/etc. Command lines and key presses require remembering or looking up syntax. So, of course you will hear stuff like that from me.

It is ok to have opionions, but it is a chicken & egg situation that I have to resolve. The sender need to pull information from grbl to configure itself. So the communications setting can not (IMO) be moved to the App settings tab unless it starts up in the App settings tab every time. I for one would not like it that way. Similarily settings needs to be pulled from the ini-file as well, again that could be selected from the settings tab but practical? So IMO using shortcuts with command line parameters still has merit.

> I didn't realize that Visualization needs an app restart to activate. 

I am well aware of this, either a prominent message is needed or auto reconfig should be done. On my todo list already. Currently most of these settings require a restart.

> I like the idea of a configurable UI. Especially if you have a way to save multiple and have them quickly available.

Of course layouts should be possible to save, if not I do not see the point of having such an option. And multiple layouts require selection, again I would prefer a shortcut with a command line option... Alternatives would be to require multiple instances of the sender or selecting a layout on startup.

---

The 3D renderer requires quite a bit of work before it is usable, and I would prefer to have a smaller version available in the main window. Currently the program list and the related buttons (Start Hold etc) resides in the same control. I want to split them in two and I am thinking about making a tab out of the program list and add a response log and a "live" 3D viewer as tab options to that. Partly because I do not want a large cluttered main window - the sender should be usable on a laptop.

I appreciate input, however implementation will take some time. For now I have focus on making the core functionality stable. Rewriting the main state machine to use function pointers now, in a similar vein to what I did with the original "spaghetti" state machine in grbl...

terjeio, 2020-03-11:
My idea for adding a tab where currently the GCode list (Program) is :

[image]

[image]

Comments?

phil-barrett, 2020-03-12:
I think that's a straightforward approach.  One thing to consider its there would be 3 sets of tabs on the app - on the top, on the right side and bottom of the "output" panel.  Each group is a little different. I think a UX person would want them to be more consistent  (shape, placement, ...) but I'm not sure it will give us much benefit.  Moving the output tabs to the top might be a good idea though it may be too busy with the tabs on the top close by.

einencool, 2020-03-22:
Hello,
today I've tested a little bit with your software, but these were only "Desktop Tests" with the Teensy Board liing on the table :-)

I like your software very much, but I also have some questions for updates. I have no idea from coding, so I can't mention how much work it could be.

I like your implementation of the jogging commands. And the Software (alpha 13) looks fast and reduced to the important things...

I'm controlling my CNC with a bluetooth keypad, so it would be great to optimise the software for more keypad control :-)

These are my "points for improvments...":
- different Step / Jog size/speed for XY and the Z-Axis (to drive the Z-Axis slower than the other axis)
- Keymapping for Macros / Start / Pause / Stop / Check
- Maybe a way to maximise the software on a screen, because it looks so small on a 24 " Screen
- I come from bCNC and it's nice to see in the "3D-View" how much from the program was done ( the part that was ready was shown in green) But I don't know if it is much work and if it will "blow up the code". It's just a "nice to have"
- Is there a way to show the "remaining" time? This was also shown in bCNC. The software was good, but it hang up to often and costs me to much of my workpieces...

So in the next days I'll try your software with my old Arduino Mega controlled CNC (if it is possible)

terjeio, 2020-03-23:
> different Step / Jog size/speed for XY and the Z-Axis (to drive the Z-Axis slower than the other axis)

Have you tried [keyboard jogging](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Usage-tips)?

> Keymapping for Macros / Start / Pause / Stop / Check

Some are [assigned already](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Usage-tips), but I guess you want user defined binding. Will put on todo list.

> Maybe a way to maximise the software on a screen, because it looks so small on a 24 " Screen

Om todo list (perhaps I should publish it?), initial design goal is that is should be usable on a laptop with limited resolution.

> ...see in the "3D-View" how much from the program was done...

Will have to investigate this, not sure how complicated it will be. Perhaps moving the vectors to a different path collection will work, this since it is the path collection that is assigned properties such as color.

> Is there a way to show the "remaining" time? 

Not yet. Doable since the program parses the gcode on load.

---

My main proirity for now is to make the program robust and add what I consider as core functionality. Currently I am working on probing, the gcode parser needs attention as does the 3D rendering (not all G-codes are supported). Additional lathe wizards is also high on my agenda, I guess not what most users would want - but I want those for my lathe rebuild.

einencool, 2020-03-23:
The keyboard jogging is really good, the only thing is, in my thoughts, on a machine with a size around 1000x 600 x 200 mm it would be nice to drive the XY axis faster than the Z Axis, but that was only a thought :-)

For the key mapping you can define them to your preferences. Today I have the "Auto Hot key " program to remap them on my Bluetooth keypad.

I have it also tested on my 15" laptop, but there is also the high resolution the problem that I have to sit directly in front of it to see what I have to click.. :-)

Like I said, I only make a test on the bench, it seems yesterday in the evening that I had a "NC" file with G17 and G19 commands. The sender stopped at these, he semms not to know what that is. Could that be, or is there an issue in my file?

Thank you for your continuous improvement.
The software has a lot of potential. 👍

terjeio, 2020-03-23:
@einencool  : if possible please attach your problem file to a comment so I can use it for debugging.

einencool, 2020-03-23:
Hi, 
this was the only Code in this program, it was to clear only one side.
`G90 G94
G17
G21
G54
S12000 M3
X125.6 Y35.2
Z15
Z5
G1 Z1 F333
Z-4.4
G19 G3 X125.6 Y35.8 J0.6 F1000
G1 Y36.4 Z-5
G17 G3 X125 Y37 I-0.6
G1 X0 Y37 F400
G3 X-0.6 Y36.4 J-0.6 F1000
G1 X-0.6 Y35.8
G19 G2 Y35.2 Z-4.4 K0.6
G0 Y35.2 Z15

Z20
M5
M30
`

On this Laptop I don't have your Software installed, so I can't say if it was wrong.
I think it will work, because on my other Software, the Code in "Check Mode" also gives an error, but it works.

Then I tried your Software on my actual CNC Router with an Arduino Mega.
Well I've had some Problems with it, because while jogging sometimes the software didn't response or better to say it doesn't move any more. I could click in every single window oder every button, but I have to press the Reset button so that I can move forward.

But then it happens, that i shortly press the right key, and the spindle moves a long way until I pressed the Reset key. It was my Luck that I was far away from my workpiece, otherwise it would have killed my bit...

But after that I switched back wo my "normal software" and worked with it the rest of the day.

phil-barrett, 2020-03-23:
grblHAL doesn't like G2 and G3.  G17 and G19 are OK.

phil-barrett, 2020-03-23:
The first G2 is the offending one.  I changed it from 

> G2 X125.6 Y35.8 J0.6 F1000

to 

> G2 X125.6 Y35.8 R2 F1000

And grbl accepted and ran the program. Note that R2 was just a guess. It also accepted all the other G2s and G3s. Something in the J0.6 is wrong but I don't know what. Here's whole program that get's accepted. 

```G90 G94
G17
G21
G54
S12000 M3
X125.6 Y35.2
Z15
Z5
G1 Z1 F333
Z-4.4
G19 
(G2 X125.6 Y35.8 J0.6 F1000)
G2 X125.6 Y35.8 R0.6 F1000
G1 Y36.4 Z-5
G17 
G2 X125 Y37 I-0.6
G1 X0 Y37 F400
G3 X-0.6 Y36.4 J-0.6 F1000
G1 X-0.6 Y35.8
G19 G2 Y35.2 Z-4.4 K0.6
G0 Y35.2 Z15

Z20
M5
M30

terjeio, 2020-03-24:
@einencool 

There is a problem with your program, line 11 reports error 33 (motion command target is invalid). grblHAL (depending on compatibility level) is a bit more strict than other versions, it does not allow the program to continue after an error as this may cause accidents. This may be the reason it stops. Also, I guess my sender is more strict than others in handling errors in programs.

Keyboard jogging with other firmware than grblHAL:
Jog cancel is buggy and does not always work. Only workarounds has been suggested for that, not a fix for the underlying bug(s). I have made a version for the [Arduino Mega](https://github.com/terjeio/grbl-Mega) with what I believe is a fix, you may try that. It was recently [brought up here](https://github.com/gnea/grbl/issues/837) by @shaise who also has suggested a fix.
I have warned about the issue in the [wiki](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Known-limitations), but it is perhaps better to make the keyboard jogging a configurable option when using other firmware than grblHAL.

Pressing the Reset button is sometimes the only way to continue, again for safety reasons. I need to put more information in the wiki...

Anyway, many thanks for your feedback - very useful for bringing both grblHAL and the sender forward. And thanks to @phil-barrett too for helping out.

einencool, 2020-04-13:
Hi @terjeio 
like I mentioned in the Teensy Topic, I would like to give you some points for your Software. Maybe there are some Points you can solve :-)

I'm using my old CNC with an old Playstation 3 Controller and map the buttons with "Joytokey" to some keyboard keys. So I don't have to look every time onto the monitor when I'm starting my job.

1: Realtime Jogging is now optional, I think that is a good point. 
But maybe you can make Shortcuts for the "normal" Jogging feature like your keys left / right and so on for "normal" Jogging with the step width choosen in your "Jog" Tab.
So if you press left, the gantry will move e.g. 0,1 / 1 / 10mm left.
Maybe you can map the "Numpad" keys 1/2/3 for 0,1 / 1,0 / 10mm step width.
Or you can adjust it like before with Shift and Ctrl or Alt Key for different Step width.
That could also be mapped to the Gamepad...

2: Shortcuts for macros, so I can start them with a button.

3: Look and feel:
Please maximize the Screen :-) On a bigger Monitor it doesn't look so good
Please place the step side to the main page or the complete Jog-tab, so that you can see, which movement you are doing.

4: Probing Tab
It's great to see, I didn't start it now, but it would be nice to have two Options for Probing speeds for fast and slow probing for more accuracy...
I've done it till now with a macro, but your Center finder looks also nice. 
Is it possible to calculate these Positions to an offset when probing with an external pin?

Maybe you can do some of these Things because I think your Software can be very good, it's very lightweight and fast, I like that.

Thank you so much :-)

phil-barrett, 2020-04-13:
If you add maximize, then please also add minimize.

terjeio, 2020-04-13:
@phil-barrett
minimize added, easy to miss out on obvious stuff when focused on internals...

@einencool 
Thanks for input. Generally I think I am slowly getting a understanding of how the WPF/MVVM coding pattern should be used and from I have learned, and used, I believe the codebase is now approaching a useful and relatively stable foundation to build upon. I have chosen to split the sender in many different projects for a number of reasons, one is to be able use the different pieces to build targeted apps (I have one dedicated for my CO2 laser), another is to ensure that I adhere to the coding pattern rules. This means that I have put some restrictions on how I want to implement stuff. My main focus this far has been on internals, not the UI, as I want the basic stuff to be as clean and robust as possible.

1. Jogging
There should be no reason to not enable keyboard jogging when the firmware is gblHAL, regardless of the `COMPATIBILIY_LEVEL` setting. Note that it is enabled automatically for some levels. Step jogging is active when the cursor keys are modified by the <CTRL>-key, perhaps one idea would be to bind the step size to the one selected in the jog control? 

Adding further shortcut keys such as executing macros or selecting stepsize is something I have already started looking into, I want that to be a part of a single centralized class shared by a number of subscribers to keypress events. Currently the code is a little bit too tightly coupled to other code to achieve that.

I do not own a gamepad so I have no idea of how flexible the configuration for these are the via Joytokey app. If you can bind keys on the keypad to the same keycodes that the cursor keys and the modifiers provide keyboard jogging via the keypad should work "out of the box".

2. Shortcuts for macros - see above. Should be bound to the function keys only?

3. Look and feel. 
On todo list - I have an idea that the UI should be user definable. The MVVM coding pattern allows that. It kind of is already, but that requires a bit of coding... The main project, GCode Sender, is surprisingly little code - its purpose is mainly to glue all the components together to form the UI.

4. Probing.
The idea is to make the left hand side parameters selectable from user defined profiles (via a drop down), a bit similar to how lathe mode profiles are implemented. You can check that out by setting the lathe mode in _Settings: App_ to _Diameter_ or _Radius_ and restart. Tabs for _Threading_ and _Turning_ will then show up.

I need users to test and provide detailed feedback for the probing functions, I am no expert in this area. And I need to get/make a probe or two, testing this far has been with a tool.

---
I have spent a lot of time on this project lately, and I have a bunch of others I want to focus on now as the spring is coming. A garden, a lathe waiting for its new lead screw+++. Give me some time and hopefully I can make you all happy.

einencool, 2020-04-14:
When the real time jogging works with grblhal without problems, it's no problem, then you can leave it as it is. 
On my old cnc I've set my step width "normally" to 1mm and when I press the key left or right on the gamepad the button will be mapped 20-times per second, and when I change the step width it will be moving slower or faster. Then I use another key press and I can only move one step per time to find my edge or something like this.

I think the one step per per time width should be aligned to the step width choose in the jog box, so you can make precise movements and "know" how far you move...

I can map up to 4 different keys to be pressed with one button simultaneously. E.g. Ctrl-Alt-Shift-"A" for one command, so it's no problem if you give maybe 12 different shortcut keys e.g. Shift-"F1" and so on and then the user can map them to different functions.
You can have a look at "Bcnc" there are all different versions for the "F-keys" and they can also be used for own macros. That would be good enough.

When it would be possible to get the shortcut upgrade and the adjustable "single-step-width" per jog-step-width e.g. 0.1mm and so on, then I would definitely test it with my new machine when it will be ready. But I think it will last for around 2-3 weeks.

Thank you for your great work and help.
Greets Chris

terjeio, 2020-04-14:
New release up for testing, this is quick and dirty implementation of enhanced keyboard step jogging.
It requires keyboard jogging enabled and will _always_ use the step feed rate from _Settings: App_ (this is not going to change).  The current step distance will be displayed in the lower right corner of the Grbl tab and can be changed with numerical keypad buttons `4` and `6` (when num lock is enabled).

The jog pane Distance radio-buttons will be synced both ways - but not sure if this is wise. Same goes for the longest jog distance (10mm/1inch) - it is currently selectable with the shortcuts but I am not sure this is wise either. I would certainly vote against having that selectable.

Step jogging is activated when the `<CTRL>` modifier key is pressed, note that this mode does not support autorepeat (and never will). IMO longer jog distances should be performed without this modifier or with `<SHIFT>` modifier key for faster feedrate.

Note that the initial step distance will be set from the step distance value from _Settings: App_ , this is a feature, not bug. Also note that those who have a keypad attached directly to the controller will not see the _Keypad jogging_ frame in _Settings: App_, this since parameters for jogging is read from the firmware (set via_Settings: Grbl_) in order to keep keyboard jogging consistent with the keypad.

I might be persuaded to allow step keyboard jogging for non-grblHAL firmware when keyboard jogging is not enabled, this since the autorepeat feature is not used and jog cancel never sent.

Test with care! - as I have not done much testing myself...

einencool, 2020-04-14:
Wow, that was fast...

I'll try it in the evening with the new PC that I want to use with the new cnc Maschine.

I think, that I have explained it not so good in my previous post. In my opinion it would be good only to adjust the single step width with the "Step width" from the jogging tab.
The original jogging was ok for me. 
Only on my old cnc with the arduino Mega was a little bit faulty...
The sipped was adjustable in the settings tab, so that was all ok and very smooth

terjeio, 2020-04-14:
@einencool 
> Only on my old cnc with the arduino Mega was a little bit faulty...

My [patched version ](https://github.com/terjeio/grbl-Mega) for Arduino Mega should work ok with keyboard jogging enabled.

einencool, 2020-04-14:
When I would use the cnc for further operation, I would think about it. But this cnc will be removed when the new one is ready, so never touch a running system :-)

terjeio, 2020-04-17:
Build [Alpha-0.17](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v0.17-alpha) is now up. Resizable UI++

einencool, 2020-04-29:
Hello @terjeio ,
Today was the first time, I fired up my new CNC with your latest GRBLHal and your Sender Nr 17. I make the first moves with it.
After some different Probs, I got it to work, but one thing was really strange while jogging the machine.
When I use the „Realtime“ Jogging Commands with the arrow keys, every time when I release the button, the machine makes a quicker move at the end. This is in Slow and fast mode visible. When I set the acceleration to 10 mm/sec2 then it was very good visible.
When I jog with your 10mm or 1mm/button controls, everything works as it should. The machine slows down as it should.
I don‘t know if you can understand what I‘m writing *g*
Do you know what could be the problem?
In the next days I‘ll try a little bit more. I hope tomorrow I‘ll get the spindle working.

terjeio, 2020-04-29:
@einencool : can you add your grblHAL settings to a comment?
The easy way: select _File > About_ in the sender and press the _To Clipboard_ button, then you can paste the settings into a comment.

einencool, 2020-04-29:
Thank you for your fast reply, I‘ll post it tomorrow.
Hopefully you can see what is the Problem :-)

terjeio, 2020-05-01:
Thanks, I have tested a bit (but not with a machine connected) and cannot see any unexpected behaviour.

Be aware that when jog cancel is sent on key up the movement is decelerated to a stop. If the jog speed is high and acceleration setting is low there can thus be a significant delay before movement stops. Try experimenting with the keyboard jogging settings in _Settings: App_ in order to find values that suits you (and your machine).

Button jogging via the jog flyout sends short distance jog commands that is not terminated with jog cancel and will therefore behave differently.

einencool, 2020-05-01:
Well, I tried today a little bit more, but it makes no difference if I put a acc of 10 or 1000 for the axis. Also the distance is not the problem if the max speed is reached, then every time at the end it makes this scary move. On every axis. 
When I look at the screen everything looks fine. It's only visible for me, when the machine is moving.

When I switch to the one key press jogging, then it accelerates and brakes as it should. And the machine is capable to be moved with around 10 m/min and now I'm on a speed with 1.5 m/min

Since I have no other software which supports real time jogging with the Teensy and your GRBLHal, I can't test it otherwise.
In this case I'll make some tests with the UGS Platform software in the next days...
But I don't know if it will work out of the box with the grblhal in compatibility mode 0, but I'll see it then :-)

With best regards Chris

terjeio, 2020-05-02:
I checked again with my scope - then I saw it. So you are right, there is an issue. I have checked with two other processors/drivers, they do not show this issue so perhaps a driver problem.

My initial attemps at solving this has failed - so I suspect this may be a tricky one to track down. I miss having access to a debugger...

einencool, 2020-05-02:
Ah Ok, the good thing is, that it's not only on my machine, but it's not so good to hear that you don't know what the problem could be. Especially when it's only with the Teensy and not with other boards.

shaise, 2020-05-02:
is it a problem also with the stm32f103 driver?

terjeio, 2020-05-02:
@einencool : can you try with commenting out `#define ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING` in grbl/config.h (line 258)? I do not see the issue on my scope when I do that. I guess it is a good lead - however, I have to go shopping now so will have to follow that later.

@shaise : no, I've checked the STM32 and MSP432 drivers - both seems fine.

shaise, 2020-05-02:
@terjeio, Yes, I tried using the same settings @einencool  uses on the STM32 on the real CNC and I can confirm it works with no issues. thanks.

einencool, 2020-05-02:
@terjeio 
Thank you for your Help, this works for me, and now the moves are really smooth, like it should be.
So I finished the mechanical assembly today, and wanted to make my first moves with the machine.

But then I got a Problem with my Z-Axis, but this I'll look for in your GRBLHal Repo, because it seems to be a little Problem with the "Direction" Signal for the Z-Axis. It doesn't power the direction Pin. So I changed the Z and the A Axis Direction Pin in the Config (Pin 7 & 9) but that also doesn't move. So I finally changed the Z and Y Axis, and that also doesn't fire up the direction Pin for Z, but for Y it does, so I think the wiring is not the Problem, it seems to be in the Config. So I have to look for it, but maybe you know where the Problem could be. In your Software the DRO for Z goes in the desired direction...

terjeio, 2020-05-02:
@einencool : I belive I have [fixed the issues](https://github.com/terjeio/grblHAL/commit/a80dcefd6592a74990cfd74cf3ea507602e6c590) now, Z-axis was due to me commenting the Z-axis output for debugging and forgetting to remove the comment. Jog problem was a bit more subtle and did not only affect jogging - sloppy coding from my side in an IRQ handler combined with (IMO) strange timer behaviour seems to be the reason for that.

So, please try this fix with `ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING` reenabled and report back.

Thank you for your patience.

einencool, 2020-05-02:
Hi @terjeio 
that seems to work pretty good.
The next Problem I have is while Homing. I don't know whats the Problem, when I press the Home Button, the Z-Axis homes and then the machine stands in Idle. I think the Problem is, that I have to enable default homing in the Config, but now I made a "Reset" from the Config, and now I'm trying to connect to the Teensy since 40 min and I can't get the "GRBL" Settings because it doesn't seem to connect ...
For now I give up and open a beer *g*

terjeio, 2020-05-02:
Can you connect with a terminal program like PuTTY? If so does the status response include `|Pn:` with `D` in its parameter list?

einencool, 2020-05-02:
I don't know how to connect with putty to this device, but in your program it says that the door pin is open (the pin is not connected to the breakout board, because it is on the underside of the Teensy and this pins I didn't solder / that's why I also didn't have the eeprom)
I tried to invert the signal in the config, but this didn't solve it.

When I start the "UGS Platform" and try to get the Status, it says Grbl didn't finish booting...

terjeio, 2020-05-02:
Ok, then comment out line 133 in driver.c to kill the door signal permanently.

//#define SAFETY_DOOR_PIN     (29u)

This signal is a pain because it disables everything when asserted...

einencool, 2020-05-02:
👍 
Just right now, I took the Teensy out of the Breakout Board and soldered the 5V USB Pin bridge.
When I only connect the Teensy without the Breakout Board, then everything will be displayed as normal, only the signals are red, because no Signals are on the board.
Now in your Software all 6 Buttons are shown and I can access all the GRBL Settings.
When the Breakout Board is connected and the 5V Pin is disconnected, then only 4 Pins are shown and I can't change anything in the Settings.

But now I'll disable the Door Pin 👍

terjeio, 2020-05-02:
Ok, I wrote driver.c, the setting is in driver.h. Sorry about that.

phil-barrett, 2020-05-02:
Sorry to be a bit late to this.  Einencool, did you earlier cut the VIN/VUSB connection on your Teensy 4? (see red circle in photo)  If so, that is why the BoB wasn't giving you anything - +5V is needed to power the optos for all the inputs.
[image]

Inverting the door pin has worked in the past. But I just checked an up to date build with a bare T4 and TSender and there seems to be a problem.  I can invert the door signal and it shows as inverted but doesn't allow me to reset/unlock.  When I save grbl settings and power cycle the T4, the door pin reverts to uninverted. Inverting other pins like limits works and persist across power cycles so it is saving some of the settings at least.

einencool, 2020-05-02:
No problem, after the last few days, I know  some of the interesting points :-)

I reflashed the device, set the conf8g and then I desoldered the 5v pin and put it back into the breakout board. Now everything is working as expected and the homing routine was working without any problems.

So I have to thank you for your continuous and fast support 👍 💯

phil-barrett, 2020-05-02:
And by the way, with a bare Teensy 4, you will also need to invert the e-Stop pin.

einencool, 2020-05-02:
Yes you're right, but on my, machine the "Start Hold and E-Stop" are NC- Buttons, so then there is no problem. But sadly I noticed after soldering and wiring the board, that I had to solder the inner points on the Teensy for the eeprom and the safety door.
But for now it's OK.

phil-barrett, 2020-05-02:
Yes, the Teensy 4 backside pins are an issue.  I've gotten pretty good at soldering them but managed to desolder one of the 402 caps on the first one I did.  Was able to replace it but not fun. 

The good news is PJRC is about to release the Teensy 4.1 which has even more pins and all are on the edges for easy soldering. No more back side pins.  I have a PCB design for it ready to go to the factory, just waiting to make sure they don't change anything. 5 axis and more goodies. Much more DIY friendly.  If you want, I will send you one when done. no charge.

einencool, 2020-05-02:
Oh, that sounds really interesting.
I'll give it definitely a try, thank you for this Great offer 👍 
Do you know when it will be available?

phil-barrett, 2020-05-02:
I don't know when. PJRC says they will release the T4.1 in May (Covid delays are possible) and it will take a week for me to get the first PCBs.  I'll will need a few days to verify and test so probably some time in mid June.  I'll let you know when I know better.

einencool, 2020-05-02:
That sounds good. I think in the next 3 weeks I'll get my new JMC Motors and then it'll be a good time for changing the pcb 😀

terjeio, 2020-05-03:
> Inverting the door pin has worked in the past.

This depends on it asserted state, if asserted then grblHAL (and grbl) enters suspend mode and only responds to real-time commands. This is the reason why the door input is sometimes a real pain, and I am contemplating disabling it as default.

terjeio, 2020-05-03:
Safety door switch handling is [now disabled in the core](https://github.com/terjeio/grblHAL/commit/0b35ef4154afb09f7f91b32957548f77cbe6f232) by default.

einencool, 2020-05-03:
Hello  @terjeio ,
today I made a first "Dry Run" without my spindle, because I'm changing the Fan for the China Spindle with an electrical 12V Fan.

The physical "Start" Button seems not to be recognized by the Software. The "Red Light" goes on, when I push the button, but the Software doesn't start the GCode. Maybe I'm missing something, that I have to enable it, but that was one point.

Another question is, that I have some "Comments" in the GCode File from Fusion 360, which part of the Code is running, sometimes when I copy & paste things in F360 then it happens, that I got a comment with two "))" e.g. "Bohren (2))" When that happens, the Software stops working. is there a way that your software can ignore this part? Because this could always happen to me, that there are two "))" signs.

Then I had a Piece with an Arc and then it stops at one Point, I think its because the "J" Value was not in tolerance or something like this. Can I change the behaivior with the GRBL "Arc" settings?

Thank you for this Software, there are some Points where I have to learn with it, because it acts not like the "bCNC" I was using for 3 Years, but the Realtime Jogging is really good now. 

The biggest thing I'm missing is the progress bar, so that you can see the time and the percentage.
For today the Macros are untested, I think, that I'll try them in a few days, when I have made the first Piece to upgrade my Z-Axis, because now the Plate is from wood, and the Aluminium is waiting to be machined :-)

terjeio, 2020-05-03:
> The physical "Start" Button seems not to be recognized by the Software. 

This is a regression I am aware of - will try to get it fixed for the next release.

> ..that I got a comment with two "))" e.g. "Bohren (2))"

This could/should be fixed in the Fusion post-processor as it is illegal G-Code.

> ...I think its because the "J" Value was not in tolerance...

Again, this could be a post-processor problem. I do not know if changing the Arc resolution setting ($12) will help. You may just try. Also, posting a few lines of the offending code may shed some light on the problem.

> The biggest thing I'm missing is the progress bar...

The scroll bar in the Program list is a kind of progress bar. Next version will show the number of blocks (lines) in the status bar. Display of estimated time/remaining time is way down on my todo list.

einencool, 2020-05-07:
Hi @terjeio 
I would like to give a little Update about your Sender after some short tests with my machine.
Yesterday I configured my Playstation Controller on the PC and it works beautiful with your Software 👍 
The Jogging is really nice and works like a charme. Also the fast movements and the „step function“  is great.
At the Moment I don‘t have a Probe connected to the machine, that will follow in the next days and then I will begin testing with some macros.
Does the „Home“ Button also have a Shortcut? Then I would like to map it to my Controller. Otherwise I‘ll make a Macro for it.

Can you tell me, if I put the GRBLHal to Compatibility Mode, does the Realtime Jogging feature still function with your Sender? Or will it be disabled? I think in the beginning I will also test with another Software like UGS Platform to compare them...
Do you know which Compatibility Mode I have to use for UGS? If not, no Problem, I‘ll try it :-)

So thank you for your Software and I‘ll keep testing 👍

terjeio, 2020-05-07:
Good to hear that you find the combo useful. Probing needs testing/more work - at least tool length probing/offset is not complete...

> Does the „Home“ Button also have a Shortcut? 

No, but I can add `<ALT>+<H>` for that.

> Can you tell me, if I put the GRBLHal to Compatibility Mode, does the Realtime Jogging feature still function with your Sender?

Yes, however you may need to enable it in _Settings: App_. There will be a checkbox visible in the _Keyboard jogging_ box when enabling is required.

> Do you know which Compatibility Mode I have to use for UGS?

No, so you need to try yourself. Perhaps I should add a list of senders to the wiki with settings known to work. Be aware that some senders may partially work with the lower settings values.

einencool, 2020-05-09:
Hi @terjeio ,
Today I made some Tests with the Sender, and I think the Software has a lot of potential.
I have some Issues with the shortcuts.
When I press "Alt-R" to start the Program, something in windows also recognize it, but I don't know what it is. 
I like your implementation of the Macros.
I mapped the F1 key and so on onto my Playstation controller and then I can start the Z-Probe and so on. That works great. First I thought that the extra question about running the macro was a little bit to much, but now I like it. I also mapped the "Enter" key to the Controller and then I can start it.

Is it possible to sort the Macros in another order? Because in the beginning I was putting all my macros into the Programm and then I saw, that they were aranged in the Order I entered them. It's just for the future, because I map them to my controller...

So wish you a nice weekend and stay tuned :-)

terjeio, 2020-05-09:
`Alt+R` problem could be due to me handling shortcuts the same way as I handle the keyboard jogging keys. This means that if a MDI, DRO or other input field has input focus they will be ignored. I will look into that.

Sorting macros by dragging them? Will have to put that on my todo list.

Have a nice weekend too!

einencool, 2020-05-11:
Hi @terjeio 
I downloaded your Alpha 18 and the new GRBLHal Version, will update my machine tomorrow.
While I downloaded the new Programm, I've seen, that I have to copy the File "App.config" from Nr. 17 so, there are all the Macros listed, then I can arrange them like I want them, so I think you don't need to make the Macros sortable, it's only to do one time, and then they can be, where they are :-)

Then I'll try the "Alt+H" Button too, or I'll put it into a macro, then the Software asks, before it will be started. I really like that 👍 

Is it possible to attach the following commands to macros?
$H --> Homing
~ --> Start Programm

Then it will always be asking if the command should be started...

Wish you a nice evening.
Greets Chris

einencool, 2020-05-12:
Today I tested a little bit more with the Teensy GRBLHal (from 02.May.20) and your Sender Alpha18
I must say, it seems to be the perfect Combination for me :-)

The Realtime Jogging with the Playstation Controller is absolutely the best I can imagine 💯 
One Stick for X + Y and the other for Z-Axis
And the Digit Controller for 1-Step Jogging for X+Y

And some other Buttons for the Macros.
What I didn‘t get to work is when I Program a Button of the Controller to „ALT + R“ It doesn‘t start the Job, but today I‘ve seen, that the „Hardware Start Button“ is working. Thank you so much for this, because that is so good to start the job when I‘m sitting directly in front of the machine...

I changed the Postprocessor from Fusion360 so that this also works really good for me.

I haven‘t tested your Probing tab, because in the moment I only use a macro for my Touchplate.
But is there the possibility to set an Offset for another Touchpin?

I have an extra ball head on a pin that has an offset of about 80 and 5mm to my cutter. But this is only for X and Y, the Z height is set by another touchprobe

terjeio, 2020-05-12:
Again good to hear that it is working out great for you, thanks for the feedback.

Probing is work in progress, hopefully this part of the program will end up useful too. A lot is still missing and it is a bit dangerous to use. Touchplate offset is not yet implemented. Give me time...

einencool, 2020-05-12:
Oh, that's no problem, I'll work with macros and that works fine. I got a touch plate for z axis and another one for x and y.

And in the macro I can set the distance to my needs and also start it directly with the controller :-)

Is there a way to work with an directly installed z-Probe and with M6 Commands?
So when I have pieces with several endmills that the zero point will automatically be calculated

[33 more replies]

closed 2020-10-07 by terjeio