terjeio/ioSender #14: 4 axis visual glitches

phil-barrett, 2020-02-28:
not urgent but for reference.
[image]

terjeio, 2020-02-29:
Next alpha version hides (for now) _Program limits_ if  `N_AXIS` > 3, will expand jog panel later. Do we want keyboard jogging for A, B and C axes? Use `<Alt Gr>` and bind to cursor keys?

I hope I am able to publish next alpha later today - testing takes a lot of time...

phil-barrett, 2020-03-09:
I may be missing something but did you not release a .exe?

terjeio, 2020-03-09:
I did, all can be found via the [Releases](https://github.com/terjeio/Grbl-GCode-Sender/releases) icon on the main page:
[image]

phil-barrett, 2020-03-09:
My bad.  sorry.  Installed and starting an overnight run.

closed 2020-05-20 by terjeio