terjeio/ioSender #17: spindle group CW and CCW eat RPM digits.

phil-barrett, 2020-03-11:
Alpha 11.

Set $30, max spindle speed to 1000, save.
ensure off in spindle group is set.
enter 1000 in the RPM field.
click on CW or CCW. note that RPM has changed to 99.
click Off, note that RPM is still 99.
Clock on CW or CCW, note that RPM has changed to 9

phil-barrett, 2020-03-11:
I resarted TSender and was not able to reproduce the behavior.  Not sure what's going on.

terjeio, 2020-03-11:
It seems like a regression...

How it is supposed to work:

The RPM field has dual functions, when spindle is off it is enabled  and shows the last entered value either from direct input or M3/M4 commands (from the last program run or MDI input). When spindle is running it is disabled and shows the _real_ RPM. If no spindle encoder is attached the _real_ RPM is the programmed speed plus the override value.

I will try to fix this by moving logic from the spindle control to the viewmodel.

terjeio, 2020-03-11:
New release uploaded, I hope this works as it should.

closed 2020-05-20 by terjeio