terjeio/ioSender #18: Testing Alpha 13.

phil-barrett, 2020-03-16:
Installed and played with it last night. Seems pretty solid though still has some rough behavior around Hold and (re)Start that I think you are aware of.  I believe the same problem exists with using the feed/hold signal.

Tabbed interface looks ok. I assume you will remove the the top 3D View tab.

In the signal section, the indicator boxes (not sure the right word for them) are clickable and change to light grey when clicked. What does that mean/do? 

I have been assuming that in the Signals section, a red box means the signal is active (as in button pushed or end stop hit). This presumes that inputs are pulled high, I believe.  Thus, if inverted in Settings: Grbl, a red box means a high level signal is present. i.e. the normally low input signal is now high.  Is this correct?

Ran a long job overnight and when I looked this morning, it was gone. Not sure what happened. Running same job again. Is there some sort of logging I could turn on?

phil-barrett, 2020-03-16:
Hmmm, was running same job and it simply halted after about 2 hours. TSender wasn't completely dead but I clicked on file and it hung. I restarted it. Is there anything you want me to look for if it does the same thing again?

terjeio, 2020-03-16:
Hold/Start works ok for me now, both the buttons and signals input. What is the rough behaviour you are seeing? 

I will let the 3D tab stay there for now, it is larger and has the _Program limits_ overlaid. 

Signals displays the asserted (physically engaged?) state regardless of pins are set as inverted or not. I prefer to use NC switches with pullup enabled, and I have decided to use that as the default mode for grblHAL (unlike the NO default for vanilla grbl which I do not like).

> Ran a long job overnight and when I looked this morning, it was gone.

You mean the sender was gone? If due to a crash the reason should be logged in the Windows event log.

terjeio, 2020-03-16:
> Hmmm, was running same job and it simply halted after about 2 hours. 

Was this with your concentric circles test file? And MKRZERO?

What happens if you press the Stop button, does it respond normally again?

I am asking since I saw a similar problem with the Due yesterday, it just hanged - but only for a few minutes before it started again. Problem was with the Due driver.

phil-barrett, 2020-03-16:
Rough behavior -  pressing hold button in TSender or pressing external hold button (NC, pulled high so "un-shorting") causes TSender to hang.  

Here's the crash info from the log. Time stamp was 3/16/2020 2:15:50 AM (my overnight run). Alpha 13.

Faulting application name: GCode Sender.exe, version: 2.0.0.0, time stamp: 0x5e6df1b2
Faulting module name: KERNELBASE.dll, version: 10.0.17134.1067, time stamp: 0xe94e2b6e
Exception code: 0xe0434352
Fault offset: 0x00114592
Faulting process id: 0x3788
Faulting application start time: 0x01d5fb66070a02ac
Faulting application path: D:\Documents\cnc\gcode sender\GCode Sender.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report Id: 2f4a2fc1-26d2-4f4f-960d-41d343c86e8d
Faulting package full name: 
Faulting package-relative application ID:

phil-barrett, 2020-03-16:
Sorry, I should be putting more info in my reports.

MKRZERO Version. Built with standard driver.h.
It was a different test program - concentric circles as segments. Basically the same but with arcs turned off. lots of very short movements. am attaching a zip of it (it's huge).
[concentric circles as segments fr5000.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/4339859/concentric.circles.as.segments.fr5000.zip)

terjeio, 2020-03-16:
> Rough behavior - pressing hold button in TSender or pressing external hold button (NC, pulled high so "un-shorting") causes TSender to hang.

Does not happen for me. "Hang" means totally unresponsive? Can you select the Console tab and check verbose mode? If so, what is the output?

phil-barrett, 2020-03-16:
Stop and start work ok. Pressing hold causes a hang.  Exit doesn't work.  If I unplug the MKRZERO, TSender will exit.  TSender with the T4 version has the same behavior in Alpha 12.  Will check it in a bit.

phil-barrett, 2020-03-16:
TSender A13, console verbose. Loaded and ran job.  Hit Hold and it froze.  Here''s the screen shot.  Not sure it's very helpful, though.
[image]

phil-barrett, 2020-03-16:
Here's the event log fault for that crash (when I unplugged the MKRZERO USB cable).
Faulting application name: GCode Sender.exe, version: 2.0.0.0, time stamp: 0x5e6df1b2
Faulting module name: KERNELBASE.dll, version: 10.0.17134.1067, time stamp: 0xe94e2b6e
Exception code: 0xe0434352
Fault offset: 0x00114592
Faulting process id: 0x220c
Faulting application start time: 0x01d5fbc478cd48ba
Faulting application path: D:\Documents\cnc\gcode sender\GCode Sender.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report Id: 0644bf9b-1781-4019-a532-67e830fcd73c
Faulting package full name: 
Faulting package-relative application ID:

terjeio, 2020-03-16:
Ok, has now done a quick test with your new file and four different processors. All responds correctly to both Hold/Start and signals input. Do you get the same behaviour with T4?

terjeio, 2020-03-16:
Oops - I did not see your previous three comments. My guess is that MKRZERO is crashing. Is the port visible after a hang and can you connect with a terminal and get responses?

phil-barrett, 2020-03-16:
Tested A13 with the T4 version. Ran CCasSegments job. Hold does NOT cause a hang. Start resumes the run.

By the way, I don't think this is a short freeze.  TSender is completely frozen. I let it sit for 10 minutes with no change.  No controls work.

terjeio, 2020-03-16:
BTW, which grblHAL core version are you using? If not the latest (20200310) then this is likely  to be the reason.

phil-barrett, 2020-03-16:
Ah, I am using one from 20200308.  Will rebuild and test.

phil-barrett, 2020-03-16:
Yes, that was it. Works fine now. Not sure why my T4 build using the same core didn't fail.

terjeio, 2020-03-16:
It is because the T4 build polls the USB input on Systick events (in interrupt contect), MKRZERO polls from the foreground "process" via `hal.execute_realtime`. `hal.execute_realtime` was not called when in suspend mode (Hold, Sleep) before 20200310.

In the next commit for T4 I will change `USB_SERIAL_GRBL 1` to poll via `hal.execute_realtime` and leave `USB_SERIAL_GRBL 2` as is.

I forgot to mention I have fixed the mouse over/click issue with the signals control.

terjeio, 2020-10-07:
Fixed.

closed 2020-10-07 by terjeio