terjeio/ioSender #19: minor issue - elapsed time runs during hold

phil-barrett, 2020-03-20:
Feature or bug?  I don't know.

terjeio, 2020-03-21:
Feature? Depends on what the elapsed time is used for, if for calculating charge for customer it should run? I decided it should run, can be changed - or perhaps make the behaviour configurable? I dont't know either...

phil-barrett, 2020-03-21:
I have thought a bit more about this since posting it.  

Why leave it running? Biggest reason is to know entire length of the job include fussing with fixtures and such.  

Why pause the timer during hold?  Knowing actual job running time. If I pause a job to walk away from the machine for a while (run an errand, attend to an interruption, ...) I would prefer to have the timer pause as well.

Does the timer pause during tool change? I haven't tested that out but I'd guess it doesn't.  What do motion controllers like Mach and UCCNC do?

The safest route is a configuration option (or app setting).

This also brings up a separate question on the GRBL side about accumulated machine running time but I will leave that for another day.

Hope you are staying safe. Things are getting pretty harsh around here.

terjeio, 2020-03-21:
I think I will add this as an option to app settings. It does run during a tool change, and it should continue to do so regardless? I will have to check what Mach 3 does - I haven't even noticed any timer when using that but can see from web images that is has one.

Accumulated machine time would require (IMO) FRAM for non-volatile storage - to be updated on transitions to `Idle `state?

This far everything is ok for me, I am lucky to live in the countryside and work from home anyway. Easy for me to take the neccessary precautions. Take care, it is a nasty virus.