terjeio/ioSender #21: probe and probe plate discussion

jschoch, 2020-03-28:
> BTW: do you own some probes/plates? I have none so I am thinking about using a piezoelectric element for probing. 

Probing is a big yack shaving diversion so I though I'd break this out into another thread.

My "pcb mill" I use a HDPE plate for my PCBs so I just need to have the spindle grounded and connect probe positive the PCB surface.  I made a puck with a spring in it which I just set onto the PCB.  This seems to work well and I don't think I want to complicate that setup any more.

My lathe and "micro mill" both have ground continuity throughout so the above method doesn't work.  My simple solution is to use some acetal to isolate a ground pin.  These are machined to a press fit.  The connection wire is soldered to a spring and then fit the spring over the pin.  You could also thread/tap something into the pin.

I've thought quite a lot about non contact type solutions.  

I did a prototype using velostat.  It works but the signal wanders and tapers in a big curve as the material decompresses.  I don't think this is a big deal if you are not doing lots of quick probe cycles. Using a microcontroller to manage what is a contact signal is easy enough and I think the approach has some merit but I have not poked at it for a while.  This is perhaps the easiest to construct and very cheap.  The complexity is in the signal processing.

Using a similar principal strain gauges can be used.  This requires a high resolution ADC for each strain gauge Wheatstone bridge.  I bought some BF350 and HX711 ADCs but I have not yet tested them.  another project for my long backlog.

There is also the Kinematic coupling approach.  you can use ground pins and ball bearings.  I attempted to use a PCB to make one but it didn't work out great.  For a non-PCB type the hard part is soldering to the ball bearings.  On my todo list is trying plumbing flux to do the soldering.  Silver solder is likely ideal but I don't have any.  I also have a tig welder now so there may be a way I can use that.  The other components are fairly easy to machine and there is a multitude of designs for these style probes.  The complexity here I believe is setting it to be concentric which typically uses a system like a 4 jaw independent chuck.  you can position the probe offset but then you have to figure out how to make it perpendicular to your work surface.

My pie in the sky project would be to make a 3d taster.  The principal they work on is pretty simple but it requires machining a very accurate "cup" for the X and Y travel.  It is essentially a regular dial indicator with a matched cup so that movement in the X/Y pushes the cup up.  There is only one sensor/gear for the dial.

For plates I like this design:  https://entirelycrimson.com/products/cnc-edge-finder-for-diy-router-machines.  the hard part is holding the plate in registration but that design can do inside/outside corners and is easy to machine.  The other downside is that if you forget to connect something you are going to crash.  

The spring type "button" is nice for z height or tool offsets because you can set your probe limit distance to match your spring travel.  The tool can still crash but it will only push the spring down.  I have snapped a number of small end mills due to coatings on the PCB or bad contact with the PCB.

You could also make a HDPE or acetal sandwich by epoxying a plate of conductive material and then re-milling both sides to ensure parallel.  This would let you attach use the fact that the spindle is grounded and you could probe the actual tool you would be using.  Copper tape may be a quick and dirty way to do this but wouldn't last long.

While there are many fun diversions the simple continuity type probes are so easy and effective it is hard to justify the time to make another type.  My lathe has enough backlash issues that I don't bother probing or setting my tool offsets.  I have a ballscrew conversion planned but it works now and it is hard to mess with something that is working.

terjeio, 2020-03-29:
[New release](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v0.14-alpha) with probing tab up now.

Thanks for input on probes/plates, will make some when I take a break from coding. Hopefully the weather here will soon improve making work in the shop more attractive...

While working on probing and seeing what can go wrong I have been pondering if the probe input should be reconfigured to act as an e-stop (or reset) input for non-probing movements during the probe sequence. Might save the probe if (when?) something go pear shaped...

jschoch, 2020-03-29:
On my centroid acorn as well as standard grbl I believe probe active throws an error.

terjeio, 2020-03-29:
Not in grbl for sure, the probe input is polled in most (all?) versions I have seen. If an error is thrown then it must be detected and issued by the sender. It needs to be assigned to an interrupt and (programmatically) rerouted to a control signal for the firmware to halt motion and issue an error.

If a probing command is issued while the probe signal is asserted grbl will throw an error, but not otherwise.