terjeio/ioSender #23: G93 G94 Issue

S2000Stefan, 2020-04-10:
Hello terjeio,

testing your GCode-Sender and GRBLHAL, i am enthusiastic, works very well despite alpha status. I come from GRBLPanel and i am looking for a station that can also display my a-axis.
Unfortunately i have a little problem with G93 and G94 commands.
When i start GCode-Sender with GRBLHAL, G94 is activ. If i now enter G93 everything is fine and the g-code is running. Comes now a single G94 cammand error22 is coming, G94 a200 f1500 as a line works.
The other way around a single G93 command to G94 works fine.
Cheers,
Stefan

terjeio, 2020-04-10:
Good to hear you find the combo useful.
As for your problem I need a bit more input to be able to replicate. If you post the gcode program that fails, or a few lines before and including the offending line, that would be helpful.

GrblHAL is a bit stricter than vanilla Grbl in that when an error occurs it will be reported for all subsequent commands until a reset or a system command is issued. This, in combination with the sender beeing out of sync, may be cause the error report to be shown for the wrong line. I suspect this may be the case as G93 mode requires a F parameter for each commanded cut motion afterwards, and I am not able to replicate your error with a G94 command followed by motion and a feedrate.

From the linuxCNC documentation:

> When the inverse time feed rate mode is active, an F word must appear on every line which has a G1, G2, or G3 motion, and an F word on a line that does not have G1, G2, or G3 is ignored. Being in inverse time feed rate mode does not affect G0 (rapid move) motions.

S2000Stefan, 2020-04-10:
Thanks for your quick reply. Yes, i could have thought of sending the g-code. 
I have already tried different variants of the postprocessor with the same result.
Here are a few screenshots of the error messages:
[image]
[image]

And here the g-code:
G17
G21
G90
M3 S16000 G4 P3
G00 Y-30.000 Z33.000 A0.000
G94
 F1100G01 Z28.000
G93
 F191.2G01 Z27.000 A12.245
 F191.2G01 Z26.000 A0.000
 F9.4G01 A360.000
G94
 F1650G01 Y-29.954
G93
 F12.5G01 A720.000
G94
 F2200G01 Y-30.000
G93
 F205.9G01 Z25.000 A732.245
 F205.9G01 Z24.000 A720.000
 F10.1G01 A1080.000
G94
 F2200G01 Y-29.954 Z25.615
G93
 F13.5G01 A1440.000
G94
 F1540G01 Y-30.000 Z24.000
G93
 F223.0G01 Z23.000 A1452.245
 F223.0G01 Z22.000 A1440.000
 F10.9G01 A1800.000
 F243.3G01 Z21.000 A1812.245
 F243.3G01 Z20.000 A1800.000
 F11.9G01 A2160.000
 F267.6G01 Z19.000 A2172.245
 F267.6G01 Z18.000 A2160.000
 F13.1G01 A2520.000
 F1000.8G01 Z17.750 A2524.898
 F1000.8G01 Z17.500 A2520.000
 F19.5G01 A2880.000
G00 Z33.000
G00 Y32.750 A2520.000
G94
 F1100G01 Z28.000
G93
 F191.2G01 Z27.000 A2532.245
 F191.2G01 Z26.000 A2520.000
 F9.4G01 A2880.000
G94
 F1650G01 Y32.800
G93
 F12.5G01 A3240.000
G94
 F2200G01 Y32.750
G93
 F205.9G01 Z25.000 A3252.245
 F205.9G01 Z24.000 A3240.000
 F10.1G01 A3600.000
G94
 F1650G01 Y32.800
G93
 F13.5G01 A3960.000
G94
 F2200G01 Y32.750
G93
 F223.0G01 Z23.000 A3972.245
 F223.0G01 Z22.000 A3960.000
 F10.9G01 A4320.000
G94
 F1650G01 Y32.800
G93
 F14.6G01 A4680.000
G94
 F2200G01 Y32.750
G93
 F243.3G01 Z21.000 A4692.245
 F243.3G01 Z20.000 A4680.000
 F11.9G01 A5040.000
G94
 F1650G01 Y32.800
G93
 F15.9G01 A5400.000

And the other one:

G17
G21
G90
M3 S16000 G4 P3
G00 Y-30.000 Z33.000 A0.000
G94
G01 Z28.000 F1100
G93
G01 Z27.000 A12.245 F191.2
G01 Z26.000 A0.000 F191.2
G01 A360.000 F9.4
G94
G01 Y-29.954 F1650
G93
G01 A720.000 F12.5
G94
G01 Y-30.000 F2200
G93
G01 Z25.000 A732.245 F205.9
G01 Z24.000 A720.000 F205.9
G01 A1080.000 F10.1
G94
G01 Y-29.954 Z25.615 F2200
G93
G01 A1440.000 F13.5
G94
G01 Y-30.000 Z24.000 F1540
G93
G01 Z23.000 A1452.245 F223.0
G01 Z22.000 A1440.000 F223.0
G01 A1800.000 F10.9
G01 Z21.000 A1812.245 F243.3
G01 Z20.000 A1800.000 F243.3
G01 A2160.000 F11.9
G01 Z19.000 A2172.245 F267.6
G01 Z18.000 A2160.000 F267.6
G01 A2520.000 F13.1
G01 Z17.750 A2524.898 F1000.8
G01 Z17.500 A2520.000 F1000.8
G01 A2880.000 F19.5
G00 Z33.000
G00 Y32.750 A2520.000
G94
G01 Z28.000 F1100
G93
G01 Z27.000 A2532.245 F191.2
G01 Z26.000 A2520.000 F191.2
G01 A2880.000 F9.4
G94
G01 Y32.800 F1650
G93
G01 A3240.000 F12.5
G94
G01 Y32.750 F2200
G93
G01 Z25.000 A3252.245 F205.9
G01 Z24.000 A3240.000 F205.9
G01 A3600.000 F10.1

Hope that helps.
Cheers
Stefan

terjeio, 2020-04-10:
Thanks,  it seems I have made a mistake when I added G95 support - sorry about that.

Try with removing line 869 and 870 in grbl/gcode.c as a temporary fix:

https://github.com/terjeio/grblHAL/blob/cdab733cae58b598ecc849a01008a5157ad93bcd/grbl/gcode.c#L869-L870

I believe this is going to be the permanent fix but will have to check if my G95 implementation requires this check/error before I commit an update. It should not...

S2000Stefan, 2020-04-11:
terjeio, you make my day.  ;)
Now is running like a charm. I make little tests yesterday without problems.
You dont´t have to apologize, a wonderfull piece of software.
Longer tests today and i think it is time to go to real machine. 
Thanks again for the quick help.
Cheers
Stefan

closed 2020-04-11 by S2000Stefan