terjeio/ioSender #25: Question about probing - height map

phil-barrett, 2020-04-17:
This is potentially a very exciting feature.  How does (will?) it work? Does TSender modify Z values as they are sent to conform with the height map? Does this require grblHAL support? i.e. could it work with legacy grbl systems? I think engravers would love this feature. PCB milling, too.

Are only linear height maps supported? I can see it useful for carving on curved surfaces.

terjeio, 2020-04-17:
> How does (will?) it work?

By modifying Z values by linear interpolation.

> Does TSender modify Z values as they are sent to conform with the height map?

No, the program is replaced in-memory with a transformed version created when _Apply_ is clicked - use _File > Save_ if you want to save it. Programs are tokenized on load, the tokens are transformed and a new program is generated from these. The same process is used for the _File > Transform_ options, and is one of the reasons I am still marking the sender as alpha quality. For most programs this will work, but it is not yet able to handle the complete range of g-codes. E.g transformed result will always be in metric and offsets are not yet handled as they should. Pending work... The testing I have done is probing a deliberately bent PCB (approx. 8mm delta) and checked that the tool follows the surface. No milling yet (I use my homemade laser exposer for making PCBs).

> Does this require grblHAL support?

No.

> Are only linear height maps supported?

Yes, but I guess it could be used for curved surfaces depending on the resolution of the map. I am using code from [OpenCNCPilot](https://github.com/martin2250/OpenCNCPilot) to do the transformation. I need to add credit for that - as well as for numerous snippets sourced from Stackoverflow...

benishor, 2020-12-28:
Is there a place where I can find documented the usage of height map (probing and usage per-se)? I tried to walk the intuitive path but after applying the probed map to my program and running a cycle start, I ended up banging my probe right through the PCB I was trying to mill. Thank you!

terjeio, 2020-12-28:
> Is there a place where I can find documented the usage of height map (probing and usage per-se)?

Not yet.

> I tried to walk the intuitive path but after applying the probed map to my program and running a cycle start, I ended up banging my probe right through the PCB I was trying to mill.

Did you zero Z at X0Y0 (at the workpiece surface) before starting? There is a check box for that which does that automatically at the end of probing the map.

benishor, 2020-12-28:
Yes, I did. I'm not sure what went wrong, either a case of pebkac or a bug. That's why I needed the docs, just so I could render out a very likely user error.

terjeio, 2020-12-28:
Can you upload the file you used for me to run a test with?

benishor, 2020-12-29:
Sure, I added it to this post. Eversince I first commented, I retried the height map probing process and got the same results; the probing works out fine, a map is being generated but machine Z seems all funked up when applying the map to the program. 

Can you please lay it out for me what actually happens during the probing process and upon applying the map to the loaded program? 
[frezare-top-one-pass.nc.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5750038/frezare-top-one-pass.nc.txt)

phil-barrett, 2020-12-29:
Could you save the map in Probing/Height Map and post it here?  Also, save the transformed Gcode and post it here as well? I transformed your file on my test system with a map file I had and then ran the code (no actual motors).  Following on the 3D view, it seemed to run ok but my map file is fairly tame.

terjeio, 2020-12-29:
> Can you please lay it out for me what actually happens during the probing process and upon applying the map to the loaded program?

During probing z surface positions are collected for the points probed, when finished offsets for each point are calculated with the probed Z position @ X0Y0 used as the zero height reference.  If _Set Z = 0 at X0Y0:_ is checked either the selected coordinate system or G92 offset is set so that Z becomes 0 at X0Y0. If not checked Z0 at X0Y0 has to be established in a separate step before milling. Do not set a different coordinate system than specified in the gcode file or used as default!

When the heightmap is applied the gcode is changed so that the Z position follows the contour of the probed object. This is done by interpolation and adds XY waypoints as needed to the gcode.

I probed with default settings, applied the map and ran the program and it went smoothly for me. I have tested with both setting G54 and G92 offset for Z0. I have a deliberately bent PCB to test on, this so that when the map is applied I can see the result in the 3D viewer:

[hmap.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/5751006/hmap.zip)

Do you probe with the tool you are going to mill with or a separate probe with a different length? If different you have to set Z0 at X0Y0 with the tool at the surface before starting the program.

benishor, 2020-12-30:
I am probing with the same tool I am using. Here are some of the steps during a probing process: https://imgur.com/a/g7W78aH
The problem here has to do with Z being set to 72 after probing. Based on your explanation I think I can simply uncheck _Set Z = 0 at X0Y0_ altogether since I tend to zero all axes before starting the probing.
[frezare-top.map.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5753683/frezare-top.map.txt)
[frezare-top-one-pass.nc.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5753684/frezare-top-one-pass.nc.txt)
[program-loaded-and-transformed.nc.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5753685/program-loaded-and-transformed.nc.txt)

I also attached some files for @phil-barrett.

Thank you!

terjeio, 2020-12-30:
@benishor : I see that you have a G92 offset active as there is a red border around the Clear button. Check in the _Offsets_ tab if Z offset is set to around 72 for G92.

terjeio, 2021-04-29:
Reopen if still a problem with the latest release.

closed 2021-04-29 by terjeio