terjeio/ioSender #26: A17 - Jog Step: 0.05 at startup

phil-barrett, 2020-04-19:
Minor issue.  Upon startup, jog step indicator in low right corner is 0.05. Actual step in the Jog Panel is 0.1 mm. When you change distance in the jog panel, thelower left "Jog Step" value correctly tracks.

On a separate point, should it say Jog Step or Jog Distance? Shouldn't the two be consistent?

terjeio, 2020-04-19:
Working as it intended for me, but then perhaps since _Jog step_ is the distance traveled only when using keyboard jogging (with <CTRL> modifier). Maybe I should disconnect the sync between flyout and keyboard jogging distances? The initial _Jog step_ is usually from _Settings: App_, perhaps I should add that too when cycling through values. Decisions, decisions...

Note that when a keypad is connected to the controller, at least with my plugin, keyboard jogging values is read from firmware settings - this to keep jogging consistent with keypad.

phil-barrett, 2020-04-19:
OK.  Thanks.

closed 2021-04-29 by terjeio