terjeio/ioSender #27: File Open/Reload/Edit/Close controls

phil-barrett, 2020-04-19:
Perhaps the current set up is just a place-holder but they seem out of place with the rest of the UI.  They are graphic in nature (as opposed to text everywhere else). Also, the vertical dots seem to imply movability - when you mouse over it the cursor changes to a 4 arrow move cursor though you can't really move it.  Also, there is an expand down control/button in the lower right corner of that bar but it doesn't do anything.

terjeio, 2020-04-19:
Vertical dots and expand button is part of the toolbar control. Will check if at least the vertical dots can be removed. The expand also? Maybe leave it since it could be that it starts to work if a lot of macros with long names are defined?
Further graphical icons may be added - so I'll keep them from now. Perhaps a bit smaller...

terjeio, 2020-10-07:
I believe this has been fixed now.

closed 2020-10-07 by terjeio