terjeio/ioSender #28: TSender A17 exits - get "MPG Mode active" message upon restart.

phil-barrett, 2020-04-20:
I don't know if this is a grblHAL issue or TSender. I've been trying to get an easily reproducible case of this but having some trouble. Several times now with long running jobs, TSender will exit and when I restart I get this message.
[image]
Closing the message box does nothing - I have to power cycle the Teensy to get it to run. The attached zip contains a long file (concentric circles as segments fr5000.cnc) that seems to reproduce the problem fairly often.
[concentric circles as segments fr5000.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/4501815/concentric.circles.as.segments.fr5000.zip)

I have been running with aggressive buffering turned on and turned up feed rate override to 200%.

Some times TSender doesn't exit but just stops.  Run time stops advancing and the start/stop/hold buttons do nothing. File Open doesn't do anything.  Restarting TSender gets the MPG Mode message.

Note, I have not updated grblHAL to the latest changes.

terjeio, 2020-04-20:
Strange, likely to be a Teensy crash but I do not immediately understand why you get the TSender message on restart. The message is shown if there is incoming data on the serial port after connection is established (and if the controller is reset, this after the resetdelay specified in _Settings: App_).

Does Tsender exit without any error message?
If, after such a crash, you connect with a terminal instead how is that working out?
Which USB mode are you using?

phil-barrett, 2020-04-20:
Using USB serial 2.  No exit message. I will check with a serial terminal next time that happens. 

I left it to run overnight and it didn't appear to crash but there is definitely something wrong. Can't zero an axis or zero all. Jogging seems to go for random distances. A few clicks and now it's not taking any input.  I closed TSender and ran a terminal.  Here's a screenshot. It's just displaying the same lines continually. After I took the screenshot, I tried an enter and it went to sending ok continually.  
[image]

Restarted TSender but it was unresponsive.

phil-barrett, 2020-04-20:
By the way, I will rebuild with the latest updates and rerun the tests.

terjeio, 2020-04-20:
Ok, then I believe this must be due to reading data from the serial stream (in the controller) has gone haywire by somehow repeating a sequence over and over.

I am running a test now with USB serial 2 as well, perhaps I should terminate that and test with USB serial 1 to avoid duplication of efforts?

As a side note It can be mentioned I have left parsing of some data in TSender deliberately vulnerable (for now) by not handling errors - however this should result in an error message before exiting. If this occurs serial data has been garbled somewhere in the chain. I have not yet decided on what action to take when errors like this happens - generally I do not like processing to continue...

phil-barrett, 2020-04-20:
I rebuilt with latest updates. (as an aside, need a way to insert my changes like ioports automatically).

I've set USB Serial to 1 and rerunning test so keep yours going.

terjeio, 2020-04-20:
My test went haywire after 4 hours+, opened a terminal and saw similar repeating messages as you did. I did not notice before (in your screenshot) that the second Bf: value alternated between two values, this could be a clue (the second value is the number of bytes free in the RX buffer). So I issued a reset command (CTRL-X),  as this flushes the input buffers - both in the PJRC usb driver and the one added by grblHAL. Oddly then the startup message appeared repeatedly indicating that reset command was somehow reread.

Is this a bug in the PJRC driver or in my code? The alternating Bf: value indicates that the data is actually read into grblHAL RX buffer. The first thing I do when polling for data is getting how many characters are available from the PJRC driver, if none available polling is skipped altogether.

I changed to getting number of available characters on March 7 as I wanted to read a block of data from the PJRC driver in order to avoid disabling interrupts too often (the PJRC driver does this when single characters are read). From how I read my code this makes me wonder if this is a PJRC bug...

I made similar changes to the USB serial 1 as well, will this fail too?

phil-barrett, 2020-04-20:
My test (USB Serial 1) has been running for 2.5 hrs. Should take about 12 hrs.  Will let you know what happens.

terjeio, 2020-04-21:
I've run a USB Serial 2 test to completion after modifying the code a little, a test running in parallel on a MSP432 (with TSender running on the same machine) also completed successfully.

The modification I made suggests that the PJRC code may be buggy - or that it is not designed to run in an interrupt context (polling is done from the systick timer interrupt) even if PJRC claims so.

Modified code:

```
void usb_execute_realtime (uint_fast16_t state)
{
    char c, *dp;
    int avail, free;

    if((avail = usb_serial_available())) {

        dp = rxbuf;
        free = usb_serialRxFree();
        free = free > BLOCK_RX_BUFFER_SIZE ? BLOCK_RX_BUFFER_SIZE : free;

        avail = usb_serial_read(rxbuf, avail > free ? free : avail);

        while(avail--) {
```

Original (first line removed, second changed):

```
        avail = avail > free ? free : avail;

        usb_serial_read(rxbuf, avail);
```
The code change was to reassign the avail value from the `usb_serial_read()` call, which suggests to me that the `usb_serial_available()` call returns a incorrect value for some reason when this issue occurs. `rxbuf` is a 20 character "prebuffer" - this explains why the same input was repeated over and over when `usb_serial_available()`  returns an incorrect value.

I'll run some further tests to see if running the polling from a interrupt context is the real reason behind this issue, or if there is a straightforward bug in the PRJC driver.

USB Serial 1 polling is done from the foreground process, I will not be surprised if this doesn't fail.

phil-barrett, 2020-04-21:
Using USB Serial 1, I am now 14 1/2 hours into the test run (forgot to override feed rate). So, you are correct, hasn't failed.  Once you push the changes, I'll rebuild and test with USB Serial 2.

By the way, the Teensy 4.1 is going into beta test, I've put a request in for a beta copy. Hopefully I can get one.  It definitely will support ethernet so you'll be hearing some questions on that from me I'm sure.

terjeio, 2020-04-23:
Commit [cc7c59a ](https://github.com/terjeio/grblHAL/commit/f730870165c24c3c5b4ef5ac3de720236cc7c59a) should hopefully end the USB saga. I have run a number of long tests with both USB 1 and 2 with no issues.

The PJRC driver does not like to be run in an interrupt context, I guess the USB IRQ has lower priority than the systick timer IRQ and that is the reason behind it randomly failing.

phil-barrett, 2020-04-23:
OK, will download, build and test today.  Thanks for tackling this problem, not a simple one.

closed 2020-05-20 by terjeio