terjeio/ioSender #29: A17: oddness with keyboard editing of grbl settings

phil-barrett, 2020-04-24:
This appears to be a bug in the keyboard manipulation of 4 digit numbers in text boxes.  My example is for Id 25 (Homing search seek rate) but it appears to apply to all numeric text boxes..

- Open the Settings: Grbl tab.
- Select id 25 by clicking on the row in the table on the left (Homing search seek rate). Enter 5000 in the text box.  Press save (or enter).
- note that the entry in the Value column changes to 5000
- press reload. Note the Value column entry has added .000 to the value

(the above is just to set the situation up to demonstrate the bug)

- select Id 25 as above
- place the cursor after the 50 in the 5000.000 shown in the text box on the right.
- delete a 0 so that the box shows 500.000 and press enter.
- note that the Id 25 Value column entry changes to 5000 (not .000 following).
- press save and then reload.  the Value column entry has changed to 5000.000 and exit box on the right still shows 500.000
- expect results are 500.000 in the Value column entry for Id 25

This appears to be the case for all values edited, not just Id 25. Note that if you change the leading digit or completely rewrite the value, it works correctly.

terjeio, 2020-05-01:
Looks like a framework bug to me, the `OnPreviewTextInput` event does not fire on all changes. I have added a handler for `OnPreviewKeyUp` to trap and process delete and backspace keys as a workaround.

phil-barrett, 2020-05-01:
Makes sense to me.  They probably a college intern working on that part... Thanks for fixing it.

closed 2020-05-15 by terjeio