terjeio/ioSender #31: Feature request

phil-barrett, 2020-04-24:
It would be nice to have total GCode file line count shown in the UI.  Perhaps next to run time. Grbl Panel does this and it is nice to get a sense of how big the program is. I use this to get a crude estimate of how long the program will take to complete.

Clearly an accurate completion prediction is useful though I believe it is a _very_ hard problem to solve.

terjeio, 2020-04-24:
Like this?

[image]

Estimated run time is way down on my todo list - lots of other stuff to complete first.

phil-barrett, 2020-04-24:
Yeah though I'd prefer total line count.

I totally understand why run time estimate is low on your list.

terjeio, 2020-04-24:
Blocks = lines, what is best to use?

> The RS274/NGC language is based on lines of code. Each line (also called a “block”) may
> include commands to a machining center to do several different things. Lines of code may be
> collected in a file to make a program.

grblHAL has support for the Block Delete Switch:

> 2.2.2 Block Delete Switch
> If the block delete switch is on, lines of RS274/NGC code which start with a slash (the block
> delete character) are not interpreted. If the switch is off, such lines are interpreted.
> As outlined in Section 1.3.1, the Interpreter runs in two stages (read and execute). The driver tells
> the Interpreter when to perform each stage. When the Interpreter reads a line starting with a slash,
> it informs the driver, “I just read a line starting with a slash.” The driver checks the setting of the
> block delete switch. If the switch is off, it tells the Interpreter, “Execute that line.” If the switch is
> on, the driver does not tell the Interpreter to execute the line. Instead, it tells the Interpreter to read
> another line, with the result that the line starting with the slash is not executed.
> In the SAI, the block delete switch may be set, and its default setting is off.

Will add UI element for this later (if the firmware does not support an input for it).

phil-barrett, 2020-04-24:
Ah, did not know the block == line. thanks.

closed 2020-05-15 by terjeio