terjeio/ioSender #32: short probing tests

S2000Stefan, 2020-05-06:
Hello terjeio, 
today i did some short probing tests and would like to tell you the results. 
First of all I would like to suggest that you can jogging in the probing tab, would be very helpful in my opinion.
Now to my first probing tests:        ;)
For all attempts no milling program was loaded and the basic settings were unchanged except the cutter diameter was 6mm.

Corner probing without z works without errors. 

Z probing, the process is okay so far, milling cutter moves to the set height (5mm) is unfortunately not displayed in the dro.
Corner probing with z is the same, wrong dro display.
Think there is a mistake in z probing somewhere.
Entry in mdi is only possible once, e.g. G0 X2 Y2, the second entry, e.g. G0 X0 Y0 is no longer accepted.
Think the probing Tab is a very nice helper.  +1:
Will continue the tests and report.

Cheers Stefan

terjeio, 2020-05-08:
Thanks for the feedback.

Keyboard jogging: I have added a button to the probing tab for this, when given focus (or clicked) keyboard jogging is activated:

[image]

Z-probing: seems like this leaves the controller in G91 mode (relative distance), could be why MDI fails. I need to look into this a bit more.

S2000Stefan, 2020-05-09:
Quick update thanks for that.
The jogging active butten is super. Only unfortunately it deactivates with every probing and tab change, also with the tab change in the probing menu, would suggest that the button remains active until you change a main tab, grbl, probing ... and so on.

Tab Edge finder:  
Z Probing o.k. and DRO display now correctly. 
Unfortunately, the edge finder moves twice to the top of the workpiece.
The same for corner probing with z query. 
The corner probing is carried out correctly z is touched, then the edge probe first moves to X0 Y0 Z0 and then to safety level e.g. Z5mm.
DRO display is ok.

MDI input now works too.

Center Probing:
There should be more offset on the inside diameter, otherwise you have to hit the center very precisely, which becomes very difficult with large diameters. I have to enter 41mm with a diameter of 45mm to have a certain security.
The same applies to probing with an outer diameter, more safety distance would be good.
DRO display is ok.

I touch wood with a touch plate so it would be very useful if the touch plate offset were included. Touch plate thickness 5mm safety height 5mm DRO 7mm.
So I sample the z height with a macro as an example:
G91 G38.2 Z-10 F50
G90 G10 L20 P0 Z5
G1 Z5.5 F200
G38.2 Z-2 F10
G90 G10 L20 P0 Z5
G1 Z7 F500

terjeio, 2020-05-09:
Keeping jogging active is something I have to investigate further, e.g. since jogging uses the navigation keys it has to be disabled when any input field is active (has focus/is given focus)... Keyboard jogging is in fact one of the more difficult parts (for me at least) to get working correctly.

* Edge finder: moving twice to probe the top is done on purpose, this due to when the probe hits the workpiece movement is decelerated to a stop. Since setting an offset uses the _current point_ a move back to the probed point, including X and Y, is required before setting the offset. At least this is how I understand the LinuxCNC documentation - e.g. a G10L20 following a successful G38 will set the offset to the decelerated point, not the probed point. Please correct me if I am wrong.
Be aware that the probing Z is done at XY plus half the probe diameter inside of the probed corner/edge - my thinking is that probing exactly at the corner is not a good idea. A checkbox is required for this?

* Center probing: _Max. probing distance_ and _Probing offset (safe distance)_ is used for calculating the travel. It is on my todo list to allow saving settings for these (and other parameters) in user defined profiles.

* Touch plate: I assume this is for the _Tool length_ probing only.  I need to get/make myself a proper probe for testing. Also, I need to understand how to implement this correctly - G10 vs G43 vs g5x offsets... grblHAL has the option to define a tool table with offsets per tool stored in EEPROM that I need to consider as well - I will use this for my lathe and the router ATC I am tinkering with. So the _Tool length_ tab is very much work in progress, e.g. the touch plate offset is currently not used in the code.

I am not a machinist so a bit slow to pick up all the details required for implementing this...

terjeio, 2020-05-10:
FYI: the DRO fields are writable and can be used to set the current position manually (by pressing <Enter> after inputting a value.

S2000Stefan, 2020-05-10:
> Keeping jogging active is something I have to investigate further, e.g. since jogging uses the navigation keys it has to be disabled when any input field is active (has focus/is given focus)... Keyboard jogging is in fact one of the more difficult parts (for me at least) to get working correctly.
 
Yes i never thought about that, right if an input field is active the keyboard jogging must be deactivated. 

> Edge finder: moving twice to probe the top is done on purpose, this due to when the probe hits the workpiece movement is decelerated to a stop. Since setting an offset uses the current point a move [...]
Be aware that the probing Z is done at XY plus half the probe diameter inside of the probed corner/edge - my thinking is that probing exactly at the corner is not a good idea. A checkbox is required for this?

I honestly have to think about that. In my macro I also scan the surface twice but not in rapid motion  ;)
I'll try to find out more about this direction.

> Center probing: Max. probing distance and Probing offset (safe distance) is used for calculating the travel. It is on my todo list to allow saving settings for these (and other parameters) in user defined profiles.

I am aware that you can set the safety distance in the field probing offset, but it seems to me that in center probing the distance is halved, each side 2.5mm instead of 5mm. All related to original settings.

>  the touch plate offset is currently not used in the code.

Which explains a lot, of course.   ;)

> I am not a machinist so a bit slow to pick up all the details required for implementing this...

I think you're doing a very good job.

> FYI: the DRO fields are writable and can be used to set the current position manually (by pressing <Enter> after inputting a value.

Very good function, I haven't noticed it yet, I will definitely use it more often now.

S2000Stefan, 2020-05-10:
Found here a probe-screen for linux cnc written in python but the macros might be helpful for you?
[https://github.com/verser-git/probe_screen_v2](url)

terjeio, 2020-05-10:
> Found here a probe-screen for linux cnc ...

Good find, the linked [blog post](https://vers.by/en/blog/useful-articles/probe-screen) contains a lot of useful information. The related [linuxcnc discussion](https://forum.linuxcnc.org/49-basic-configuration/29187-work-with-probe) is 37\(!\) pages long - a lot to digest there too.

Thanks a lot, very useful - likely _the_ reference we need. And obviously probing is not something completed in a hurry...

jschoch, 2020-05-10:
> I need to get/make myself a proper probe for testing.

what kind of probe do you want?  Maybe I (or we) can make  you one as a gift for all of your fine work here!

terjeio, 2020-05-10:
@jschoch : Thank you for the offer. My current idea is to integrate one into the prototype ATC on my [mini mill](https://cdn.hackaday.io/images/607301527419852157.jpg) so I'll try that first.

I expect to get my [mini lathe](https://hackaday.io/project/165248-mini-lathe-emco-compact-5-cnc-conversion) conversion up and running the coming week, Then I will be able to machine the parts I need. And hopefully I will be able to complete spindle sync support (for G33/G76) in grblHAL too.

I have too many nearly completed projects on my hand...

jschoch, 2020-05-10:
> I have too many nearly completed projects on my hand...

you and me both!

It would be fun to collaborate on a probe so let me know if there is anything I can do to help.

jschoch, 2020-05-10:
> My current idea is to integrate one into the prototype ATC on my mini mill so I'll try that first.

taking this quite a bit off topic but I took a look at your mill and I had some thoughts if you don't mind the feedback.  

I went down the road of cheap spindles and it was a pretty big waste of time.  If your spindle has .1 or .2mm of runout you aren't likely to have much luck milling something small like a  8-Pin TSSOP.  I am now running an NSK NR-40 which literally is a dream.  It has pneumatic tool changing and it has no vibration while cutting.  The only downside is the cost of the collets and getting them with the motors is quite expensive however I was able to pick mine up for $200 on ebay with the collet being $100.  This is well worth it in my opinion.  Plan B would be for you to make a spindle on your lathe.  You can pickup angular contact bearings from aliexpress for quite a reasonable price.  They make a huge difference for milling.  You can also pick up an er "extension" from aliexpress  and hard thread the end (required for adding preload to your bearings).  You can then reuse that spindle to drive an o-ring/pulley combination.  The downside is that you'd be stuck with ER collet changing which gets old very quickly for frequent tool changes.  It looks like you are going to be attempting to capture the collet nut for tool changing, is that what the thingie is center left of the bed?  That seems like a very tough engineering challenge but it has been done.

If you are only doing very light milling that spindle mount may be ok but I doubt it will work for milling aluminum.  I'd suggest you look at nema 23 steppers and running it at 48V (24 at the bare minimum).  STSPIN820s can do $45V if you want to stick with a polalau type stepper driver, also TMC2130 can do 46v.  Both likely require a better rated capacitor than is typically on most boards if run > 24v.    That Z lead screw will also cause you problems with backlash which brings me back to the idea of donating something to you for your efforts.  I got super lucky and picked up 8 small ball screws for $80 shipped on ebay a while ago.  If you want I can send you one of them for your Z axis.   You'd likely have to re-think your Z axis to accommodate it but they make a huge difference in rigidity and accuracy.

terjeio, 2020-05-10:
I do not mind. ;-)

The mini mill is for drilling PCBs, milling the outline and as a test platform for grblHAL - so for light use. I use my homemade laser exposer to image the tracks and solder mask. No problem with exposing for 0.5mm pitch ICs and 0.15mm tracks.

The big brother at our workshop - the first CNC project I did:

[image]

Hiwin rails (new - the real deal), second hand ballscrews from Korea, THK KR33 actuator for Z-axis second hand from Thailand, Nema 23 for X and Y, DQ542MA stepper drivers and a Kress spindle. Heavy duty enough for my general use - mostly milling aluminium. For this I use Mach 3 with a ethernet SmoothStepper controller, no grbl for that (yet). IIRC I run the motors at 36V.

The ATC is a kind of fun project, happy if I get it to work and if not a good challenge. Not shown in the picture is the mechanics for clamping the spindle so I can unscrew/screw the collet nut. I need some small diameter metric screws to finish that, M1.2?

Again, thanks for your offer - but I believe it is a waste of good parts upgrading the mini mill beyond the requirements for the intended use... Oh, and the spindle runout is not too bad.

I could have used a small ball screw for my lathe X-axis, but I already got a second hand NSK from Korea so too late for that.

I appreciate that users of my code takes time to test and give feedback, that is giving back good enough for me.

jschoch, 2020-05-13:
back on topic, while I had good luck this morning and milled out a double sided board without much drama, tonight I tried to probe Z and auto level and several times the G92 tool height setting operation set my Z height to 192 mm.  It was impressive to watch my v cutter plunge through FR4 and into the MDF below without breaking, but I don't think that is good for my machine.  I caught another crash, not sure if it was the tool height or the auto leveling that threw it off.  I think it would be good to have the DRO screen visible while in the probing tab while the kinks are worked out.

terjeio, 2020-05-20:
New [release up](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v0.19-alpha), refactoring inspired by the [LinuxCNC probing plugin](https://vers.by/en/blog/useful-articles/probe-screen). Still work on todo list though.

S2000Stefan, 2020-05-21:
Short tests of the probing tabs have been very successful. I tried the corner probing and diameter probing. So far all tests have been very positive.

Outside diameter probing:
[https://www.youtube.com/watch?v=yVnzTChy0VE](url)

But what I noticed with the outside diameter probing it would be an advantage if the probing would take a little distance after the second exact probing before the z axis moves up. For this you could use e.g. the latch distance.

Corner probing:
[https://www.youtube.com/watch?v=POftttWroPo](url)

Is it really wanted that the z axis after the probing again on Z0 on the corner drives? I do not find it very advantageous.
X axis Y axis on zero and safe z axis height is ok. My opinion again z0 zero should not be?

Further tests will follow.  :)

terjeio, 2020-05-23:
Thanks for the feedback.

> But what I noticed with the outside diameter probing it would be an advantage if the probing would take a little distance after the second exact probing before the z axis moves up. 

I have to check my code again, it should move XY clearance distance.

> Is it really wanted that the z axis after the probing again on Z0 on the corner drives?

Lazy coding by me? By subtracting the deceleration distance it could be avoided, but then I need to get the current point from a status report to be able to set the offset precisely. The easy way out was to move to the probed point using G53 before setting the offset. 

From LinuxCNC documentation (I added the bold):

> G10 L20 is similar to G10 L2 except that instead of setting the offset/entry to the given value, it is set to a calculated value that makes the **current coordinates** become the given value.

and

> G92 makes the **current point** have the coordinates you want (without motion), where the axis words contain the axis numbers you want. All axis words are optional, except that at least one must be used. If an axis word is not used for a given axis, the offset for that axis will be zero.

Note that the probing motion decelerates to a stop after triggering the probe, e.g. probing with a tool is thus not a good idea?

S2000Stefan, 2020-05-23:
> Lazy coding by me? By subtracting the deceleration distance it could be avoided, but then I need to get the current point from a status report to be able to set the offset precisely. The easy way out was to move to the probed point using G53 before setting the offset.

Okay, now I understand your approach. It's legitimate for you to take the best part. Personally, I don't think that the restarting of Z0 is so good. 
    ;)
 
One more small note about G92, please use the command with care. It does not just move one coordinate system. I personally prefer to use G10 P20 P0 XYZA... to zero the axis in the current coordinate system. As can be seen in my macro above.

> e.g. probing with a tool is thus not a good idea?

Probing with a tool is not a good idea, but it's much faster and with wood that I work on very often it doesn't really matter. 
       ;)

terjeio, 2020-08-05:
I have added some new tool change modes to grblHAL that may be relevant to this discussion as some can use probing/tool length offset. Comment at [this issue](https://github.com/terjeio/grblHAL/issues/79) if of interest.

S2000Stefan, 2020-09-18:
Today I had found some time to test the new Edge version (GCode Sender) of the revised Edge Finder External Tab extensively. 
I have to say I like it very much, the process and the settings are great, I couldn't find any mistakes. (This time I fixed the workpiece). 
Thumbs up.
The only thing I found is if you click on the z-probe in the middle of the work piece and then deactivate the z-probe in the box. So if you don't select anything and then press start, the program will still run a z-probe. Wouldn't it be better to give a warning message like at the beginning when you enter the tap again and have nothing selected?
("Select edge or corner to probe ........")
More tool change tests will follow this weekend with the new GrblHal version.

terjeio, 2020-09-18:
Thanks again for testing, very much appreciated. I hope we are getting closer to a usable system now.

> The only thing I found is if you click on the z-probe in the middle of the work piece and then deactivate the z-probe in the box.

This is no longer possible, I now disable the Z-probe checkbox when the x-probe graphics is clicked.

I have also added interlocks for the fields on the left hand side, this means that the only the fields used for the selected probing action will be enabled for input. This will make it clearer which values are actually used.

Later I will add some configuration options for probing in the _Settings: App_ tab such as XY offsets if the probe is not in the center of the spindle.

The [Edge version](http://www.io-engineering.com/downloads/) has been updated with these changes.

S2000Stefan, 2020-09-24:
Today the Edge Finder was once internal and the Edge Finder external under strict observation.  ;)
All with the latest version of the GCode Sender and latest version of grblHal.

The process of **_all_** corner and side scans is excellent. However, there are problems with the z-sample.
If you do a corner scan, whether inside or outside, plus Z-sample and press stop in the GCode Sender during Z-sample, the Z-axis (attention) moves to Z minus with Rabid feedrate on and on.

I also noticed that the entries in XY Clearance are not correct later in the process.  For example if you enter 5mm the sample is only 2mm or if you enter 6mm the sample is only 3mm ...... away from the edge.
All other inputs like offset and speed ..... seem to be ok, at least I did not notice any inconsistencies.

terjeio, 2020-09-25:
New edge version just uploaded.

> However, there are problems with the z-sample.
If you do a corner scan, whether inside or outside, plus Z-sample and press stop in the GCode Sender during Z-sample, the Z-axis (attention) moves to Z minus with Rabid feedrate on and on.

I believe this is fixed now.

> I also noticed that the entries in XY Clearance are not correct later in the process. For example if you enter 5mm the sample is only 2mm or if you enter 6mm the sample is only 3mm ...... away from the edge.

I have added preview as suggested by @jschoch earlier but in a less intrusive way. Can you check the code generated with preview enabled when this happens? Note that the preview lists the moves as relative to where the tool is placed and that the wrap up after XY probing is currently not added to the preview.

Again, I appreciate the time you spend testing!

S2000Stefan, 2020-09-25:
> However, there are problems with the z-sample.
If you do a corner scan, whether inside or outside, plus Z-sample and press stop in the GCode Sender during Z-sample, the Z-axis (attention) moves to Z minus with Rabid feedrate on and on.

>I believe this is fixed now.

I have to report that I get the craziest messages and movements with the new version when I touch the corner and/or plus z-probe. 
Once error26 comes, rehearsing completely but the dro display is completely wrong. The z-probe is executed next to the workpiece and other crazy things?
There is no testing possible too many crazy errors.
[image]

> I also noticed that the entries in XY Clearance are not correct later in the process. For example if you enter 5mm the sample is only 2mm or if you enter 6mm the sample is only 3mm ...... away from the edge.

> I have added preview as suggested by @jschoch earlier but in a less intrusive way. Can you check the code generated with preview enabled when this happens? Note that the preview lists the moves as [...]

As I can see from the preview window you do not take into account the diameter of the probe. Wouldn't it be better to include it in the XY clearance? Because if I have a diameter of 12mm and a clearance of 6mm the probe would not be offset from the touch surface. Hope you understand what I mean. Note the third line in the preview window diameter 1mm and diameter 12mm the line is the same. What i think is wrong.
[image]
[image]

The z-probing is not shown in the preview window even though it is activated and was sometimes executed even though it was not visible in the window?
Sorry for all the bad news.

terjeio, 2020-09-25:
> Sorry for all the bad news.

I am sorry for causing this, a single line of code is creating havoc.

I will also add 0.5 x probe diameter to XY clearance in the next upload, I have to admit I did not consider that myself but it does make sense.

Z probing is a bit tricky to get into the preview window from the real code as bits of that needs probing positions and is waiting for the machine to come to rest at commanded positions during execution. I will add "dummy" code later to show which commands are issued.

BTW, waiting for "come to rest at commanded positions" could be the reason for probing sometimes hanging - currently I wait for a delta of max 3 micrometers, this should be increased, made configurable or handled in some other way if the condition is never met. Perhaps even terminating the probing sequence with an error.

The next version of grblHAL will have an option to enable simple probe protection. By no means perfect but better than nothing. It works by cancelling movement if the probe is triggered when a non-probe move is executed. Cancelling is executed by the sender and may thus be delayed a bit. A faster and safer protection scheme is on my todo list. This will be added to the grblHAL but is harder to get right.

S2000Stefan, 2020-09-26:
I saw this morning that you uploaded a new edge version at 00.10am. Tell me, do you never sleep? ;)
I tested it as well. The 0.5 x probe diameter to XY clearance is already integrated and seems to work. :+1: 

Unfortunately I am still unable to do further tests because I always get a probing faild. If I set the speed very slowly the probing sometimes runs until the latch through but then breaks off.

> BTW, waiting for "come to rest at commanded positions" could be the reason for probing sometimes hanging - currently I wait for a delta of max 3 micrometers, this should be increased, made [...]

This is probably a bit too sharp for my machine because I always get the probing error messages?
But I think it is a very good idea to stop the probing immediately in case of inconsistencies.

> The next version of grblHAL will have an option to enable simple probe protection. By no means perfect but better than nothing. It works by cancelling movement if the probe is triggered when a [...]

This is a big step forward to make the machine really safe, I am excited.

terjeio, 2020-09-26:
> Tell me, do you never sleep? ;)

 I do now and then, but hard to go to sleep when working on a problem or exiting new idea.

> Unfortunately I am still unable to do further tests because I always get a probing faild.

Oops, I left the new probe protection snippet in place, it should only be enabled if enabled in the controller.
I have commented this out for now, new edge uploaded.

I have changed the  "come to rest at commanded positions" handling as well.
The wait is now terminated with success if the position is within two steps of the target.
"Two steps" is a bit fuzzy since there are conversions to and fro steps and millimeters. I may have to increase this a bit or even change the algorithm.

It is terminated with failure:
* if the controller has transitioned from Run to Idle state before reaching the target.
* if no state transition has been detected, the the state is Idle and the position delta is less than 0.01 mm.

It could still hang but I believe that is unlikely. Testing, testing...

S2000Stefan, 2020-09-26:
I have just come off the machine and can only report positive news. The probing works again as it should and half diameter is implemented, as written above. :+1: 

I also saw that you set a logical relation between the probing distance and XY Clearance.
The probing distance must be greater than the XY Clearance.
Unfortunately you forgot this with the Latch Distance? If the Latch Distance is greater than the Probing Distance the Probing will still be executed and logically end with an error.
I would also like to ask the question: if both logical values are the same, will the Probing be started, wouldn't it be better to point out a logical higher value?
At least half a millimetre difference or more? Nothing can happen, but the user might wonder why his probing always fails.

Hopefully tomorrow I will find the time to continue testing, and I will also try to get Gremlin out again.  ;)

Translated with www.DeepL.com/Translator (free version)

terjeio, 2020-09-26:
> I have just come off the machine and can only report positive news.

Great, hopefully we are getting closer to a fully functional version now.

> I also saw that you set a logical relation between the probing distance and XY Clearance.
The probing distance must be greater than the XY Clearance.
Unfortunately you forgot this with the Latch Distance?

Maybe I forgot, maybe I did not think about it... Anyway added now but with no arbritary distance added in. Impossible to make probing bomb proof for inexperienced users, some understanding is required....
Since the fields had tooltips assigned to them the error message(s) did not show up on hover - fixed now.

New edge build uploaded.

S2000Stefan, 2020-09-27:
As announced I tested the Edge finder, external/ internal and the centre finder extensively today.
What can I say, I could not find any inconsistencies or errors. Everything went perfectly. :+1: 

But I have to think about this message in the Edge finder external, do you think the statement is not quite correct? 

[image]

Couldn't have said it better: The offset minus half the diameter of the Probe tip/tool is smaller than the XY clearance, use Offset and.....?

Unfortunately I also have to report that the G-Code Sender has crashed once. The controller was still there, but after restarting the G-Code Sender it could be addressed normally. The only thing that was strange in the DRO display of the 4th axis was that there were no more numbers to be seen. Here is the event log of the error, maybe you can read something out. But it only happened once during a nearly two hours long probing test. The following Probing went without any problems and went through correctly.
[prodokoll.rar.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5288384/prodokoll.rar.txt)

S2000Stefan, 2020-10-03:
I also took a look at the height map today. I know it from Open CNC Pilot.
I would suggest to shorten the sampling time, of course at the expense of accuracy, to insert an option to select to do only one probing, of course only with the latch feedrate?

jschoch, 2020-10-03:
if you set the latch distance to 0 it will only probe once.  There is no sampling, the signal is on or off from the controller's perspective.

terjeio, 2020-10-03:
> Unfortunately I also have to report that the G-Code Sender has crashed once. 

I have looked into this now, the crash happened in code that runs often when probing so a real Gremlin... Currently I do not know how to track this down or fix it.

S2000Stefan, 2020-10-03:
> if you set the latch distance to 0 it will only probe once. There is no sampling, the signal is on or off from the controller's perspective.

I should have thought of that myself. :-) Thanks jschoch

> I have looked into this now, the crash happened in code that runs often when probing so a real Gremlin... Currently I do not know how to track this down or fix it.

And I thought you had found and changed something, because I haven't had a single crash of the programme since.
But I suspect it has something to do with my probing connection, I'm going to debounce the connection as I did with my safty door connection.

closed 2021-04-29 by terjeio