terjeio/ioSender #34: probe feedback center finding rapids

jschoch, 2020-05-10:
The rapids on the center finding are a bit scary, maybe better to tone them down a bit with the defaults.  My first test I had something that could move but I could have easily crashed it.  It isn't clear that it rapids to the "safe distance".  I like that it is quick but it may be unexpected for some.

terjeio, 2020-05-10:
You want a new input field for probing "rapids"? For both the Edge and Center finders?

S2000Stefan, 2020-05-10:
I am not jschoch but I would like that also very much.

jschoch, 2020-05-10:
for hole finding: The way I would expect it to work is that by default it feeds at the probe feed rate from the starting point of center finding and would not rapid at all.  For the default behavior I would remove the hole diameter setting and show the max probing diameter based on the safe distance settings.  So by default with 5mm safe distance your non-editable "max diameter" field would show 10mm.  It may be useful to have the center finder routine use it's own values for safe distance and an optional "rapid" setting.

Probing with rapids is nice and it would be good to have as an advanced feature particularly for large holes but I think you'd want to do it in a wizard type setup so it is very clear that getting the diameter wrong might crash.   When I tested it first I was not expecting the rapids and I just guessed the hole diameter.  I guessed 30 but it was 23 and it crashed right away (luckily it was not setup rigid).

I don't believe you need the diameter of the hole to calculate the center.  http://mathforum.org/library/drmath/view/55233.html

terjeio, 2020-05-20:
Perhaps the [latest release](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v0.19-alpha) is better? Default workpiece diameter is now 0 and some sanity checks on parameters added.