terjeio/ioSender #36: height map probe feedback

jschoch, 2020-05-10:
I really struggled with the height map probing.  My part is setup to flip on the X axis and X is setup on the centerline.  The load from file boundaries does not seem to work correctly in this case.  

The "top right" and "lower left" are also very confusing.  I think it would be clearer to label them so it is clear which coordinates are X +/- and Y +/-.  

It is also unclear what height and width are so clarifying with which is X and which is Y would be helpful.

I'm not sure how much work it is but it would be nice to see the heightmap over the gcode viewer display.  I'm guessing you have planned to do this.

The tool offset setting does not seem to update.  I was not able to get the Z height accurate and had to reset it after the probing routine.  The height map did seem accurate though.

Finally, when doing PCB's it would be nice if the height map would stay in memory and if you load another file it would ask you if you wanted to apply the heightmap to the new gcode.  For a double sided PCB milling where I'm doing 2 parts I would first load the drill file and do the height map probing.  Then I'd load the traces cutout and apply the heightmap.  Next I'd flip the part and re-run the traces cutout with the heightmap applied a 3rd time.  Finally I'd run the board cutouts with the heightmap applied a 4th time.  Since the spoil board degrades over time I'd not want to save it or have the potential of getting confused with an old one.   

I'm guessing there are use cases where you want to keep it and have it auto-load but I don't see myself needing that.  BCNC's dialogs allow for this but are very murkey and unclear in any use case.

I'm planning on giving it another go tomorrow but I had a few to many crashes and almost crashes for one day!

terjeio, 2020-05-10:
> My part is setup to flip on the X axis and X is setup on the centerline.

Can you upload a test file for me?

I will change labelling of the input fields,  but not sure what is the best labels to use. Perhaps Xmin, Xmax etc is best? Using + and - suffixes looks a bit odd to me.

> ... heightmap over the gcode viewer ...

On todo list, but I need to move the needed data structure over to the main code first. This as the probing code is in a separate project along with the data structures it uses.

> The tool offset setting does not seem to update.

It does not, should it update to the first probed position? Or should one probe this with the _Tool length_ tab?

> ... it would be nice if the height map would stay in memory ...

It does for me, how did you manage to unload it?

> ... if you load another file it would ask you if you wanted to apply the heightmap to the new gcode.

Will look into that, perhaps I should add a function to unload the heightmap as well? For now you have to go to the probing tab and _Apply_ the map again. The window title bar will show that the map has been applied by prefixing the file name with _Heightmap applied:_

> I'm guessing there are use cases where you want to keep it and have it auto-load but I don't see myself needing that.

The map does not auto load, you have to use the _Load_ button or drop a map file on the 3D viewer. A use case for reloading a map could be when milling parts that has exactly the same shape.

jschoch, 2020-05-10:
Gcode attached

>> The tool offset setting does not seem to update.

>It does not, should it update to the first probed position? Or should one probe this with the Tool length tab?

I was assuming it was applied to height map since my z hieght was off after probing.  In bCNC you need to first probe Z, then run the autolevel routine, then "apply".  Is this the same for your sender and does it use the tool offset value?

>>... it would be nice if the height map would stay in memory ...

>It does for me, how did you manage to unload it?

>The map does not auto load, you have to use the Load button or drop a map file on the 3D viewer. A use case for reloading a map could be when milling parts that has exactly the same shape.

So it sounds like the procedure is:

1. load some gcode
2.  scan and create the heightmap
3. click apply after 
4. click save
5. run job
6. load new gcode
7. use load button or drag map file
8. run job
9.  6..9 as needed 

if your tool height changes, what do you need to do?

[Combo_traces_cnc - Copy.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/4605712/Combo_traces_cnc.-.Copy.txt)

jschoch, 2020-05-10:
I think I figured out my issue.  the probe must start in the "lower left".  I expected it to start at x/y 0,0 then move to 0,-30 and start probing and probe to 0,30.  instead it starts at 0,0 and probes to 0,60.

terjeio, 2020-05-10:
Thanks for the file, what is missing from the probing algorithm is a rapid move to the "Lower left" corner. I will add that. Oh, I see you just found out yourself.

The map has it origin at the "Lower left" corner and the Z-offsets are relative to that. So setting zero (or offset) at that point before starting the job is required. Since map files can be reloaded perhaps a workflow like this is appropriate:

1. Load some gcode.
2. Scan and create the heightmap (possibly from program limits), or load a previously saved one.
3. Optional: save the newly scanned map if it is to be reused.
3. Apply the heightmap.
4. Probe the "Lower left" corner to establish the controlled point at the workpiece surface.
5. Run job.
6a. Load new gcode file, goto step 3. Or
6b. Load new workpiece, goto step 4. Or
6c. Done

jschoch, 2020-05-10:
I re-ran the job this morning and was able to re-apply the heightmap!

One thing I was a bit worried about was re-running the heightmap after flipping the board.  The board may not be consistent thickness but it now has tons of holes in it. I wonder if it is possible to ensure the height map probe scan avoids holes.  This may be better suited for a post processor.  I don't think flatcam can do this though.

terjeio, 2020-05-10:
Milling the back side first without drilling is easier? Note that is quite easy to flip a program with grblHAL as it supports G51 scaling. I flip my drill files with G51X-1 as I mostly make single sided PCBs and drill from the back side. Do not forget to issue a G50 afterwards to cancel scaling... I have not tested this with a heightmap though - perhaps the heightmap needs to be applied first and then the file edited to add G51.

Also, if I have done my homework correctly, the sender should be able to import Excellon drill files (.drl) and translate them to gcode. However this is not going to work with other firmware than grblHAL since the translation is to canned cycle G81.

I assumed the need for a heightmap for milling PCBs was more due to the PCBs beeing a bit warped than thickness variation. If the former I would rescan the map after flipping.

jschoch, 2020-05-10:
this may help, it is a bit long winded though

https://youtu.be/FjvuWzZFyHI

The idea is to make 2 boards, you put the front on the top, back on the bottom.  You flip once and presto you have 2 finished double sided boards!

jschoch, 2020-05-12:
I worked on the grblHAL board this morning, another end mill lost!

I keep assuming that the Z height is set during height map probing.  BCNC also requires that you set the Z height first, but I wonder if that is the right default.  It is also unclear where you should measure the Z height.  For instance if your origin is centerline X and your board is 80mm high in Y do you measure Z from 0,0 or drop down to 0,-40 where you start your heightmap scan?

I'm guessing there is a good reason Z isn't set from the heightmap scan but I don't know what that reason is.  If it is possible it would be nice to be able to tick a box to have Z height set on the first scanning point.  this way you can be sure +/- Z is calculated correctly with the correct reference.

Finally, though I though it was working on the first try, this time I didn't notice any movement in the Z motor after applying the height map which would seem to indicate it was not applied.  I had some purple on my heatmap which I would assume needed the Z motor to move.

terjeio, 2020-05-12:
Sorry to hear that another mill is lost - you have to be super careful with probing and running jobs after.

As for setting the tool offset at the map origin is surely someting that can be added, as you say it should be a checkbox for that. I do not think it is wise to always set it from the heightmap since that can be reloaded. And also since probing with a tool is not good practice? (I do it myself all the time).

Perhaps it is a good idea to add some guide text where the warning is?

It should be fairly easy to see if a heighmap has been applied to the gcode, there should be lots of additional lines and/or Z movements.

terjeio, 2020-05-20:
Checkbox for tool zeroing added in [latest release.](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v0.19-alpha)

jschoch, 2020-05-22:
I'm not really liking the G92.  If you are running a job (on non-grblHAL) and say you need to stop it and adjust your Z height it resets and looses your Z0 position.  I believe this caused at least one of my crashes.  

The heightmap seems to be working well.  Below is a comparison, left is with the height map, right is without.  The holes on the left are lined up better due to running traces and drill/cutout in a single setup though they need a bit more deburring.  The offset holes on the right are due me having to re-home and I must have lost some position on the right.  You can also see that the copper is not isolated in many places thought the lines look very sharp.  

I ran the cutter over the right board 4 times due to me trying to dial in the z height manually.  The machine repeats very precicely but it does lose something in homing and i'm dealing with noise from the stepper drivers.  I tried to get the machine running with a grblHAL board but there is some issue causing it to reset over and over if it is installed in my breakout board.  It may be related to the noise issue.

[image]

terjeio, 2020-05-23:
> I'm not really liking the G92.

I have forgotten about that, grblHAL [keeps more data over a warm start](https://github.com/terjeio/grblHAL/blob/73604aa13c0b748a760d6f2e49c77830f38ce4f3/grbl/gcode.c#L118-L132) (reset) such as G92 offsets, and should perhaps [keep the offsets over a cold start](https://github.com/terjeio/grblHAL/blob/73604aa13c0b748a760d6f2e49c77830f38ce4f3/grbl/gcode.c#L2263-L2267) too (I am a bit reluctant to do that though):

LinuxCNC keeps the offsets even over a cold start:

> LinuxCNC stores the G92 offsets and reuses them on the next run of a program. To prevent this, one can program a G92.1 (to
erase them), or program a G92.2 (to remove them - they are still stored).

and IMO even NIST specifies so:

> You can set axis offsets in one program and use the same offsets in another program. Program
G92 in the first program. This will set parameters 5211 to 5216. Do not use G92.1 in the
remainder of the first program. The parameter values will be saved when the first program exits
and restored when the second one starts up. Use G92.3 near the beginning of the second program.
That will restore the offsets saved in the first program. If other programs are to run between the
the program that sets the offsets and the one that restores them, make a copy of the parameter file
written by the first program and use it as the parameter file for the second program.

I do not know if Grbl is coded to intentionally reset the G92 offsets or not, but I remember now beeing annoyed enough about losing the offsets to change the behaviour in grblHAL. So perhaps this a Grbl bug?

Anyway - I will add a drop down for setting a coordinate system and make this default for non grblHAL firmware. And perhaps reset the offset when [COMPATIBILITY_LEVEL is set > 0](https://github.com/terjeio/grblHAL/wiki/Changes-from-grbl-1.1#workaround).

> Below is a comparison, left is with the height map ...

The bottom PCB in the image is the one milled with heightmap applied?

jschoch, 2020-05-23:
g92:  i believe if the motor enable flips you have the potential to lose some position, it seems intentional for vanila grbl and I'd expect that if you keep making the sender awesome you will have more folks using it but maybe only up until one of their assumptions about how it works fails.  I'm highly excited to see improvements on grbl's highly constrained design now that it doesn't have to cram into such a limited architecture however folks have been using it for years will likely be surprised unless you have some great documentation of the differences.  Some of these changes are needed  and it is sad that there is no way to try to have coordination.  At some point maybe it makes sense to call it something other than grbl though to ensure folks are checking their assumptions.  I don't think you are there yet.

The biggest problem for the ephemeral offsets is that you may have cut away your datum...

> The bottom PCB in the image is the one milled with heightmap applied?

the quarter should be a the top (is it oriented differently in your browser?), the bottom left is the heightmap milled pcb, the bottom right is the one without the heightmap.  The traces on the right are cut but the 2-8 is only half cut through the copper.

They were cut from the same 70x100 pcb blank on the same setup (tape/ca glue) on MDF.

terjeio, 2020-05-23:
> g92: i believe if the motor enable flips you have the potential to lose some position, it seems intentional for vanila grbl ...

Why is vanilla grbl keeping the position over a soft reset then? It should definitely be reset too if loosing position due to flipping motor enable signals is a problem. And if resetting G92 offsets is required then why not the rest (g54+) as well? And a soft reset should not touch the motor enable signals either. So IMO not compliant with neither NIST or LinuxCNC specs.

> ... however folks have been using it for years will likely be surprised ...

Yes, this is a problem - I should have kept better track of the changes breaking backward compatibility. I will try to rectify that. But I want to move forward with grblHAL too, disabling my sender for vanilla grbl may thus be an option...

> ... there is no way to try to have coordination ...

There isn't? I would prefer a third party to take the initiative.

> ... is it oriented differently in your browser? ...

The quarter is on the left side in my browser (Firefox).

jschoch, 2020-05-23:
Honestlly I had never used G92 before using this sender with GRBL or Centroid's CNC 12 software.  Digging into g92 it updates all coordinates g54-g59 which is a bit confusing.  

[this](https://github.com/grbl/grbl/issues/1109) is a decent discussion on pros/cons of G92 but what stuck out for me was 109JB's description 

"G92 is not a work coordinate and is misused very very often. G92 is global coordinate system offset."

terjeio, 2020-05-23:
Thanks for the link to that discussion. Seems I should get rid of the G92 option, perhaps add an option to use the tool length offset(s) instead. Which does not survive a soft reset in vanilla grbl either (but does in grblHAL).

I am not a machinist so I do not know/understand all the finer details, again - thanks for providing input.

jschoch, 2020-05-23:
I think keeping the persistence is ok as long as it is well documented.  It also doesn't seem like a good idea to use g92 for offsets since it is a global offset that overrides all the WCOs and it seems like it should be used in a very intentional way.  

It may be good to start a matrix of things like this similar to your driver matrix.

terjeio, 2020-10-07:
I have added a [compatibility matrix](https://github.com/terjeio/grblHAL/wiki/Compatibility-level) to the grblHAL Wiki so I am closing this issue.

closed 2020-10-07 by terjeio