terjeio/ioSender #38: vanilla grbl jogging error:15 soft limits exceed

jschoch, 2020-05-13:
Not critical but wanted to report this.

My machine has the Y axis homing inverted and I run in $10=3 so I can use bCNC

When I enable soft limits I cannot jog, it throws error:15.  I can move around in g53 and g54 with g90 g91 without having any errors thrown.

I can run bCNC and jog around.  I can see that it is issuing $J=G91

with grbl-gcode-sender I can't see the command that is issued but here is the console output when trying to jog

jog error:

`<Idle|MPos:-5.000,-130.000,-5.000|Bf:15,512|FS:0,0>
error:15
<Idle|MPos:-5.000,-130.000,-5.000|Bf:15,512|FS:0,0>
<Idle|MPos:-5.000,-130.000,-5.000|Bf:15,512|FS:0,0>
<Idle|MPos:-5.000,-130.000,-5.000|Bf:15,512|FS:0,0>
<Idle|MPos:-5.000,-130.000,-5.000|Bf:15,512|FS:0,0|WCO:-95.180,-58.882,-158.402>
`

if I issue $J=G91 y1 F200 it works

`
<Idle|MPos:-5.000,-129.000,-5.000|Bf:15,512|FS:0,0>
ok
<Jog|MPos:-5.000,-128.817,-5.000|Bf:14,512|FS:200,0>
<Jog|MPos:-5.000,-128.151,-5.000|Bf:14,512|FS:110,0>
<Idle|MPos:-5.000,-128.000,-5.000|Bf:15,512|FS:0,0>
`

terjeio, 2020-05-13:
Can you try with this setting?

[image]

jschoch, 2020-05-13:
it is plain old grbl so I don't see that setting.

terjeio, 2020-05-13:
Ok, then either disable keyboard jogging (jog cancel is not reliable in plain old grbl) or reduce jogging distances in _Settings: App_. I'll try to limit jog moves in the sender later.

jschoch, 2020-05-19:
reducing the distance works, thanks.

closed 2020-05-19 by jschoch