terjeio/ioSender #39: "Allow manual" and jogging problems ($22) Alpha18

S2000Stefan, 2020-05-17:
I use a four-axis machine.
If I activate "allow manual" in Alpha18 ($22=55) jogging is possible normally. But if I set the A axis back to zero with $HA (manual homing)  and then jogging, no matter if X, Y or Z axis is running, the A axis will also run simultaneously. Whether keyboard jogging or using the tap.
If "Allow manual" is disabled ( $22=23) the jogging works perfectly.

terjeio, 2020-05-18:
That is really strange and I am not able to reproduce. Are you sure the sender is the culprit? Code for keyboard jogging and flyout jogging lives in two separate classes - and the keyboard class is not issuing jog commands for axes other than X, Y and Z.

Can you try with sending a jog command, e.g. `$J=G91X0.1F100`, from a terminal such as PuTTY? Does the same happen then?

Are the simultaneous movements always of the same distance?

S2000Stefan, 2020-05-18:
Because my english is not so good I made a small video, hopefully you can see something is very blurred. GrblHal restarted and the machine homed.
[https://www.youtube.com/watch?v=CrFvFL-V5xc](url)
Hope that's allowed here.

> Are you sure the sender is the culprit? 

With the alpha17 there are no problems - there is also no possibility to click on "Allow manual".  ;)

> Can you try with sending a jog command, e.g. $J=G91X0.1F100, from a terminal such as PuTTY? Does the same happen then?

Yes the same happens with putty, the A achses moves, too.

> Are the simultaneous movements always of the same distance?
Answers the video very nicely.   ;)

Hope this helps a bit

terjeio, 2020-05-18:
Thanks for the video - the last few seconds was helpful. It seems to be a combination of settings that trigger this. Does the problem disappear if you disable _Limit jog commands_ -> `$40=0`?

S2000Stefan, 2020-05-19:
If i set the Limit jog commands to $40=0 the problem is gone.
But when i want to jog with the keyboard i have to open the jogging tab. Otherwise jogging with the keyboard is no longer possible.
If i then only want to jog with the keyboard the following error message appears:
"Travel exceeded: Jog target exceeds machine travel. Jog command has been ignored."
And as soon as i move the A-axis with a e.g. G0 A200 command the error message also appears with the jogging tab folded out when jogging with the keyboard.
And jogging with the keyboard is then no longer possible.

terjeio, 2020-05-19:
> If i set the Limit jog commands to $40=0 the problem is gone.

Good, this is what I need to know. I have [committed](https://github.com/terjeio/grblHAL/commit/7e4ee4ed8f9afd1815c0f3645acb23c65245d9e6) what I believe is a fix for grblHAL. Pulloff distance was not skipped for manually homed axes, causing the movement. It is not always easy to foresee the side effects of even small and localized changes - as I thought manual homing was...

Please report back if the commit fixes the issue completely (or not).

S2000Stefan, 2020-05-19:
> Please report back if the commit fixes the issue completely (or not).

After intensive testing i can report that the issue is fixed. It was only a test on the desktop pc but thinking about the real machine it will work too. Tomorrow i will test it on the real milling machine as well. 

Thanks for the quick fix of the issue.

S2000Stefan, 2020-05-20:
the real tests on the milling machine have confirmed the simulations, so I close the issue.    :+1:

closed 2020-05-20 by S2000Stefan