terjeio/ioSender #41: end user probing testing

jschoch, 2020-05-24:
here is a long winded video of me bumbling through some testing.  I didn't test edge finding prior to today.  Corner finding works well, as does center finding.  I'm guessing you didn't put much time into edge finding and there are some easy fixes.

https://youtu.be/NthS9aIBZzg

It seems that for edge probing it uses some of the corner parameters in unclear ways, for instance the x/y offset seems to control the pull off distance.  It is also unclear if the probing distance in the 2nd from the top box controls the probe distance or if the offset in the bottom box controls the distance for edge probing.

[Here](https://youtu.be/NthS9aIBZzg?t=273) is an example where g92 was set somehow but not by the settings I had configure and I had a "crash"

[Here](https://youtu.be/NthS9aIBZzg?t=358) g92 gets set again but not based on my input

[Here](https://youtu.be/NthS9aIBZzg?t=594) I forgot to reset the diameter since the settings need a few click to persist and get lost when you switch tabs (or maybe I restarted?)

[Here](https://youtu.be/NthS9aIBZzg?t=634) I am not sure what coordinates are being used and the dro values are not helping.  Maybe a feature request: can you track the coordinate system and other things like absolute vs relative mode and have them display in all tabs?  I would also like to see the DRO in the probing tab, if the DRO could display the current WCS that would be great!

[Here](https://youtu.be/NthS9aIBZzg?t=1218) is an example of the GUI crashing.  It did this many times in a row while I was attempting to record this and the crashing is what motivated me to record it in the first place.  Once I finally got the recording setup decent it stopped crashing.  The audio is very bad because I had the wrong mic setup.

terjeio, 2020-05-25:
I have to look at your video again, not sure I picked up everything...

A few comments:

G92 option will be axed. Not sure about (how to) add(ing) tool length offset option for _Tool length_ and _Height map_ tabs, at least for vanilla grbl as it does not survive a soft reset (from checking the code, I may be wrong - will test later).

I assume you are testing in a single axis test jig? If so then IMO the only axis that can be tested in a meaningful way is Z. Ref. "Move the probe _to above the position indicated by green dot_ before start." in the _Tool length_ tab (I have added the italics).

The _Edge finder_ graphics needs to be updated, IMO not good.

I wrote "Inspired by probing [plugin for LinuxCNC](https://vers.by/en/blog/useful-articles/probe-screen) (IMO a good reference)." - in the release notes did you check out that link? It may explain some of your issues.

Profile pop-up menu: _Add_ - add profile to drop down and saves the data to to a file, _Update_ - updates the file. No need to update unless you want to save your changes. Perhaps a tooltip or better wording of the menu options should be used. Note that a profile does not contain data from the probing tabs. Maybe I will add some later.

Crashing - I have had a couple of crashes recently too, always when interacting with the program and I have not been able to replicate. Hopefully I will be able to track this down.

Probing puts the machine temporarily in G90 mode (relative distance), this is a problem if probing fails the approach as grbl then enter alarm mode. This means the G90/G91 mode cannot easily be reverted to its original state when exiting the _Probing_ tab. I will have to find a way around that.

A tip, to see the current parser state in the sender check the _Parser state_ option for $10 (available for grblHAL only).

Note the _Center finder_ does not pull off (#32) before retracting in outside mode, I will fix that. 

I'll dig into this more later, I have some paid business to take care of first.

jschoch, 2020-05-25:
>G92 option will be axed. Not sure about (how to) add(ing) tool length offset option for Tool length and Height map tabs, at least for vanilla grbl as it does not survive a soft reset (from checking the code, I may be wrong - will test later).

I am not against G92 but I don't think it should be the default and it shouldn't sneak in.  Somehow I think your probe page states are getting reset without your view updating.  I have no other explaination for getting a g92 when G56 was ticked.  I've reviewed the recording and it snuck in but not because I had the wrong checkbox active.  This may be a more serious problem with states in your UI.

G92 should be an options since some folks will not be using end stops.  

> I wrote "Inspired by probing plugin for LinuxCNC (IMO a good reference)." - in the release notes did you check out that link? It may explain some of your issues.

The documentation is very dense but there is a ton of room in the probe screen so why not use it!!
I will take a stab at mocking something up.  

I'm also wondering if the 3d viewer can be used to dynamically display the current parameters in a 3d model.  For instance the detail on the vers documentation has a great illustration of the edge moves and just seeing it tells you which param you need to pay attention to.

> I assume you are testing in a single axis test jig? If so then IMO the only axis that can be tested in a meaningful way is Z. Ref. "Move the probe to above the position indicated by green dot before start." in the Tool length tab (I have added the italics).

i don't understand the quoted text, where is that from?  Seems like edge finder tab is ok to use for any axis.  I was testing Y.  

> Profile pop-up menu: Add - add profile to drop down and saves the data to to a file, Update - updates the file. No need to update unless you want to save your changes. Perhaps a tooltip or better [...]

Add is greyed out, only update is available unless you type in a new name (i just discovered this).  maybe more obvious is you enable the Add button which creates a modal input prompt for the new name? Another fairly typical way to do this is to put a <new> item in the dropdown list under the existing list and for a new install like [ <Default> ,[user's profiles], <new> ].  Finally user provided names really shouldn't be your lookup key (in my opinion) so maybe just let the user click add and create another default labeled profile.  This feature really seems to have potential though!

>Probing puts the machine temporarily in G90 mode (relative distance), this is a problem if probing fails the approach as grbl then enter alarm mode. This means the G90/G91 mode cannot easily be reverted to its original state when exiting the Probing tab. I will have to find a way around that.

G90 isn't relative so I don't understand this.  I see a G91 down below is that what you mean?  I think you can code around that.

here's what I see when the edge finder runs and fails due to the motors not powered.

`
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
[GC:G38.3 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F50 S0.]
[PRB:0.000,10.800,-3.000:0]
[GC:G38.3 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F50 S0.]
`

I see some things in there i'm not sure about, like:

G49.  I don't think you want to cancel your tool offset when edge or corner finding.

G98 and G50: is this for lathe mode?  Maybe good to omit machine type gcodes when not being used.

>I'll dig into this more later, I have some paid business to take care of first.

This is getting long, no rush on this.  The main thing i'm interested in is a 2d 3d model showing the moves in scale so I'll poke around with that and see if I can come up with something helpful.

terjeio, 2020-05-26:
> G92 should be an options since some folks will not be using end stops.

Ok, I'll leave it in then, but not as the default mode.

> there is a ton of room in the probe screen so why not use it!!

Yes, but it requires work - IMO it is better to get something out for discussion that trying to make it perfect on the first go, and then perhaps have to rip everything apart again. Eg. on my todo list is adding internal corner probing to the edge finder.

> ... wondering if the 3d viewer can be used ...

It may be easier to use static bitmaps, but of course not possible to rotate/scale as a 3D rendering.

> i don't understand the quoted text, where is that from?

[image]

The information I have added under the probing tabs could be better/more precise, english is not my first language and I am not very familiar with the nomenclature used by machinists either...
The same applies to field labels and tooltips too.

> G90 isn't relative so I don't understand this.

My bad, sorry for the confusion - probing may leave the controller in G91 (incremental distance) if probing results in an alarm. I save the current G90/G91 state when entering the probing tab and try to restore it when exiting, this may fail if an alarm has been triggered. I am not sure I like the soft reset behaviour of vanilla grbl - something I need to explore further before deciding how to handle this. 

> G49. I don't think you want to cancel your tool offset when edge or corner finding.

It is the current parser state, indicating no tool offset is active.

> G98 and G50: is this for lathe mode?

G98 is retract mode for canned cycles. G50 is scaling off, G51 is scaling on, this is a Mach3 extension I find useful when drilling and milling PCBs. Perhaps I should add a `#define` option for that?

>  Maybe good to omit machine type gcodes when not being used.

Yes, this is how I do it - eg. G7/G8 is not shown unless lathe mode is active.

jschoch, 2020-05-27:
I was poking around the rendering code last night to try to use the gcode parser to show the probing moves.  How would I get the G38 to render?  I couldn't quite figure out how start and end are calculated.

terjeio, 2020-05-27:
>  How would I get the G38 to render?

Sorry, I am not sure I understand. Do you mean you subscribe to position updates and that fails? I switch off polling for the real time status when probing, could that be the reason?

https://github.com/terjeio/Grbl-GCode-Sender/blob/2bf3f1600e5cc14f798fd772384e93f26956c312/CNC%20Controls%20Probing/CNC%20Controls%20Probing/Program.cs#L83-L89

terjeio, 2020-05-27:
I wrote: "switch off polling for the real time status when probing"

It is in fact active during polling, reactivated later in the init method - a bug?

The  DRO I added to the probing tab gets updated during probing:
[image]

jschoch, 2020-05-27:
I wanted to render the gcode for the probe moves in the 3d viewer.  G38 does not render, and it seems to need a start and end for the 3dviewer to show it but I couldn't figure out how to get that set correctly.  

The idea is to setup a new instance of 3dviewer and put it to the right of the probe config screen.  Then add a preview button which would generate the gcode and send it to the probing 3dviewer so you can see what moves will be made.  If it works well I could add some labels to show the steps (like the static vers probe docs).

terjeio, 2020-05-27:
Ok, I think I get it now. Generation of rendering moves is done in [GCodeEmulator.cs](https://github.com/terjeio/Grbl-GCode-Sender/blob/master/CNC%20Core/CNC%20Core/GCodeEmulator.cs), by the Execute() method, moves for G38 has not been added there.

jschoch, 2020-05-27:
yes, i tried to add it but i wasn't able to get the start and end set correctly.

terjeio, 2020-05-27:
Try:

```
                    case Commands.G38_2:
                    case Commands.G38_3:
                    case Commands.G38_4:
                    case Commands.G38_5:
                        {
                            var motion = token as GCLinearMotion;
                            setEndP(motion.Values, motion.AxisFlags);
                            action.Token = new GCLinearMotion(Commands.G1, token.LineNumber, machinePos.Array, motion.AxisFlags);
                        }
                        break;
```

jschoch, 2020-05-28:
this worked.  I am poking at adding labels.  One issue is that the "!" commands are not parsed and the point at which it moves after the probe will not be a single line.  I may try to show a triangle of moves possible to represent the range of motion which might be made based on the current settings.

this for example where i've moved things to G90 so I can know where to pull the probe back to after the G38 is mocked in:
`
G90G0Y5
G90G0X1
G90G0Z-3
G91
G38.3F100X-10
G90G1X2
G91
G38.3F25X-10
G90G0X2
G90G0Z3
G90G0X-1Y-1
G90G0Z-3
G91
G38.3F100Y10
G90G1Y-2
G91
G38.3F25Y10
G90G0Y-1

`
[image]

This example is what is spit out from the corner probing:

`
G91F25
G0Y5
G0X1
G0Z-3
G38.3F100X-10
!G0X2
G38.3F25X-10
G0X1
G0Z3
G0X-1
G0X-1Y-1
G0Y-1
G0Z-3
G38.3F100Y10
!G0Y-2
G38.3F25Y10
G0Y-1
G0Z3
G0Y1
`

[image]

Also, the renderer does not seem to reload correctly, you have to change tabs.

Do you think I should try to re-use the existing rendering gcode or make a new class?

Also Machine.StartPosition seems to be the machine home position, is that correct?

Does the animation work?  This would make it very slick.

Finally, there is an example in the helix toolkit demo app that allows you to see X,Y,Z mouse coords and it does fact detection to show where the mouse is on the model.  This would be really handy.  I can take a pass at integrating that if you are ok with it.

terjeio, 2020-05-28:
> One issue is that the "!" commands are not parsed

You mean the ! prefix on G38 I use to skip the previous probed point if latch distance is > 0?

> Also, the renderer does not seem to reload correctly, you have to change tabs.

A minor bug - I do not want the 3D tab to render before it is visible. Fixed.

> Do you think I should try to re-use the existing rendering gcode or make a new class?

I do not know. Start with a new or subclass if possible?

> Also Machine.StartPosition seems to be the machine home position, is that correct?

It is the current controlled point (tool tip). It is where the green lines intersect.

> Does the animation work? This would make it very slick.

Yes, but not in the main 3D View as it is not possible to open that while a program is running.

> see X,Y,Z mouse coords

Could be cool, but I guess that would require correct coordinate system handling in the emulator/renderer. Coordinate system handling and improving grid rendering is on my todo list, but currently way down.

einencool, 2020-06-28:
Hello,
I‘ve got two questions about Probing and some ideas.
I‘m thinking about to get a 3D-Finder but, I have a ER20 Collet System, so I didn‘t have a fix length on the machine.
Is it possible to measure the workpiece in X and Y and make a Z-Probe with the 3D-Finder and then measure the length of the 3D-Finder with a Z-touch probe which is mounted to the Base?
Then make a M6 toolchange and measure the Length of the Endmill with the Touch Probe?
Is this possible? 
Also I think about mounting the 3D-Finder outside the Spindle so I would eventually need a offset for this, would this also be possible?

Another question while I was looking at e.g. Estlcam that you can make a edge finding and measure the angle of the workpiece, so that the G-Code will be „rotated“ to match the workpiece. This would be much easier when you could mount the Workpiece on the surface without the high accuracy you would normally need...

Do you see any chances for these points ?

Thank you in Advance
Greets Chris

terjeio, 2020-07-02:
@einencool  Your questions are valid and it should be possible to implement these features. Rotation is perhaps best implemented by expanding G10L2 to recognize the R parameter in grblHAL?

I have been busy with maintenance work in my home lately - and I am not yet finished. It may take a little while before I am able to restart coding for the sender.

einencool, 2020-07-02:
No hurry, take your time, these projects are more for the winter time :-)
I have some pins in my surface so aligning the parts couldn’t be easier ;-)

einencool, 2020-07-07:
One question about the Probing Tab. Today I made exactly one test, and in this way I won‘t use it, because it fires the hole machine with G0 Codes, and with 5m/min I don‘t want to destroy my new 3D-Finder...
Is there a way to set the Feedrate for this down?
I made a simple macro for this and in my Opinion a speed with around 500mm/min is for me OK, maybe 1m/min. I think this needs to be adjustable for the user.

terjeio, 2020-07-07:
The probing tab is work in progress so changes will happen. I have some in the pipeline but I need to revisit the code before making any new commits. I use this [document as a reference](https://vers.by/en/blog/useful-articles/probe-screen) and this stipulates RAPID_SPEED for the initial movements. In grbl terms this is the maximum rate for a given axis. So perhaps we need an additional setting for the speed of initial movements then?

einencool, 2020-07-08:
Yes, I think that makes sense.
So everybody can use the speed he feels good with. 
For me it's way to fast and I couldn't use it with a good feeling :-)
I think "rapid" means only quicker than the measurement speeds...

einencool, 2020-07-11:
Hello @terjeio 

One question. Today I set the Speed in the GRBL Settings to max 750mm/min So I could make a little test for the 3D-Finder with better speeds.

Is it possible to while in Center Finding Mode to view the measured diameter from the Hole or Pin?
I've got a Ring with exact known Diameter and want to calibrate my 3D-Finder to the software, but I don't see any way in the Console to see the diameter.

This could also be helpful to set the Origins of a Workpiece with manual measurement methods

terjeio, 2020-07-11:
> Is it possible to while in Center Finding Mode to view the measured diameter from the Hole or Pin?

It should be, but perhaps after running two or more passes to ensure that the measurements are from the exact center?

My plan is do some grblHAL coding before I restart coding for the sender so it could be a little while before it get done.

einencool, 2020-07-11:
Sounds good, I‘ve seen some Center Finding Macros, where the Position will be measured in this direction:

X-
X+
Y-
Y+
And then X will be measured a second time from the real Center Point —> X- and X+

It would be great to get the possibility to measure a workpiece and get the „real“ dimensions from the touch points.
Sorry for this challenging work, but I didn‘t know how to do that

einencool, 2020-08-08:
Please, Please add the possibility to slow down the search feedrate while Probing. That would make things much easier.

einencool, 2020-08-30:
@terjeio 
Just a smiling reminder while you are working on the sender.

Here is a little summary from the last points.
- rotation of the g-code with a measurement of the workpiece position 
- get the "diameter" or width of the measured piece. (Maybe To calibrate the diameter of the 3D-Finder)
- when using a touch plate or the 3D-Finder is not in the axis of the spindle, where to set an offset for X and Y Axis

Maybe you find some time in the next weeks :-)
Thank you for your great support

terjeio, 2020-08-30:
> Maybe you find some time in the next weeks

I hope so, currently I am busy with maintenance work on my house. Rainy days might give me more time for coding...

My plan is to get a stable grblHAL version out first (moving test to master), then I will restart "serious" coding on the sender. Recent changes has been relatively minor: bug fixes and for testing of new features in grblHAL.

einencool, 2020-11-15:
Hello @terjeio 
I would like to ask about two things for the probing tab i mentioned before.

Is it possible to add a offset for the Touch probe, which is not mounted in the spindle?
I made a macro for the X and Y Offset, but when i only probe the  left side, i have to set it by hand, and it would make things much easier when this would be set automatically.

Second question is about the possibility in center finding only to measure one direction, maybe only X or only Y Axis.

I think the Rotation off the part is much more work, and the other two parts can be implemented much more easier.

Thank you in advance.
I hope i can test the autosquaring function in the next weeks, but for now it seems, that i have to make a bigger maintenance for the ballscrews...

terjeio, 2020-11-18:
> Is it possible to add a offset for the Touch probe, which is not mounted in the spindle?

This should be part of a probe profile? Add a button for advanced settings?

> Second question is about the possibility in center finding only to measure one direction, maybe only X or only Y Axis.

This could be done by unlocking and setting the distance for axis not to probe to 0?

> I think the Rotation off the part is much more work

It is as the gcode has to be transformed. I wonder if it will be possible to do that in the controller by implementing the `R` word of `G10 L2`...

einencool, 2020-11-18:
> > Is it possible to add a offset for the Touch probe, which is not mounted in the spindle?
> 
> This should be part of a probe profile? Add a button for advanced settings?
Yes, I think that would be a very good idea. 

> 
> > Second question is about the possibility in center finding only to measure one direction, maybe only X or only Y Axis.
> 
> This could be done by unlocking and setting the distance for axis not to probe to 0?
I think this is not the best option, when I think that I want to Probe a small hole (maybe 5mm ) then it could be not so good, or did I get into the wrong direction and when I have a small hole, I enter the 5mm and the rest will be executed with the clearance for the travel distance?

When you think this would be ok, then I would also recommend it :-)
But please make a little hint in the software about the possibility :-)

> 
> > I think the Rotation off the part is much more work
> 
> It is as the gcode has to be transformed. I wonder if it will be possible to do that in the controller by implementing the `R` word of `G10 L2`...

terjeio, 2020-12-14:
> Second question is about the possibility in center finding only to measure one direction, maybe only X or only Y Axis.

This is now available in the [beta 8 release](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v028-beta8), by unlocking XY sync and setting the direction not to probe to 0.

einencool, 2020-12-14:
Nice to hear, I‘ll give it a try in the next days :-)
Thank you Terje

einencool, 2020-12-30:
Since beta 8 makes some problems on my machine, I‘ve seen in Beta 7.3 that when probing „only the right outer wall side“ the probe touches the workpiece and after that drives directly die Z-Axis up, it doesn‘t clear the tip from the workpiece.
Haven‘t tried all other points, on the left side it‘s ok and on the front also.

terjeio, 2020-12-30:
Does this happens with beta 8 too? I would rather fix problems with this than reverting to beta 7.3 for bug fixing.

terjeio, 2021-01-14:
@einencool I have added some initial code for probe offset, currently only for setting G5x offsets. Do you want to test it to see if I am on the right track? If so I can upload to the edge version.

[image]

Note that the probe XY offsets are not visible with the default window size, the height has to be extended a bit first. Due to that I have added a new configuration for the sender in _Settings: App_ for restoring the last screen size on startup.

einencool, 2021-01-14:
Good Evening @terjeio 
Wow, that sounds great, I would love to test it :-)
Maybe I have tomorrow in the evening some time for testing, so if you could make the version available, I‘ll give it more than a try :-)

Then I can test if all sides are handled correctly.

One point from my side for testing, but I don‘t know if it is possible.
As I made some tests yesterday, I‘ve seen something, that would be great if it would be also possible to handle it separated .

When in Probing tab, I can configure the feeds & speeds for probing, and also there are F&S in the Config part for the tool Change.

I have configured the speed in the Toolchange Config relatively slow, because I don‘t want the Touchprobe „Search“ speed to high (50mm/min Search and 25 for Probing) 
I set it so slow, because I manually drive the Endmill and Touchprobe near the Touch Sensor.

But when doing the Reference Tool Length Measurement, it will use the Speed given from the Probing Tab. And for sideway probing, I would like to use the 200mm/min, but for Z-Probing only 50mm/min.

Would it be possible to separate X/Y and Z Speeds?

And while using the $TPW Command, the Tool probes two times, thats all Ok, but it drives the Z-Axis really high between the two probes, is there also a way to change this separately?

Only some points, and when it‘s not possible, it‘s ok, but when I don‘t ask, then I don‘t know :-)

Thank you :-)

terjeio, 2021-01-14:
@einencool ok, I'll upload tomorrow.

> Would it be possible to separate X/Y and Z Speeds?

A lot is possible but I am a bit conservative in not adding too many options to the user interface, probing is quite confusing already. Maybe an appsetting for advanced users could be a solution to that dilemma?

> And while using the $TPW Command, the Tool probes two times, thats all Ok, but it drives the Z-Axis really high between the two probes, is there also a way to change this separately?

This is currently hardcoded in the controller in _grbl/tool_change.c_ by `TOOL_CHANGE_PROBE_RETRACT_DISTANCE` and you can change it there. 

It can be mentioned that I have now completed an overhaul of the controller settings system making it easier to add new ones, and this is a candidate for joining the settings list. A remaining issue is that adding new settings will often trigger a settings reset - more work is required to fix that.

Part of the overhaul is that ioSender now gets setting information from the controller and groups settings in the UI - making it easier to navigate for the end user. This also eliminates the need to update the sender when settings are added.

einencool, 2021-01-14:
Perfect, thank you very much, and also for the travel distance, that‘s no Problem for me, to change it in this file.
When nobody else has a problem with it, then you can leave it, as it is.

You are absolutely right, that not to much settings should be added to confuse the users.

Then I think, I‘ll change the search speed to 100mm/min and go this way. Or another question.
I have a macro for setting the Tool length with my own speeds
Can I use this macro also for the reference measurement? 
When that works, everything is fine :-)

einencool, 2021-01-15:
One question for the "probe Retract Distance"
At the beginning of the file is the value 2.0f (line 37)
[image]

and a little bit deeper, there is "this value" and an additional "+2.0f" (line 200) means that, that there are 4mm of retract?
[image]

That would match my thoughts, that the distance is really high.

terjeio, 2021-01-15:
> there is "this value" and an additional "+2.0f" (line 200) means that, that there are 4mm of retract?

No, the extra 2mm is for setting the target of the second probe motion 2mm below the first probed z-position to ensure the probe is triggered again.

einencool, 2021-01-15:
Ah, alright that sounds good :-)

terjeio, 2021-01-15:
> That would match my thoughts, that the distance is really high.

It seems to be 2mm when I am testing - do you see a larger distance?

The [edge sender](http://io-engineering.com/downloads) for probe offset testing has just been uploaded. Be careful when testing, do not damage your probe!

einencool, 2021-01-15:
The distance looks so big, but I think you are right and it is only 2mm, I have 0.5mm in my Macro for probing.

One open point.
Can I make the reference measurement with the checkbox enabled, also with the macro?

Will try in the evening or maybe tomorrow your new version, now I'm not at home, and don't know how much time I get today :-)

terjeio, 2021-01-15:
> Can I make the reference measurement with the checkbox enabled, also with the macro?

Yes, use `$TLR` to set the reference after probing. Note that `$TLR` will not set the reference unless the last probe command was successful. If not successfull any previous reference will be cleared. Either way the _TLO ref'd_ LED will indicate the status.

Also note that `$TLR` can be issued at any time, data from the last probe command, whenever that was executed, will be used.

einencool, 2021-01-15:
I will now begin with some tests, the first thing I noticed, that the "Offset" is not available in Edge Finding Mode, Only in Center finding.
[image]
[image]

einencool, 2021-01-15:
> Yes, use `$TLR` to set the reference after probing. Note that `$TLR` will not set the reference unless the last probe command was successful. If not successfull any previous reference will be [...]

That means, that I need an extra Macro for this, if I understand this right?
My normal Macro for Probing is:
g91
g38.2 z-8 f50
g1 z0.35 f150
g38.2 z-0.5 f10
g90
G10 P1 L20 Z39.14
g1 z41 f250

And with this Macro I also Set the height for the Z-Axis.

In my understanding, I need a Macro like this one:
g91
g38.2 z-8 f50
g1 z0.35 f150
g38.2 z-0.5 f10
**g1 z5 f150**    // drive Z-Axis 5mm up
g90
**$TLR**

Would that be right?

The Offset seems to work really good.
I tested it in the outer and inner Edge Tabs, and that works great 👍

terjeio, 2021-01-15:
> Would that be right?

You'll have to try, I am not 100% sure about what you want to achieve.

> In the main tab it shows me the red "TLO ref'd" light, but I don't have the "clear" button in Probing Tab

The only way to clear that light is to use red _Reset_ button or issue `$TLR` after a failed probe.
The _Clear tool length offset_ button is for the TLO light - and it clears an actual tool offset, not the reference:

[image]

> that the Probe touches the Workpiece the first time and then raises an error, but I think, that's because I only had a piece of wood on the machine and pressed it down by hand...

This could well be so since an error may be raised if the probe triggers when not expected, e.g when retracting. This is enabled in the controller by the `$10` setting _Run substatus_ option (it is used by the sender for "simple" probe protection).

Thanks for testing, I am happy that it works as expected.

einencool, 2021-01-15:
What do i have to do, that i can see the “tlo” light?
In my picture, one comment above you can only see the “tlo ref’d” light....

and the question for the probe macro I’ll answer myself tomorrow 👍

terjeio, 2021-01-16:
> What do i have to do, that i can see the “tlo” light?

You have to set a tool length offset with `G43.1`. It can be cleared with `G49`.

The tool length reference offset is only used by the [tool change extension](https://github.com/terjeio/grblHAL/wiki/Manual,-semi-automatic-and-automatic-tool-change) to calculate and set `G43.1` offset on a tool change.

einencool, 2021-01-16:
Now I have tested a simple GCode with two tool changes.
The workflow is ok, and both macros are working fine.
- First I measure the Workpiece with the 3D-Finder and set my Zero Point,
- Then I measure the machine base with the $TLR Macro and the reference light will be red.
- But also after the complete GCode, I only have the one red light.
The Point "TLO" doesn't appear in my software, but after the measurement, I get the red "clear" button in Probing tab.
But the Clear Button is NOT there, when I have made the TLR measurement...

After TLR Measurement and also when Programm is finished:
[image]

Clear button appears after new tool is measured, I think in the past it was also there after the Reference was set:
[image]

This is the little test file
[Schubladeneinsatz-atc-test .txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5824309/Schubladeneinsatz-atc-test.txt)

This is the output from the console
[Probing to set zero.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5824305/Probing.to.set.zero.txt)
[Probing to set reference.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5824306/Probing.to.set.reference.txt)

einencool, 2021-01-16:
> Thanks for testing, I am happy that it works as expected.

I have to thank you 👍 
That makes working much easier, because now I don't have to set the offset every time by hand 🥇

terjeio, 2021-01-16:
> But also after the complete GCode, I only have the one red light.

This is due to `$10` setting option _Parser state_ not beeing checked, enabling this is the only way the sender can keep the light in sync with the controller status without constantly polling for the parser state. And I am not going to add polling for the parser state - too much overhead.
Visibility of the button in the probing tab does not need this since a check can be done manually on events triggered there.

I can be noted that legacy grbl does not support these status lights at all.

einencool, 2021-01-16:
Ah ok,
It‘s not a big Deal, but I was every time wondering, that you have both lights :-)

For me the Probing is now working really good.
Maybe in the future there would be the possibility to have two separate speeds for X/Y and Z Axis.
But since I can use my Macros now for Z-Probing, that‘s absolutely OK for me.

One point could be also interesting. 
The Probe-Profile will always switch to „default“ Maybe this could be saved to the last state or whatever.
I have changed the order in the Probing.xml, so that my 3D-Finder is in first place and will be the „default“ profile...

terjeio, 2021-05-06:
> Another question while I was looking at e.g. Estlcam that you can make a edge finding and measure the angle of the workpiece, so that the G-Code will be „rotated“ to match the workpiece. This would [...]

See issue #94 for edge version with gcode rotation.