terjeio/ioSender #44: 3d view - zoom out, zoom in, perspective

MARIOBASZ, 2020-06-18:
Is it possible to adjust the 3d visualization to taste, as it appears in the title?
I upload two examples of cutting with trochoidal strategy.
Maybe improve performance using G5.
In the second it seems that he is so close as to say "Houston, we are in trouble, I think we are going to collide, do not wait for us for dinner"
[Hexagon Trochoid example_1.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/4801012/Hexagon.Trochoid.example_1.txt)
[Hexagon Trochoid example_0.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/4801013/Hexagon.Trochoid.example_0.txt)
Congratulations and thank you!

terjeio, 2020-06-18:
> Is it possible to adjust the 3d visualization to taste, as it appears in the title?

I believe so: scroll wheel or <CTRL>+right mouse button for zoom, right mouse button for perspective and <SHIFT>+right mouse button for pan.

The second one was a bit odd, something is wrong with the initial scaling of the display that I need to take a look at.

MARIOBASZ, 2020-06-18:
Thanks, terjeiro

closed 2020-08-15 by terjeio