terjeio/ioSender #45: Number of Axes issue

phil-barrett, 2020-07-07:
Does the grblHAL in the microcontroller define the number of axes to the Sender?

I have a situation where a 5 axis build causes the G-Code Sender to only see 3 Axes on one PC but all 5 on another.  Here's a screen shot of a $i showing 5 axes but the sender only sees 3.

I can power cycle it and restart the sender - sometimes it will show 5 axes, sometimes only 3.
[image]

I have an ammeter on my USB feed and it will show lower power consumption, commensurate with 3 axes, when only 3 are seen. When it show 5 axes, it shows the higher power 5 axes will draw.

This only happens on one PC.  My main dev machine always show 5 axes.

terjeio, 2020-07-07:
A sender issue? Number of axes is reported on a $I as part of the `[OPT` string:

`[OPT:VNMSL,35,1024,3,8]`

Second last parameter is number of axes defined.

If you open the _Settings: Grbl_ tab is the settings list empty when this happens?

phil-barrett, 2020-07-07:
If you look in the picture I attached you can see [OPT:VNMSL,35,1024,5,0] but it only shows 3 Axes in the DRO section.

When I open Settings:Grbl it shows all the options, I can change them and see the changes take effect (I changed $0 to 5 and saw it take effect in the oscilloscope trace). It runs X - Z steppers just fine when it's like this.

What is strange, I can see my breakout board drawing less current when it comes up as 3 Axis - around 780 mA vs 1200 mA for 5 Axis. So in some ways grblHAL is behaving like it has 3 Axes but still reporting 5 Axes in $i.

When I move the board to my dev PC, it comes up as 5 Axis every time.

phil-barrett, 2020-07-07:
By the way, here's a photo of the test setup. You can see it's fairly portable. The ammeter shows the USB current draw in 100s of uAs (though only accurate to 1 mA).
[image]

terjeio, 2020-10-07:
I believe this is fixed in later releases, reopen if still a problem.

closed 2020-10-07 by terjeio