terjeio/ioSender #46: Manual Tool Change with M6 Commands

einencool, 2020-07-11:
Hello @terjeio 

I would like to ask you for the handling with M6 Commands in the G-Code.
I didn‘t test it in your Software, but I know the Workflow in the old Software that I was using.
In the Software I had to define a Position for the Tool length Sensor (TLS), where the length will be measured.

So the workflow was:
- Measure the length of the tool at the fixed Position
- Measure the height of the Workpiece (with the TLS)
- Start Milling will the first Tool

when the M6 Command appears, then the following happens:
- Spindle was Stopped and was driven to the specified Postion of the Sensor
- A Pop-Up says, which tool should be installed next
- After changing, the pop-up will be closed
- Tool length will be measured on fixed Position
- Machine goes to last Position of the Program, start the Spindle and begin to mill with the second tool

Can you please give me some advise, how to use your Software for Tool-Changing??

terjeio, 2020-07-11:
> Can you please give me some advise, how to use your Software for Tool-Changing??

Currently a TxM6 command will enter tool change mode and manual touch off can then be done via jogging. Earlier I did some work on the tool length offset tab, this is not yet complete but my aim is to integrate this into M6 handling. The 59.3 offsets will be used for the tool length sensor position. I am also going to add support for indexable tools, grblHAL can be compiled with a tool table that is saved in EEPROM.

einencool, 2020-07-11:
So I can manually jog to the Position of the Sensor?
Will the Spindle be automatically stopped and started?

And in the 59.3 offset I could enter the Position and it will drive there automatically?

terjeio, 2020-07-11:
> So I can manually jog to the Position of the Sensor?

Yes.

> Will the Spindle be automatically stopped and started?

No, code needs to be added to move the spindle to the tool change position and stop the spindle.
Then _only_  jogging and manual touch off is possible, a cycle start will then restart the program.
An example from a PCB drilling program:
```
M5
G0X0Y0Z25
T3M6 (MSG, 1 mm drill)
M3
```
Note that grblHAL will output (MSG, ) comments in sync with code, these will show up in the sender status line:

[image]

Note that only my sender supports the tool change extension.

> And in the 59.3 offset I could enter the Position and it will drive there automatically?

That is the intention, however this is currently on my todo list so not yet.

einencool, 2020-07-11:
Ok, thank you for your help.
Then I’ll give it a try in the next days. I hope i find where to put it into the Fusion360 Postprocessor, or do you or someone have a working PP for Fusion ?
I changed one two years ago, but sometimes it has problems with circles and helix and then the program didn’t run properly. Since these problems i always start with a „check mode“ cycle...

For my other CAD Software thats no problem in changing the code.
Thank you, so I’ll close this issue, but maybe someday you can make some improvements on this topic :-)

terjeio, 2020-07-12:
Then I’ll give it a try in the next days. I hope i find where to put it into the Fusion360 Postprocessor, or do you or someone have a working PP for Fusion ?

I am using Vectric VCarve so sorry no. The sender itself can convert Excellon PCB drill files to gcode and will insert the needed commands.

> Thank you, so I’ll close this issue, but maybe someday you can make some improvements on this topic :-)

It is on top of my agenda for the sender, so perhaps not too far away. Sometime after the summer holidays?

Usually I travel to a warmer land in the autumn - when there I normally work rather intensely on grblHAL and the sender as there is not much to distract me. The latest sender was born during last years visit. Willl it be possible to travel this year?

einencool, 2020-07-12:
Oh one last point, which is not absolutely clear to me.
Which coordinate system I have to choose for measuring the tool length? That was done always by the sender so I don't know it right now :-)
After measuring the length, then I set the height of the workpiece, right?
Today I'll try some points in Fusion, maybe I'll get it to work, but coding isn't my world :-(
Maybe there is one in the internet :-)

Well, traveling this year is really a good point... One month ago we traveled to the north see in Germany (where I come from) and a week later we got a new lock down in our region... And I think after the holidays the virus will spread more than before :-((

But we hope we will make the best of our time...

einencool, 2020-07-12:
So, I can't test it today, because I don't have the machine next to me, but this is the sequence, that Fusions PP will drop, when it comes to a Toolchange.
The Comments I have all the time in the NC, so I can see them, when I scroll trough the File..

G0 Z15
G17
M5

(+++++++++++++++++++++++++++++++++++++)
(++ 2D Fase1 ++)
(+++++++++++++++++++++++++++++++++++++)
T41001 M6 (MSG, 90g Entgrater)
(++ 90g Entgrater ++)
(+++++++++++++++++++++++++++++++++++++)
S12000 M3
G4 P5
G54

terjeio, 2020-07-12:
> Which coordinate system I have to choose for measuring the tool length?

Any but G59.3? It depends on how you want to apply the tool offset. If by changing the current offset Z-value and G59.3 is used for the measuring position then changing G59.3 is not a good idea? If by setting the offset with G43 then G59.3 can be used as well. Note that it is currently only possible to jog, set DRO values and Cycle Start when the TOOL state is active.

An idea to test with the PP is to assign the tool change position to above the measurement position to G59.3 and then issue:

```
G59.3
G0Z0
G0X0Y0
G54
Tn M6
```

Change the tool and manually touch off by jogging and set the new Z position (manually calculated?) in the DRO when the P led lits up.
I do not know if it is possible (or safe) to do automatic touch off after the tool is changed by issuing a G38.2 (or G38.3) command and then use G10L20 to set the offset. If it is then the deceleration distance should perhaps be subtracted from the G10L20 offset provided.

The sender needs to be extended to handle manual tool changes better. Since I am not a machinist I have a hard time understanding all the details surrounding this, but hopefully I will be able to create a usable solution that caters for all the different ways probing can be done...

einencool, 2020-07-12:
Ah Ok, then I'll stop it and make it the way I used before. Touch of per Hand is not the way I want to go. That's why I have a TLS  :-)
Now I'm using the G10L20 command after the G38.2 

Then I have to look around, what I can do with separate files and go with the offset of the TLS...
For now I don't know how, but I think it'll be possible to do.

I know exactly what you mean. And I'm absolutely thankful for your great work. We all know that there is much to do, but you are on a very good way :-)

So I open it, if someone has a good solution for this.

einencool, 2020-07-17:
One next question.
I've got Tool number with 4 digits, and the sender says everytime when I use more than 2 digits, that the number is to high.
Can you please adjust this? In the Postprocessor of GRBL, there is the possibility to set it to 9999.

I use this, because so I can see, which tool should be used in Fusion
x000 - Diamater of Tool
0x00 - Special Shape maybe Ball Nose, Taper or something else
00x0 - number of Flutes
000x - Counter for Endmills from different length / Stores etc...

terjeio, 2020-07-17:
grbl uses a 8-bit value for storing the tool number so max is 255. I'll change it to a 32 bit number in the next commit thus changing max to 4294967295.

NIST defines tool number as an integer without specifying a max value so I guess this change is ok, however large tool numbers may break senders so at least I have to check mine for how those are handled.

Also note that if grblHAL is configured with a tool table then max tool number is limited to number of entries in the tool table. In such a case "intelligent" tool numbers as you show above will not work as the tool table is stored in persistent memory (typically EEPROM) that has limited space.

einencool, 2020-07-18:
The tool table makes in my opinion only sense, when I have a ATC Spindle.
Then you're absolutely right. Because then I have fixed offsets which could be stored. 
But with manual tool changing and a ER Collet system it must be set every time.

And since the M6 command didn't work for me now, it's no problem at all, because the tool number is commented out. :-)

terjeio, 2020-07-18:
> The tool table makes in my opinion only sense, when I have a ATC Spindle.

That is true, however for me it makes sense even for manual tool changes since I have a small collection of indexed tools for my lathe.

terjeio, 2020-08-05:
@einencool Take a look at the [coming changes in grblHAL](https://github.com/terjeio/grblHAL/issues/79) to see if it makes sense for you.

einencool, 2020-08-27:
Hello @terjeio,
I hope I have a little bit of time in the next days to test your Tool Change commands and your Probing tabs.

I have a little question just to reduce the chance that I'm doing some steps in the wrong order.

On my machine I have a movable Z-Touch Probe (40mm height) and a 3D-Finder (which is not mounted in the Spindle, it's offset from the spindle (lets say X-80 / Y-80mm). I think the Tool Change Mode # 1 ($341=1) should work best for me

When I want to machine a part with several bits, I have to use the "Establish reference Offset" Option in the "Tool length Offset" Tab first on the "base" of my machine (or a repeatable Place).
[image]
and then [image]

After that, I can move to the top of my workpiece and start the "TLO" Command the second time without the "reference"
[image]
and then [image]

Then the heigth of the workpiece should be set to "0mm" and the DRO should show something like 40mm + probing clearance 
Where do I have to enter these 40 mm (Touch Plate or Fixture Thickness?) 
I tried it just with the teensy 4.0 on my desk with an switch for the touch probe, but I didn't get it to work, when I touch off the "workpiece" it should the say 40,00 + xx mm lift up
Or doesn't it work because I can't home my machine on startup?
[image]

terjeio, 2020-08-27:
I have made a mistake for $341=1 - by automatically continuing the program after the $TPW command is issued and probing is complete. The correct behaviour is to retract a bit after probing and wait for a cycle start so there is time to remove the touch plate. Will test a bit more tomorrow and come back with info.

einencool, 2020-08-29:
Now I tried to set the Tool length in the Program while I'm at the machine.
First I set the reference to the base of my machine and after that I want to Zero the tool on my workpiece with the "normal" TLO" command.

But it doesn't show me any values near to 40 or 0 mm in the DRO. It displays -77mm which is from the Homing Position.
Did I miss something?

Another question:
I normally have a macro for the Z-heigth. 

When now in Probing Tab, I have always to go to the laptop to activate the command and then the machine starts to run...
Is it possible that a shortcut could be activated like "Shift + Enter" or something like that to the "START" Button with an extra Clickbox like  you have for the Macros before they were activated?
Then it would be possible to watch the machine while starting it, because the laptop is standing behind the machine and I can't see anything what happens...

terjeio, 2020-08-30:
I think the correct way to establish correct offset is to first ensure there is no tool offset active. Then probe Z with _Edge finder_ and _Touch plate_ height set to 40mm. This should establish 0 at the workpiece height. After this probe again with _Tool length offset_ to establish the tool reference offset (this should be possible to do when probing with _Edge finder_?).

> Is it possible that a shortcut could be activated like "Shift + Enter" or something like that to the "START" Button with an extra Clickbox like you have for the Macros before they were activated?

I have assigned ALT+R for start to be consistent with the main screen, should I rename the button to _Run_? I have tried to add ALT-S for stop but I could not get it to work - I'll look into that when I have more time. No extra clickbox yet - is that needed?

Beta 5 is now up, grblHAL needs to be updated as well for the latest changes to work.

einencool, 2020-08-30:
Thank you for your fast support , I'll test it in the next days.

So I made always a mistake with my work flow.
I have to measure the Z-height in the Probing / EdgeFinder tab and after that I have to use the TLO measurement, is this right?
That was not clear for me, so I started always with the TLO measurements.

How do I delete the tool offset if there is any? Or where can I see that. Sorry for this noob questions, but I haven't been working with it until now.

I think the shortcut will be absolutely ok. And I thought a click box could help, that the machine doesn't start without confirmation, like your macros, but I don't think that it will be needed.

terjeio, 2020-08-30:
> I have to measure the Z-height in the Probing / EdgeFinder tab and after that I have to use the TLO measurement, is this right?

For now yes. I need to write more documentation and figure out what common workflows will be like... And many small issues still to be resolved.

> How do I delete the tool offset if there is any?

There will be a _Clear_ button in the _Tool length offset_ tab if any is active:

[image]

IMO it should somehow be visible that an offset is active in the _Work Parameters_ box in the main tab as well.

einencool, 2020-08-30:
Ah Ok, I understood.
Then I just made a mistake every time *g*

It would be great if there would be a little guide, so everyone knows in which order the commands should be.

I hope I can try it in the next days. Thank you :-)

einencool, 2020-09-05:
I'll close it here, because we have two open issues it it is not so good to follow both :-)

https://github.com/terjeio/grblHAL/issues/79#issue-673516579

closed 2020-09-05 by einencool