terjeio/ioSender #47: Ethernet configuration questions.

phil-barrett, 2020-07-12:
I see how you can specify an IP address for grblHAL to the sender. There does not appear to be a mechanism to find the IP address. In grblHAL there are extensions to set all the network parameters but I don't see them in TSender.  Do they become available under some circumstance? How does one make the initial connection when grblHAL is configured to use an Ethernet protocol. I am thinking that maybe an initial connection could be made via USB to configure?  Is it possible to connect via USB and then switch to an Ethernet protocol?

I've looked at other motion controllers and they seem user unfriendly. Lots of comments on Ethernet config problems on CNCZone. Some involve directly connecting the PC to the controller via an Ethernet cable. Then changing the IP address of the PC to match the default IP address on the controller to connect and change the controller to a local router IP address. A number of them suggest a second Ethernet NIC and to stay directly connect to the controller. Given that most machinists are pretty far from network gurus, I think that is a poor approach.  Having USB can make it pretty simple to set up.  

Anyway, I think making this super easy for a user would be a huge win for grblHAL.

terjeio, 2020-07-12:
> In grblHAL there are extensions to set all the network parameters but I don't see them in TSender.

They are available in the _Settings: Grbl_ tab.

[image]

[image]

> Do they become available under some circumstance?

When the networking plugin is added to the build.

> I am thinking that maybe an initial connection could be made via USB to configure?

grblHAL always starts up in serial communication mode (UART or USB) and switches protocol on connect/disconnect.

> I am thinking that maybe an initial connection could be made via USB to configure? Is it possible to connect via USB and then switch to an Ethernet protocol?

Switching is automatic and happens on connect.

---

I have not yet added a way to input the connection parameters in tsender, the config file has to be modified manually. Another todo item.

> A number of them suggest a second Ethernet NIC and to stay directly connect to the controller.

I have my SmoothStepper controller for Mach3 connected that way, it ensures no other traffic on the cable and is the optimal configuration.

phil-barrett, 2020-07-12:
Thanks. Sometimes I am a bit slow but I think you are saying something similar to what I was thinking.  It makes sense.

To recap what I think you said: You connect to a newly flashed grblHAL system via USB with TSender. You enter the network params (which are visible because of the network plugin) and then disconnect TSender.  Then you should be able to ping the grblHAL system (assuming network connected) and with the right ip address, mask and gateway in Tsender, it should connect. 

What defines a connection to grblHAL - physical presence of USB or an actual sender or terminal connecting?  i.e. Do I need to remove the USB cable to force it to switch? (er move the cable to just a power source?)

Also, I think this means that a static IP address is required as there is no discovery mechanism for a DHCP allocated IP addr.

terjeio, 2020-07-12:
> What defines a connection to grblHAL - physical presence of USB or an actual sender or terminal connecting?

The actual sender connecting, when it disconnects communication falls back to serial. There is no need to remove the USB cable. In fact some of the reporting still appears on the serial output.

> Do I need to remove the USB cable to force it to switch?

A reboot of the controller may be needed after networking parameters are changed.

> Also, I think this means that a static IP address is required as there is no discovery mechanism for a DHCP allocated IP addr.

It is a configuration option - DHCP is supported. When enabled then the settings for IP address etc. are no longer relevant/available. From [settings.h](https://github.com/terjeio/grblHAL/blob/master/grbl/settings.h):

```
...
    // Normally used for Ethernet or WiFi Station
    Setting_Hostname = 300,
    Setting_IpMode = 301,
    Setting_IpAddress = 302,
    Setting_Gateway = 303,
    Setting_NetMask = 304,
    Setting_TelnetPort = 305,
    Setting_HttpPort = 306,
    Setting_WebSocketPort = 307,
...

typedef enum {
    IpMode_Static = 0,
    IpMode_DHCP,
    IpMode_AutoIP
} ip_mode_t;
```

Switching to/from MPG mode and SD card streaming is still possible when network comms is in use so IMO a pretty flexible solution. It is all achieved by swapping HAL function pointers...

phil-barrett, 2020-07-12:
Great! It seems you have thought it all through.

By the way, I understood that DHCP is possible. What I meant was that it's likely to be hard to use because you will have to  figure out the new IP addr when it changes (or not ever power down the grblHAL machine).

terjeio, 2020-07-12:
> What I meant was that it's likely to be hard to use because you will have to figure out the new IP addr when it changes (or not ever power down the grblHAL machine).

Two (three) options:

1. Assign a unique hostname and use that to connect.
2. Bind the controller MAC address to a fixed IP address in the router. I've used this during testing.
3. Use SSDP?

Not sure about the last one, I have not yet added that protocol to the networking plugin.

phil-barrett, 2020-08-08:
Hi, now that we are making progress on the Teensy 4.1 Ethernet support, I am seeing somewhat different behavior than described above.  I'm using Beta-1, if that matters.

$302-$304 (IP Addr, gateway and mask) are not showing up in the Settings: Grbl tab when I am connected via websocket or telnet.

Also, with Ethernet running, I am unable to connect via USB at all.  Even with invoking from a cmd.exe window "GCode Sender.exe"-port COM23:115200,N,8,1 (com23 is the correct port).  Or changing App.config to portparams line:

> <PortParams>COM23:115200,N,8,1</PortParams>

 Even after power cycling the teensy. In each case it sits with:

> Waiting for controller (COM23:115200,N,8,1)...

in the lower status bar.

terjeio, 2020-08-08:
> $302-$304 (IP Addr, gateway and mask) are not showing up in the Settings: Grbl tab when I am connected via websocket or telnet.

Is this with DHCP enabled? If so that is intentional as these settings does not make sense then, see enet.c - `ethernet_settings_report()` function.

> Also, with Ethernet running, I am unable to connect via USB at all.

Strange, it works for me. Are you sure you are using the latest driver.c?

phil-barrett, 2020-08-08:
Ah, that makes sense.  Is there are way to turn DHCP off from the sender or console?

terjeio, 2020-08-09:
>  Is there are way to turn DHCP off from the sender or console?

$301 can be used for that. There is an error in the value assignments in my sender: 0 = static, 1 = DHCP, 2 = AutoIp is the correct ones. I'll have to fix that. Some sensible values should be set as default when DHCP is enabled in the config file - more todo...

After changing to static mode settings should be reloaded to make $302-$304 show up. Another todo item is to somehow declare settings dependencies in the definition file used by senders in order to avoid that, my long term plan is to switch to xml or json to add such information (and grouping for settings too...).

phil-barrett, 2020-08-09:
Playing with it now.  Looking pretty good with some minor issues.

There seems to be a bug in dealing with $301.  Sometimes it doesn't update the value shown (after clicking on save) and a couple of times TSender crashed when saving (could be unrelated).  

If I click on autoIP to get 0 for static, $302 - $304 aren't displayed.  I have to exit TSender and restart to get it to show.

I got $301 set to 0, entered a static IP addr and GW, Mask was there.  Saved.  Power cycled teensy. Restart with -port \<static addr\> and it works.  Nice. 

One issue. if you enter an error IP addr or GW or Mask, it seems to brick the Teensy.  I'll report this in the grblHAL issues section.

A minor annoyance - Editing the octal fields (correct term?) xxx.xxx.xxx.xxx looks like it's just text. As such if you just start typing you run into the field width limit - need clear it out first.  It would be nice to have a dot or tab move to the next octal field.

Minor visual issue - the Value column in Settings: Grbl tab isn't wide enough to show all 4 fields and no way to resize columns.

phil-barrett, 2020-08-09:
I don't know if I'd call this a bug but in order to get App.config/PortParams to update when you change the IP address, you have to go into Settings: App and press Save Settings.  Would be nice to have any change that needs to be reflected in the PortParams field to be automatically done.

terjeio, 2020-08-09:
>  It would be nice to have a dot or tab move to the next octal field.

Well, this is not easy to code - I believe Windows does this with overlapping input fields looking like a single one. It was hard enough to get numerical fields working properly, this would add another layer of complexity. For a rainy week...? Or use a 3rd party library? I try to avoid that.

> Would be nice to have any change that needs to be reflected in the PortParams field to be automatically done.

I need to think about this - I am not sure it is a good idea.

Anyway, more testing is needed.

phil-barrett, 2020-08-09:
> I need to think about this - I am not sure it is a good idea.

Here is the sequence of events that currently needs to be done:

1. change ip/gw in Settings: Grbl
2. hard reset
3. start sender with -port \<new ipaddr in specific protocol form\> from a cmd window
4. go into Settings: App and press save
5. from then on they can run TSender without -port

Now, they can just start sender and it will connect via what ever protocol was selected.  Note, it has been my experience that the average windows user hates/fears running a cmd.exe window. 

Do you agree it would be a lot more friendly to have this sequence?
1. change ip/gw in Settings: Grbl (app.config Params field gets updated) with a warning that they need to do a hard reset
2. hard reset
3. Start TSender without -port...

Maybe in #1, give them the option of making it the default.

As it currently works, there is no indication that network changes need a hard reset to become active. 

And as long as I am at it, the only way to make a -port value "permanent" in app.config is to go into Settings: App and press Save Settings. (They could also edit app.config. Not sure that is desirable.) It would be nice to have that behavior noted/shown in the tab. Maybe show both the current default connection and the current active one with a button to update.

terjeio, 2020-08-09:
How likely is it that most users will use a static IP vs. DHCP? What kind of users will want to use a static IP? Will they typically set the static addresses in driver.h? What about other senders? more?

As for my sender the configuration UI has currently no knowledge of what the settings means, it is completely driven by a text file (hal_setting_codes_en_US.txt). I would like that any further UI-"intelligence" should be added there and not hardcoded in the sender. What I have on my todo list is to add IP connection options in the connection dialog, and possibly a menu item for bringing that up. So IMO a bit of thinking is required. I agree it needs to be improved, an a first step to get us going is perhaps a Wiki page? An optional wiki link could also be added to hal_setting_codes_en_US.txt and presented in the UI as well? This would be nice for other features as well, e.g. the new tool change funtionality.

phil-barrett, 2020-08-09:
I agree - a wiki entry is a good start. And I really like the idea of an optional wiki link. This would make it easy to solve problems and quickly provide support.

I would like to hear more about your IP connection dialog. It sounds like the right direction.

The problem with a dynamic IP is that it is, well, dynamic. Consider a machine with a dhcp ip address and the user doesn't use it for a week or two.  He powers up the machine and tries to connect.  If a different IP addr gets assigned, it won't connect to the sender. He will have to do something to fix it.  A static IP address eliminates this issue. How likely is it? That depends on what we tell them to do. I would prefer to tell my users to set a static IP address. I just worry that these really great features will be a source of frustration for the users. Networking always seems to cause problems for people. There are a number of complaining threads about setting up networking for various commercial motion controllers. 

By the way, I do appreciate your points on this. It is a tricky area to get right.

terjeio, 2020-08-09:
> The problem with a dynamic IP is that it is, well, dynamic.

It does not have to be. IMO it is a safer bet to connect with DHCP and then reserve the address in the router. If assigning a static IP and then someone connects with a device (a mobile phone?) to the network and get your static IP then what?

In my router it is a matter of checking a box and I get a static lease, nobody else can then get the assigned IP:

[image]

terjeio, 2020-08-10:
> change ip/gw in Settings: Grbl (app.config Params field gets updated) with a warning that they need to do a hard reset

I have added notes in the `hal_setting_codes_en_US.txt` file about this. I have commited this to the subversion repo, you can find it in the Resources folder. Replace the one in your sender folder to see the changes.

I have commited an update for the T4 driver as well, network settings values are copied at stack init so that any later changes does not mess it up. ESC followed by CTRL+T will now hard reset the controller, currently available in all connections modes - but should perhaps be restricted to UART mode as connection will be lost for others. For USB I need to disconnect/reconnect the USB cable before I can connect again (Win7). This is perhaps a reason for the lack of a reset button on the T4?