terjeio/ioSender #49: beta 1 z probe

jschoch, 2020-08-10:
I'm getting a NaN in place of the Z postion below when z probing.

```
G53G0Z-163.048
ok
PM:ok
<Run|MPos:-84.446,-62.372,-164.066|Bf:14,512|FS:108,0|Pn:P>
<Run|MPos:-84.446,-62.372,-164.020|Bf:14,512|FS:180,0>
<Run|MPos:-84.446,-62.372,-163.322|Bf:14,512|FS:160,0>
<Run|MPos:-84.446,-62.372,-163.195|Bf:14,512|FS:88,0>
<Idle|MPos:-84.446,-62.372,-163.048|Bf:15,512|FS:0,0>
G10L2P1ZNaN

```

terjeio, 2020-08-10:
I am not able to replicate, but I have this comment in the code so I have seen the problem before:

`pos.Z += probing.Depth; // Sometimes NaN...?`

I could be settings related, can you please post them?

jschoch, 2020-08-11:
[image]

terjeio, 2020-08-11:
I meant the controller settings. Somehow one of the sender internal position structs is likely not initalized when this happens.

By the way, is this issue consistently repeatable for you?

terjeio, 2020-08-16:
A new [beta release](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v022-beta2) is now available, with luck this will resolve this issue. If not I need enough information to be able to replicate it.

jschoch, 2020-08-16:
Still have the problem with beta-2.  Below are my settings, this is bart's port.  Beta 2 also does not retract the probe when doing heightmap.  It runs the seek, then the latch, then it just moves in X with the probe down.

verbose output 

```
ok
<Idle|MPos:-84.310,-109.231,-108.789|Bf:15,512|FS:0,0>
Probing...
G91F100
ok
PM:ok
G38.3F100Z-10
<Run|MPos:-84.310,-109.231,-109.343|Bf:14,512|FS:100,0>
<Run|MPos:-84.310,-109.231,-109.697|Bf:14,512|FS:100,0>
<Run|MPos:-84.310,-109.231,-110.011|Bf:14,512|FS:100,0>
<Run|MPos:-84.310,-109.231,-110.344|Bf:14,512|FS:100,0|WCO:-84.310,-112.231,-112.835>
<Run|MPos:-84.310,-109.231,-110.687|Bf:14,512|FS:100,0|Ov:100,100,100>
<Run|MPos:-84.310,-109.231,-111.035|Bf:14,512|FS:100,0>
<Run|MPos:-84.310,-109.231,-111.370|Bf:14,512|FS:100,0>
<Run|MPos:-84.310,-109.231,-111.701|Bf:14,512|FS:100,0>
<Run|MPos:-84.310,-109.231,-112.035|Bf:14,512|FS:100,0>
<Run|MPos:-84.310,-109.231,-112.370|Bf:14,512|FS:100,0>
<Run|MPos:-84.310,-109.231,-112.724|Bf:14,512|FS:100,0>
[PRB:-84.310,-109.231,-112.788:1]
ok
PM:ok
G0Z0.5
ok
PM:ok
G38.3F25Z-10
<Run|MPos:-84.310,-109.231,-112.884|Bf:14,512|FS:144,0|Pn:P>
<Run|MPos:-84.310,-109.231,-112.412|Bf:15,512|FS:0,0>
<Run|MPos:-84.310,-109.231,-112.492|Bf:14,512|FS:25,0>
<Run|MPos:-84.310,-109.231,-112.561|Bf:14,512|FS:25,0>
<Run|MPos:-84.310,-109.231,-112.650|Bf:14,512|FS:25,0>
<Run|MPos:-84.310,-109.231,-112.734|Bf:14,512|FS:25,0>
[PRB:-84.310,-109.231,-112.790:1]
ok
PM:ok
G53G0Z-108.79
ok
PM:ok
<Run|MPos:-84.310,-109.231,-112.794|Bf:14,512|FS:144,0|Pn:P>
<Run|MPos:-84.310,-109.231,-112.267|Bf:14,512|FS:432,0>
<Run|MPos:-84.310,-109.231,-111.910|Bf:14,512|FS:522,0>
<Run|MPos:-84.310,-109.231,-110.587|Bf:14,512|FS:550,0>
<Run|MPos:-84.310,-109.231,-109.920|Bf:14,512|FS:406,0>
<Run|MPos:-84.310,-109.231,-109.111|Bf:14,512|FS:190,0>
<Run|MPos:-84.310,-109.231,-108.815|Bf:15,512|FS:0,0|Ov:100,100,100>
<Idle|MPos:-84.310,-109.231,-108.790|Bf:15,512|FS:0,0>
<Idle|MPos:-84.310,-109.231,-108.790|Bf:15,512|FS:0,0>
G10L2P1ZNaN
Probing completed
error:2
<Idle|MPos:-84.310,-109.231,-108.790|Bf:15,512|FS:0,0>
G91
ok
ok
```

settings

```
$0=3
$1=255
$2=0
$3=3
$4=0
$5=1
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=2
$24=200.000
$25=1000.000
$26=200
$27=5.000
$30=1000.000
$31=0.000
$32=0
$33=5000.000
$34=0.000
$35=0.000
$36=100.000
$80=0
$81=0
$82=0
$83=0
$84=0
$90=0.000
$91=0.000
$92=0.000
$93=0.000
$94=0.000
$100=3850.000
$101=3850.000
$102=10495.000
$110=1000.000
$111=1000.000
$112=600.000
$120=50.000
$121=50.000
$122=30.000
$130=240.000
$131=136.000
$132=200.000
$140=0.250
$141=0.250
$142=0.250
$150=50.000
$151=50.000
$152=50.000
$160=16
$161=16
$162=16
$170=16
$171=16
$172=16

```

terjeio, 2020-08-16:
Thanks.

I have found the NAN issue, it is due to the sender expecting tool offset values for all axes in the TLO report. grblHAL and some other ports outputs outputs values for all axes, but not vanilla grbl. A minor issue remaining for vanilla grbl is that there is no way of knowing which axis the offset is reported for, so I will assume Z for now. It is a compile time option in vanilla grbl, `TOOL_LENGTH_OFFSET_AXIS`.

Heightmap probing is working as it should for me, here is the verbose output:

```
Probing...
G91F100
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F300 S0.]
ok
PM:ok
G38.3F100Z-10
<Run|WPos:0.000,0.000,2.738|Bf:34,1023|FS:100,0|WCO:-58.395,49.993,-14.726|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0>
[GC:G38.3 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F100 S0.]
<Run|WPos:0.000,0.000,2.405|Bf:34,1023|FS:100,0>
[GC:G38.3 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F100 S0.]
<Run|WPos:0.000,0.000,2.072|Bf:34,1023|FS:100,0>
<Run|WPos:0.000,0.000,1.738|Bf:34,1023|FS:100,0>
<Run|WPos:0.000,0.000,1.384|Bf:34,1023|FS:100,0>
<Run|WPos:0.000,0.000,1.051|Bf:34,1023|FS:100,0>
<Run|WPos:0.000,0.000,0.717|Bf:34,1023|FS:100,0>
<Run|WPos:0.000,0.000,0.385|Bf:34,1023|FS:100,0>
<Run|WPos:0.000,0.000,0.030|Bf:34,1023|FS:100,0>
[PRB:-58.395,49.993,-14.734:1]
ok
PM:ok
G0Z0.5
ok
PM:ok
G38.3F25Z-10
<Run|WPos:0.000,0.000,-0.151|Bf:34,1023|FS:240,0|Pn:P>
[GC:G38.3 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F25 S0.]
<Run|WPos:0.000,0.000,0.316|Bf:34,1023|FS:25,0>
<Run|WPos:0.000,0.000,0.233|Bf:34,1023|FS:25,0>
<Run|WPos:0.000,0.000,0.150|Bf:34,1023|FS:25,0>
<Run|WPos:0.000,0.000,0.061|Bf:34,1023|FS:25,0>
<Run|WPos:0.000,0.000,-0.022|Bf:34,1023|FS:0,0|Pn:P>
[PRB:-58.395,49.993,-14.724:1]
ok
PM:ok
G1F1500Z3
ok
PM:ok
G1F1500Y5
...
```
G1F1500Z3 above is the retract command, is this incorrect or missing for you?
Retract distance is from _Diameter_ in _Probe tip/tool_, I guess it should have been _Offset_ from _Probing clearances_?
Tooltips for _Probing clearances_ seems to be wrong to me - I need to check them again.

I have a fix for the NaN issue, but I would like resolve the heightmap issue before uploading a new build.

jschoch, 2020-08-16:
The "Diameter" in probe tip/tool threw me off.  I had it set to 0.  

Many of the settings in the probing tab are ambiguous, for instance what is the difference between touch plate and fixture?  Why is there a "work piece height" setting for Z probing in the "edge finder" tab?  This is why I attempted to have the 3d renderer draw the tool path for the given settings.   Perhaps I should revisit that if you would support merging it.  Alternatively we could just do static drawings with clear labels that correlate the settings visually based on the current mode.  I'd be happy to work on that if you wanted help with it. 

I'd honestly prefer discrete settings for ever mode over ambiguous re-used settings.  This would require having the defaults/custom probe config updatable for each probing mode.

terjeio, 2020-08-17:
Fix for NaN issue added in [latest release](https://github.com/terjeio/Grbl-GCode-Sender/releases/tag/v023-beta3).

As for the how probing is implemented a good starting point would be to add a wiki page as a reference/specification and proceed from there? I am not a machinist so I am open for input and I have to admit I find the probing stuff challenging, e.g. the concepts used in the info I have found on the web does not seem to be consistent. It does not help that english is a second language for me...

I have used [this document](https://vers.by/en/blog/useful-articles/probe-screen) as a basis for how I have implemented probing. The [LinuxCNC documentation](http://linuxcnc.org/documents/) should, IMO, also be used as a reference for how concepts are defined/used.