terjeio/ioSender #51: Linux support

leventelist, 2020-08-12:
Does this code tested on Linux host?

terjeio, 2020-08-12:
Not tested, it may be possible to run it under [Mono](https://www.mono-project.com/)? I do not have hardware to try.

leventelist, 2020-08-12:
Unfortunately, id doesn't work. I guess it misses some libraries.

The entry point method could not be loaded due to Could not load file or assembly 'PresentationFramework, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies.

kdomanski, 2021-03-15:
Unfortunately this program uses WPF, which can be used neither with Mono, nor the new Linux release of .NET.

benishor, 2021-03-15:
What would it take to make a Linux compliant version? How decoupled is the GUI code from the core?

terjeio, 2021-03-16:
What would it take to make a Linux compliant version?

That would be a complete rewrite?

> How decoupled is the GUI code from the core?

It is [MVVM](https://docs.microsoft.com/en-us/xamarin/xamarin-forms/enterprise-application-patterns/mvvm) based.

dresco, 2021-03-16:
> That would be a complete rewrite?

I wonder if something like [Avalonia](https://avaloniaui.net/) would be viable? Though I'm sure you have more than enough to do already! ;)

olijouve, 2021-12-20:
> Unfortunately this program uses WPF, which can be used neither with Mono, nor the new Linux release of .NET.

With .NET Core 3.0’s support for WPF, a WPF application can run on Linux under Wine.
https://ccifra.github.io/PortingWPFAppsToLinux/Overview.html

Now as commented by @dresco, Avalonia cross plateform MVVC could be viable.
Here is a feed back of a ported where the autor claims a super easy time porting from WPF to Avalonia...
https://www.reddit.com/r/dotnet/comments/pl4d1o/i_finally_ported_from_wpf_to_avalonia/

Avalonia could be a great choice for an upcoming release if everything else can be built with mono but WPF.