terjeio/ioSender #52: MDI focus

jschoch, 2020-08-17:
The MDI box isn't keeping focus and I keep accidentaly jogging with the arrow keys vs using the command history.  I almost think entering focus on the MDI box should be fully modal so you can't do anything else by accident.  I don't think this was a problem in my initial fix.  Happy to take a look myself if you've done something unintentionally here.

terjeio, 2020-08-17:
Note that focus is taken away from the MDI on tab changes or on pressing the Send button, has always been like that. The first is coded for, the second is default Windows behaviour? AFAICT your initial fix did not handle this differently.

jschoch, 2020-08-17:
would you be open to making it full modal?

terjeio, 2020-08-17:
Can look into that yes, not sure if it possible/easy to do for all possible scenarioes. Clicking on a button taking away focus is IIRC a property of the button as clicking on the override buttons does not take it away from the MDI - so fixing those should be easy. Perhaps tab change and DRO input will be easy as welll but for those I guess I need to save away the currently focused element if it is the MDI in order to refocus later.

terjeio, 2020-08-18:
I have added a [test branch](https://github.com/terjeio/Grbl-GCode-Sender/tree/test) if you want to compile yourself.
MDI will no longer lose focus on a tab change or by pressing Send. When editing DRO values focus is not given back as this will be harder to change, hope this is ok.
Pressing the reset button will also result in the DRO loosing focus - it is not enough to set the button as not focusable, there is a lot going on behind the scenes when a reset is detected.

terjeio, 2020-08-27:
New release now up that includes the change in the test branch.

closed 2020-08-27 by terjeio