terjeio/ioSender #53: Thanks!!! Winning!

jschoch, 2020-08-17:
I just wanted to post a success for the probing work.

Here is a very simple board I just ran using the height mapping via my kinetic probe and with tool changing using a tool offset probe.  It was PERFECT!!!! (well the holes are a smidge off, need to tune my rapids down).  The holes are interpolated and this was cut at 500mm/m.  The process I used was:

traces loaded, heightmap probed (it did not seem to set the Z height to G54 Z0 though so I ran G38 manually :( ), G30 to tool height setter, TLO reference set, 90deg dual flute cutter installed, TLO calculated, traces cut, G30, 0.8mm endmill installed, TLO calculated, File 2 for drill and cutout opened, heightmap applied, cycle start and done.

[image]

Here is the before on the traces:

[image]

And a little 320 sanding and scotchbrite + Isopropyl

[image]

I also cut some 0.1 mm traces today with some new bits I am testing out.

Thanks for sharing your hard work!!!!

jschoch, 2020-08-17:
ok so it turns out I have "kinetic" on the brain...  My probe wasn't kinetic, it was kinematic however after I posted this I thought I'd do another board and I actually turned my probe kinetic and the tip has a nice 90deg bend in it.  It was not the sender's fault but rather the little JST connector didn't do it's job and while the probe's LED was brightly lit, no signal made it to grbl :(

terjeio, 2020-08-17:
> it did not seem to set the Z height to G54 Z0 though so I ran G38 manually

The reason for this is the same as issue #49.

> Thanks for sharing your hard work!!!!

Which is based on the hard work of many others, currently only credited in the source files. Will have to add credits somewhere more prominent. Anyway, thanks for appreciating what I have done - I hope this sender at some point will become a useful offering for grbl users.

> little JST connector didn't do it's job

Probing is scary - don't know if a probe connected signal could have helped in this case. I have added this to grblHAL (but no driver support yet) and the sender will error out if reported as disconnected.

closed 2020-09-18 by jschoch