terjeio/ioSender #55: Probing failed, or did it?

phil-barrett, 2020-08-28:
Using probing with a simple home made probe (alligator clip, PCB material), I do a center probe and get "Probing Failed message in the bottom status bar.  However, the Z axis was correctly set to the thickness of my PCB (1.42 mm) + 3 mm backoff.  Cutting after probing gave me perfect depth. 

Due to a rather weak design, my little CBeam machine has some bed flex to it that you can see in a video of the probing.  Is that the cause of the Probing failed message? Should I slow down the search rate feed?

Overall, I like the Probing capability so far.

phil-barrett, 2020-08-29:
I spoke too soon. I'm getting Probing Failed and incorrect results.

terjeio, 2020-08-29:
Any error messages in the Console log? Also, can you run a probing sequence with Verbose mode to get some more details?

phil-barrett, 2020-08-29:
ok will do that tomorrow.

phil-barrett, 2020-08-29:
Here's the results of verbose while probing. I included $i output as well. It appears it probes in machine co-ords (makes sense).

<Idle|MPos:210.746,87.703,-68.553|Bf:35,1023|FS:0,0>
Probing...
G91F50
ok
PM:ok
G38.3F50Z-10
<Run|MPos:210.746,87.703,-68.683|Bf:34,1023|FS:50,0|WCO:210.744,87.701,-71.732|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Run|MPos:210.746,87.703,-68.854|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-69.020|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-69.186|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-69.367|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-69.533|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-69.698|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-69.869|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-70.035|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-70.206|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-70.382|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-70.548|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-70.714|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-70.884|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-71.050|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-71.226|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-71.397|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-71.563|Bf:34,1023|FS:50,0>
<Run|MPos:210.746,87.703,-71.734|Bf:34,1023|FS:50,0>
[PRB:210.746,87.703,-71.774:1]
ok
PM:ok
G0Z0.5
ok
PM:ok
G38.3F25Z-10
<Run|MPos:210.746,87.703,-71.387|Bf:34,1023|FS:25,0>
<Run|MPos:210.746,87.703,-71.457|Bf:34,1023|FS:25,0|Ov:100,100,100>
<Run|MPos:210.746,87.703,-71.548|Bf:34,1023|FS:25,0>
<Run|MPos:210.746,87.703,-71.633|Bf:34,1023|FS:25,0>
<Run|MPos:210.746,87.703,-71.714|Bf:34,1023|FS:25,0>
<Run|MPos:210.746,87.703,-71.799|Bf:34,1023|FS:0,0|Pn:P>
[PRB:210.746,87.703,-71.779:1]
ok
PM:ok
G53G0Z-68.779
G10L2P1Z-73.199
Probing failed
ok
<Run|MPos:210.746,87.703,-71.764|Bf:34,1023|FS:450,0|Pn:P>
ok
// $i sent
[VER:1.1f(IMXRT1062).20200818:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:ES,ETH]
[DRIVER VERSION:200823]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[IP:192.168.86.210]
[NETCON:Telnet]

einencool, 2020-08-30:
Maybe you should test this with beta 3 or beta 4.

terjeio, 2020-08-30:
I found a bug or perhaps two. First, if the `G53G0Z-68.779` command fails then `G10L2P1Z-73.199` should not be executed. Secondly there is a 100ms wait for an ok reply from `G53` before it times out - there is no 'ok' in your log so perhaps this is too short. It is the reason for the failure. I will increase this delay a bit to see if that helps. A bit strange that the T4 would need more than 100ms to respond to the `G53` command though...

A correct sequence should look like this:

```
G53G1F300Z-10.391
ok
PM:ok
<Run|MPos:91.805,68.350,-13.427|Bf:34,1023|FS:270,0|Pn:P>
[GC:G1 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T2 F300 S0.]
<Run|MPos:91.805,68.350,-13.426|Bf:34,1023|FS:270,0|Pn:P|Ov:100,100,100>
[GC:G1 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T2 F300 S0.]
<Run|MPos:91.805,68.350,-12.704|Bf:34,1023|FS:300,0>
<Run|MPos:91.805,68.350,-12.661|Bf:34,1023|FS:300,0>
<Run|MPos:91.805,68.350,-11.682|Bf:34,1023|FS:300,0>
<Run|MPos:91.805,68.350,-11.661|Bf:34,1023|FS:300,0>
<Run|MPos:91.805,68.350,-10.661|Bf:34,1023|FS:51,0>
<Run|MPos:91.805,68.350,-10.660|Bf:34,1023|FS:51,0>
<Idle|MPos:91.805,68.350,-10.391|Bf:35,1023|FS:0,0>
<Idle|MPos:91.805,68.350,-10.391|Bf:35,1023|FS:0,0>
G10L2P1Z-23.891
Probing completed
ok
```

terjeio, 2020-08-30:
Beta 5 is now up, I have fixed the bug and increased the delay to 200ms.

phil-barrett, 2020-08-30:
Got Beta 5, thanks for the quick update. 

I am still seeing Probing failed.  See the verbose console output below.

I did a G38.2 Z-10 F50 and it stopped when it touched the probe so at least the probe HW is responding.  Is this maybe a noise issue? The probe interface goes through optoisolators on the BoB but it is still possible there is noise. The probe line does run near the power supply though it is shielded.  I can reconfigure it away from all noise sources for a test you think it is worth trying. 

The output below was from a run before I homed the machine.  But then I homed it and repeated probing with the same results.

> <Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0>
Probing...
G91F100
ok
PM:ok
G38.3F100Z-10
<Run|MPos:0.000,0.000,-0.196|Bf:34,1023|FS:100,0|WCO:210.744,87.701,-73.529|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Run|MPos:0.000,0.000,-0.553|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-0.889|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-1.226|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-1.558|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-1.894|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-2.226|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-2.583|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-2.920|Bf:34,1023|FS:100,0>
[PRB:0.000,0.000,-3.000:1]
ok
PM:ok
G0Z0.5
ok
PM:ok
G38.3F25Z-10
<Run|MPos:0.000,0.000,-2.668|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-2.683|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-2.764|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-2.854|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-2.940|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-3.020|Bf:34,1023|FS:0,0|Pn:P>
[PRB:0.000,0.000,-2.995:1]
ok
PM:ok
G53G0Z0.005
Probing failed
ok
<Run|MPos:0.000,0.000,-2.985|Bf:34,1023|FS:450,0|Pn:P>
<Run|MPos:0.000,0.000,-1.372|Bf:34,1023|FS:439,0>
<Run|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0|WCO:210.744,87.701,-73.529|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>
G90
ok
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0|WCO:210.744,87.701,-73.529|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Idle|MPos:0.000,0.000,0.005|Bf:35,1023|FS:0,0>

terjeio, 2020-08-30:
> Is this maybe a noise issue?

I do not think so - `G53G0Z0.005` is acted upon but the `ok` response is arriving too late for some reason. Network latency or network stack delay?

> Let me know if there is anything I can do to help track it down.

Can you try with UART or USB communication?

phil-barrett, 2020-08-30:
Yes, with USB, it works correct.  I get probing completed message and the probe tip does appear to be position at the correct height.  

I notice a fair amount of lag with ethernet. The machine is not directly connected to my PC but through a local switch.  I've not noticed any other networking latency issues with the PC and switch. Ping results from PC to router BoB.  Is there a way to test telnet latency?

> C:\Users\phill\Documents\gcodesender>ping 192.168.86.210

Pinging 192.168.86.210 with 32 bytes of data:
Reply from 192.168.86.210: bytes=32 time=2ms TTL=255
Reply from 192.168.86.210: bytes=32 time<1ms TTL=255
Reply from 192.168.86.210: bytes=32 time<1ms TTL=255
Reply from 192.168.86.210: bytes=32 time=1ms TTL=255

Ping statistics for 192.168.86.210:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 0ms, Maximum = 2ms, Average = 0ms

terjeio, 2020-08-30:
You may also try to increase the network polling frequency, line 102+ in enet.c:

```
    if (ms - last_ms1 > 100)
    {
        last_ms1 = ms;
        enet_poll();
    }

```
Try with 50 or 25 instead of 100.

phil-barrett, 2020-08-30:
crossed in email.  will give that a try.  100mS is pretty long.   I'll try to get to it this AM but we are leaving for an overnight trip so it might not happen today. depends on when SWMBO gets up.

terjeio, 2020-08-30:
Yes 100ms is a long time - taken from the networking examples. I'll check socket options on the sender side as well, could be that sending is delayed by the stack there as well.

Have a nice trip!

phil-barrett, 2020-08-30:
Built the binary but having update problems with the teensy (seen this before). It seems to not want to update with the new .hex file. Not sure what is going on - will keep trying until I can't.  lol.

phil-barrett, 2020-08-30:
OK, resolved update issue (it was actually updating).  10 ms polling frequency (tried 50 and 25, too) and probing still fails. verified that when connected via USB it does not fail.

terjeio, 2020-08-30:
> Edit, maybe a clue - I have 2 grblHAL systems running, both on telnet. My network manager app on android can see the upstairs but not the one in the shop.

They have different hostnames? You can ping both with IP?

Attached is a modified Core.dll with NoDelay set to true for telnet comms, can you try with that?

[CNC.Core.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/5147422/CNC.Core.zip)

phil-barrett, 2020-08-30:
Yes, I can ping them both. Will be back tomorrow afternoon so will try it
then

phil-barrett, 2020-08-31:
Unzipped and copied both files (CNC.core.dll and CNC.core.pdb) into the G Code Sender folder.  No change - still get Probing failed error message.

terjeio, 2020-09-01:
ok, I will have to somehow measure the time taken for a round trip the G53 commad then.

phil-barrett, 2020-09-01:
If there was some way to see both the verbose console output and the Probe Panel at the same time, I could video it and then count frames to get the time it takes to see the OK.

terjeio, 2020-09-01:
I have to measure the time taken in code, I don't want the UI thread to be a part of the measurment. And I want to find out where most of the delay is, in the sender or in the controller.

terjeio, 2020-09-03:
Attached is another dll to try. It will log time taken (to the console) to ensure there is no pending input from the controller and the roundrip time for the following G53.

For the first the prefix should always be F and time taken should be 200 ms, for the second the prefix is always G.
I have increased the waiting time to 1200 ms for the G53. Examples:

F00:00:00.4041040
G00:00:00.4684911

F00:00:00.2028257
G00:00:00.2065307

F00:00:00.3997611
G00:00:00.4047060

The G53 roundtrip time is the difference between F and G. What kind of timings do you get?

Note that I am using the .Net StopWatch to measure time - it should be accurate, but since Windows is multitasking it is hard to know exactly what is measured...

[CNC.Controls.Probing.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/5169468/CNC.Controls.Probing.zip)

phil-barrett, 2020-09-03:
Here are the results.  looks like it took almost 1700 mS

<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0>
Probing...
G91F100
ok
PM:ok
G38.3F100Z-10
<Run|MPos:0.000,0.000,-0.020|Bf:34,1023|FS:100,0|WCO:0.000,0.202,-4.745|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Run|MPos:0.000,0.000,-0.196|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-0.392|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-0.573|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-0.754|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-0.935|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-1.116|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-1.296|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-1.472|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-1.663|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-1.849|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-2.030|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-2.206|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-2.387|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-2.568|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-2.759|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-2.940|Bf:34,1023|FS:100,0>
<Run|MPos:0.000,0.000,-3.121|Bf:34,1023|FS:0,0|Pn:P>
[PRB:0.000,0.000,-3.050:1]
ok
PM:ok
G0Z0.5
ok
PM:ok
G38.3F25Z-10
<Run|MPos:0.000,0.000,-3.206|Bf:34,1023|FS:268,0|Pn:P>
<Run|MPos:0.000,0.000,-2.854|Bf:35,1023|FS:0,0>
<Run|MPos:0.000,0.000,-2.734|Bf:34,1023|FS:25,0|Ov:100,100,100>
<Run|MPos:0.000,0.000,-2.779|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-2.824|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-2.869|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-2.915|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-2.960|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-3.005|Bf:34,1023|FS:25,0>
<Run|MPos:0.000,0.000,-3.050|Bf:34,1023|FS:25,0|Pn:P>
<Run|MPos:0.000,0.000,-3.095|Bf:34,1023|FS:0,0|Pn:P>
[PRB:0.000,0.000,-3.050:1]
ok
PM:ok
F00:00:00.4222876
G53G0Z-0.05
G00:00:01.6832847
Probing failed
<Idle|MPos:0.000,0.000,-3.126|Bf:35,1023|FS:0,0|Pn:P>
<Idle|MPos:0.000,0.000,-3.126|Bf:35,1023|FS:0,0|Pn:P|WCO:0.000,0.202,-4.745>
<Idle|MPos:0.000,0.000,-3.126|Bf:35,1023|FS:0,0|Pn:P>
<Idle|MPos:0.000,0.000,-3.126|Bf:35,1023|FS:0,0|Pn:P>
ok
<Run|MPos:0.000,0.000,-3.101|Bf:34,1023|FS:360,0|Pn:P>
<Run|MPos:0.000,0.000,-2.623|Bf:34,1023|FS:690,0>
<Run|MPos:0.000,0.000,-1.568|Bf:34,1023|FS:468,0>
<Run|MPos:0.000,0.000,-0.528|Bf:34,1023|FS:168,0>
<Run|MPos:0.000,0.000,-0.075|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-0.050|Bf:35,1023|FS:0,0>
G90
ok

phil-barrett, 2020-09-03:
I misread that, it was a little over 1200mS.

terjeio, 2020-09-03:
Crazy results - something going on in your network? I have response times down to around 5 ms here.

Do you have a second network port available in your computer? If so we can set up a separate network for the controller, same as I did for my Mach3 SmoothStepper installation.

phil-barrett, 2020-09-03:
No, the PC next to my router is a cheap Dell - one NIC (it was my
Mother-in-law's machine).  The PC and Router do share a switch.  Ping times
are sub 1 mS. The PC is shared on the network here. That might be worth
investigating.

I connected the PC directly to the router, reconfigured networks on both PC
and Teensy (teensy on 192.168.1.2 and PC on 192.168.1.1). Glad the PC has
an autocrossover NIC. Established connection, ran TSender and did a probe.
It also failed.

here's the full verbose console output (no edits). Again the difference is
slightly more than 1200 mS.  Looking at firewall and other apps running
next.

MSG:TELNET STREAM ACTIVE]
[VER:1.1f(IMXRT1062).20200818:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:ES,ETH]
[DRIVER VERSION:200823]
[DRIVER OPTIONS:USB.1]
[BOARD:T41U5XBB]
[IP:192.168.1.2]
[NETCON:Telnet]
[GC:G0 G54 G92 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
[G54:0.000,0.202,-4.465]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,-0.280]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
[GC:G0 G54 G92 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
[PRB:0.000,0.000,-10.000:0]
[GC:G38.3 G54 G92 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F100 S0.]
[PRB:0.000,0.000,-12.769:1]
[PRB:0.000,0.000,-12.774:1]
F00:00:00.2012770
G00:00:01.4154811
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.774|Bf:35,1023|FS:0,0>
[... 238 lines cut ...]
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
G91
ok
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,-9.809|Bf:35,1023|FS:0,0>

phil-barrett, 2020-09-03:
Hmmmm, I wonder if we are using exactly the same network stack?  Could you
zip up the network library you are using and I can try building with that.

phil-barrett, 2020-09-03:
By the way, in case you did not figure it out, I pasted the console log
twice.  sorry.

terjeio, 2020-09-03:
Network stack attached. I am going to create a simulator - starting with probing. Hard to simulate with a switch outside of a machine. I have some PSoC 5 code I can use as a starting point.

[teensy41_ethernet-master.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/5170136/teensy41_ethernet-master.zip)

terjeio, 2020-09-03:
Attached is a new dll with the wait increased to 5 seconds. Wonder if that works and if so what the delay is.

[CNC.Controls.Probing.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/5170477/CNC.Controls.Probing.zip)

phil-barrett, 2020-09-03:
Hmmm,  using the new DLL, it still fails. More than 5 Seconds!  verbose
logs attached.  Interestingly, it probes, retracts, reprobes and retracts.
Then several seconds later it reports "Probing failed".

I haven't gotten around to building with your network lib yet.  Will do in
a bit.

[NEWOPT:ES,ETH]
[DRIVER VERSION:200823]
[DRIVER OPTIONS:USB.1]
[BOARD:T41U5XBB]
[IP:192.168.86.125]
[NETCON:Telnet]
[GC:G0 G54 G92 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F25 S0.]
[G54:0.000,0.202,-4.465]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,-0.280]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,-2.975:1]
[GC:G0 G54 G92 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F25 S0.]
[PRB:0.000,0.000,-2.905:1]
[PRB:0.000,0.000,-2.899:1]
F00:00:00.2156234
g00:00:05.2205943
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.101|Bf:35,1023|FS:0,0>
Probing...
G91F100
ok
PM:ok
G38.3F100Z-10
<Run|MPos:0.000,0.000,-0.010|Bf:34,1023|FS:100,0|WCO:0.000,0.202,-4.745|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0|Enc:1>
<Run|MPos:0.000,0.000,-0.191|Bf:34,1023|FS:100,0>
[... 114 lines cut ...]
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0|WCO:0.000,0.202,-4.745|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
<Idle|MPos:0.000,0.000,0.070|Bf:35,1023|FS:0,0>
[PRB:0.000,0.000,-2.970:1]
[PRB:0.000,0.000,-2.980:1]
F00:00:00.2990631
g00:00:05.3006699

terjeio, 2020-09-03:
Got the simulator working now, need to level shift signals as the PSoC is a 5V device. Will do that tomorrow.

It has four 32-bit counters with directly wired step (clock) and direction inputs - will expand it later so I can test auto squaring as well...

terjeio, 2020-09-03:
Forgot that your board has level shifters - and a quick test revealed the same issue here. Why this is happening I do not yet understand, is the g53 sometimes stuck in a queue? It is past midnight here so will sleep on it.

phil-barrett, 2020-09-04:
Tested against a build with your library - as I am sure you expected - same
problem.

Perhaps another clue, sometimes probing just stops.  It will do the
initial probe down but when it contacts the plate (ie probe signal is seen)
it just stops. Of course, I am not able to reproduce it at will but it has
happened a number of times.

terjeio, 2020-09-04:
> Perhaps another clue, sometimes probing just stops.  It will do the
initial probe down but when it contacts the plate (ie probe signal is seen)
it just stops. Of course, I am not able to reproduce it at will but it has
happened a number of times.

This could be the same issue. Does it happen with USB too?

Logging in the controller shows me that data enters and exits the network stack there in order and with no delay. The G53 test I did earlier was outside event handlers in the sender, the G53 that fails is inside. Since I execute the G53 in a task that waits for the ok reply I now believe responses received when executed from the event handler are not handled until the task times out. However it is a bit strange that USB is working, networking not.  Anyway, I will check out if this is the reason and try to work around it.

terjeio, 2020-09-04:
Can you try with this? I have modified the edge finder to run the completion code (including the G53) outside of the event handler. Executing the G53 from the handler somehow blocked input processing from the networking stack, .Net black magic?

[CNC.Controls.Probing.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/5172987/CNC.Controls.Probing.zip)

phil-barrett, 2020-09-04:
Success!!!

That worked perfectly. Probing completed message was a very welcome sight.
Using a piece of paper, I verified that the height set when probing
completed was correct.

I guess context is everything.

Thanks for the hard work on that.

terjeio, 2020-10-07:
Fixed.

closed 2020-10-07 by terjeio