terjeio/ioSender #57: A few feature suggestions.

phil-barrett, 2020-09-14:
I am very happy with the latest version.  The best simply gets better!

Anyway, here are a few suggestions based on my experiences:
1) Add current executing block to the status line at the bottom next to number of blocks.  During a run, I often watch the 3D View tab and have to switch to Program View to find out what block it is on.  Alternatively, show percentage of blocks completed. (True percentage completed would be great but not worth the amount of effort - percentage of blocks is close enough.)
2) In 3D View, have a way to show where you have already cut (or where you haven't cut yet, either way is fine).
3) In Probing/Height Map, have an option to show a height number grid rather than a "heat map". I would like to know how off the bed is - if it is small, I would ignore it. For example, my current bed is off by .03 mm.  I can safely ignore this for a lot of operations though not for shallow engraving or machining PCBs.
4) In Probing/Height Map, have a "manual" option.  I'm not exactly sure the best way to do this.  Maybe a pause before each probe operation to allow positioning the touch plate and pressing a key to continue.  Or perhaps just a pause before each probe operation with a user settable delay.
5) In Probing, I think Probing Z is the most common operation but it is kind of hidden how you do it (in edge finder, click on the center of the diagram). It should be clearer how you probe for Z. If I were doing it, I would make a separate tab just for Z but anything to make it more obvious is good.

Also, in Probing, it appears you can not have multiple profiles. Add and Delete in Profile ... always seem to be grayed out.  I can change the name but never add a new one. Bug? I can see having different probes and would like to have a profile for each one.

By the way, probing is pretty solid now. I'm using the Height Map for uneven stock that can not be surfaced (prefinished material, for example). Even with manual height probing, it works nicely.  Makes me want to buy an XYZ probe.

terjeio, 2020-09-14:
1. Put on todo list.

2. On todo list, not sure how complex or time consuming this will be. Will have to move vectors in and out of paths. Even more complicated if you want it to follow the "tool" cursor, then vectors has to be split on each real time report containing moves larger than a certain distance.

3. Add min/max deviation to the probe completed message will do? I have added more info to this message in the _Edge finder_ tab already (in a "bleeding edge" version downloadable from [here](http://www.io-engineering.com/downloads/)) so this approach could work here too?

4. Pause and reactivate the _Start_ button before each probe?

5. IMO the "Tool length offset" can now be used for this - check the "bleeding edge" version.

> Also, in Probing, it appears you can not have multiple profiles.

Known bug, adding profiles is not working due to a regression. Will be fixed, perhaps tomorrow.

> By the way, probing is pretty solid now.

There are still some Gremlins lurking in the code somewhere, see [this issue](https://github.com/terjeio/grblHAL/issues/79) about the new tool change functionality in grblHAL.
I am trying to get rid of them, will look at that again tomorrow - it has been a busy day today so no time for coding.

phil-barrett, 2020-09-14:
> 2. On todo list, not sure how complex or time consuming this will be. Will have to move vectors in and out of paths. Even more complicated if you want it to follow the "tool" cursor, then vectors has to be split on each real time report containing moves larger than a certain distance.

Maybe it's not worth the effort?

> 3. Add min/max deviation to the probe completed message will do? I have added more info to this message in the _Edge finder_ tab already (in a "bleeding edge" version downloadable from [here](http://www.io-engineering.com/downloads/)) so this approach could work here too?

That probably is good enough. The heat map doesn't tell magnitude.

> 4. Pause and reactivate the _Start_ button before each probe?

Didn't think to  try that.

terjeio, 2020-09-15:
For 2:
> Maybe it's not worth the effort?

I want to give it a try but I want the sender out of beta first.

For 3:
>That probably is good enough. The heat map doesn't tell magnitude.

Check out the latest [edge version](http://www.io-engineering.com/downloads/) just uploaded.

For 4:
> Didn't think to try that.

It is not possible in the current version, in the edge version just uploaded I have added a check box for adding pauses and a new button _Probe_, I decided against using the _Start_ button.

The "bug" blocking adding new profiles is fixed in the edge version.

phil-barrett, 2020-09-15:
A new physical button or one in the GCSender? (or both...)

terjeio, 2020-09-15:
In the sender, can use \<Alt\>+\<P\> as a shortcut. If you mean controller connected physical button then I have to "wire" the cycle start signal from a real time report to sender button. It can be done...

[image]

phil-barrett, 2020-09-15:
Tested the edge version.  Pause between mapping probes works really well.  I like it a lot. Definitely a keeper! No need for physical buttons, imo.  

I don't know if this is a minor visual bug in the edge finder - if you first check the probe Z checkbox, the center of the edge finder diagram doesn't show the "LED".  But if you click on the center of the edge finder probing diagram it, of course, shows up and the probe Z checkbox gets checked.  Subsequent clicking on the checkbox hides/shows the center "LED".  

By the way, the edge finder probing diagram is nicely done - very clear how it works.

terjeio, 2020-09-15:
> I don't know if this is a minor visual bug in the edge finder - if you first check the probe Z checkbox, the center of the edge finder diagram doesn't show the "LED". But if you click on the center [...]

This is per design as the Z probe can be added to all edges. Where the LED lits is dependent on the edge chosen when the checkbox is checked. Perhaps a better UI design is possible - it is not my greatest strength...

> By the way, the edge finder probing diagram is nicely done - very clear how it works.

It is misleading in how it represents the movements, @jschoch has some ideas for improvement.

I will add probing of internal edges/corners as well, I think I will add a separate tab for that to keep the code cleaner. So rename _Edge finder_ to _External edge finder_ and add _Internal edge finder_? A separate tab will also leave room for a better representation of the movements to be performed.

The final one will be for rotation - but that requires gcode transformation...

terjeio, 2020-09-17:
Latest [edge version](http://www.io-engineering.com/downloads/) uploaded now displays the current block while a program is running and has the internal edge finder implemented.

[image]

terjeio, 2020-10-07:
Some features has already been added, remaining ones on the [Roadmap](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Roadmap).

closed 2020-10-07 by terjeio