terjeio/ioSender #59: Height Mapping - non uniform grid request.

phil-barrett, 2020-09-20:
I am making yet another wedding a sign.  The workpiece is not flat and I wanted to use height mapping. The piece is approximately 45mm x 276mm (long and narrow) and I set the w54 origin in the middle of the board.  I set the Area to Probe from program limits.  All I really wanted to do was probe the 4 corners but since I have to have gridsize, I used 45 mm, hoping to have a 2 x 5 grid map (and not 3x10 one). Apparently, the minimum number of grid points in either dimension is 3 so it probed off the workpiece in the Y direction.  No problem, I can fake the extra point hand holding the prob. But it also probed off the workpiece in the X direction. I think it should not probe outside the probe limits setting.  See the screen shot. It has all the specifics. Anyway, I wound up probing 3x14 with 22 mm grid size which got the job done.

So, I would like to see a way to specify x and y gridsize separately. And, make the minimum number of grid points in either x or y to be 2.  That way, I can just probe the corners or perhaps probe a 2x3 or 2x4 grid.

[image]

terjeio, 2020-09-20:
> Apparently, the minimum number of grid points in either dimension is 3 so it probed off the workpiece in the Y direction.

Hmm, this has to be checked in some way.

> So, I would like to see a way to specify x and y gridsize separately.

I use a third party implementation for the heightmap handling, it did not have a constructor for separate x and y sizes so I added one, and it seems to work. For you to confirm?

[image]

>  And, make the minimum number of grid points in either x or y to be 2.

Already is, but since number of grid points are calculated from width and height then hard to get right?

terjeio, 2020-09-20:
>> Apparently, the minimum number of grid points in either dimension is 3 so it probed off the workpiece in the Y direction.

> Hmm, this has to be checked in some way.

Apparently I am a bit dim today... If the dimensions divided by the grid size leaves a reminder then the probed area has to be extended outside the program limits or else the area will not be covered by the map. A solution to this is to "trim" down the x and y grid sizes so the area is covered exactly. A new checkbox is required for this - doable.

With hardcoded "trimming" and a 15 x 15mm initial grid size entered I then get this grid size:

[image]

phil-barrett, 2020-09-20:
Oh, that's very nice!  Both ideas. Will definitely test as soon as edge has it.

I have been setting the initial Area to Probe from program limits but then adjusting it to something that makes the uniform grid size work. In your example, probably 81x54 and a 27mm grid.

Does Height Mapping clip at the AtP boundaries? 

I will write up a how to probe a height map tutorial when the changes settle down.  Pretty sure this feature will get a lot of interest.  It works with plain-jane Grbl?

terjeio, 2020-09-20:
> Does Height Mapping clip at the AtP boundaries?

Not sure what you mean by this. If what happens to moves outside of a mapped area then the 3D viewer will show it.

> It works with plain-jane Grbl?

Can't see any reason for why it should not.

phil-barrett, 2020-09-20:
What I meant by that is it not probing outside the Area to Probe boundaries. There may not be any workpiece material there or may be fixtures to hit. This can be solved with documentation and a non-uniform grid size makes it easier.

terjeio, 2020-09-20:
It was a bug after all, my bad. So no need for a new check-box for trimming as it was already performed in the third-party code. A bit strange that a single value for grid size was used for the constructor, and then this was used to create x and y sizes internally that was silently used (but not by my code). In the latest edge version uploaded I now display the actual "trimmed" size values used for probing when _Start_ is pressed.

> What I meant by that is it not probing outside the Area to Probe boundaries. 

This is now taken care of by default.

phil-barrett, 2020-09-20:
I have been thinking about this a lot today.

Why do we need grid size? The obvious answer is to know where to probe.  But in general, there is an area to probe and you do not really care about the exact grid size. 

So, instead of setting a grid size, why not specify X and Y points and use that to determine grid size? The probing code will do the math.  This is particularly nice when you set the Area to Probe from program limits. Then the user doesn't have to do any math and there is no question of probing outside of Area to Probe..

It would be nice if the preview showed the grid points.

phil-barrett, 2020-09-20:
Sorry, missed your response while I was composing.  That sounds good, will put it to the test asap.

phil-barrett, 2020-09-20:
Quick test.  Works well.  Effective, you get what I was suggesting with grid points though it isn't as obvious.  You need to specify a grid size that is above what your target size would be.  (example: 276mm X with > 92.0 or greater gridsize will get you 3 probe points)

Anyway thanks, this makes the next 6 signs easier!

closed 2020-10-07 by terjeio