terjeio/ioSender #60: edge finding start point image

jschoch, 2020-09-20:
Per the previous conversation in the pull request I am used to the edge finding start point being in a different location and somehow the drawing wasn't clicking for me on starting the probe center on the corner edge.  In order to help users I'd like to suggest an improved drawing that may help new users.  The old drawing had the arrows and the dot not agreeing.  This could be displayed in the empty space once an edge has been selected or in the wiki.  If this goes in the wiki the old drawing should have the arrows lined up with the corner to better show that is the start point.

Here is a draft more accurately showing the moves.

[image]

This draft shows "0" offset which is my preferred method.

[image]

terjeio, 2020-09-20:
The first drawing above is not correct, and the sender graphics is misleading too. A corner probe will first be a move from the corner _Offset_ along the edge, then "XY Clearance_ away from it, then _Depth_ down and then up to "Probing distance" towards the edge. After a succesful probe sequence a move back to "XY Clearance" away from the edge then back up to the initial Z position and then diagonally across the corner to the opposite edge _Offset_ along that edge. Etc...

jschoch, 2020-09-20:
I can change the order.  Hard to describe huh?

Are you open to this format, using colors to show the variables and move vectors?  I don't think 3d helps unless you can orbit around to see it from multiple perspectives.  For 2d i'm not really sure how to show the "depth"

[image]

terjeio, 2020-09-20:
Hard to describe and hard to make drawings that does not make a mess of it. Are you talking about showing a more detailed image on the right side of the current edge/corner selectors or replacing them altogether. Or a _Help_ button that opens a dialog with image and description?

I would think that once a user gets used to the program he would rather like an uncluttered UI, at least I do.