terjeio/ioSender #61: XY clearance cannot exceed probe depth

jschoch, 2020-09-20:
If you prefer to avoid using the edge finding offset value you may want the "XY Clearance" value to be larger than your probing distance since your probe macro starting point is inside the corner.  In any case with the current rule your probe must be inside the corner to detect the work so having a rule for "XY Clearance" <= "probe depth" doesn't really make sense.

terjeio, 2020-09-20:
I am sorry I do not understand your argument.

> since your probe macro starting point is inside the corner

The probe should be above the green point, that is above the corner itself regardless if it is an internal or external corner. The offset is the distance away from the corner along the edge to where the probe will touch the edge. The _XY Clearance_ is the distance away from the edge. When the controlled point is at _XY Clearance_ away from the edge it will first be moved _Depth_ down and then the actual probing motion will be performed.

>  so having a rule for "XY Clearance" <= "probe depth" doesn't really make sense

Where is this rule to be found in the code? If it exists it is definitely incorrect.
_XY Clearance_ must be less than the _Probing distance_ not the _Depth_.

The _Center finder_ had an invalid rule and message that I have fixed but not yet comitted.
The rule used the _Offset_ as part of the rule and the error message was "_Probing XY clearance too large for workpiece diameter._" This has now been corrected to check the _XY Clearance_ against the X/Y sizes \(formely _Diameter_\) and the message has been changed to "_Probing XY clearance too large for workpiece X size._" and "_Probing XY clearance too large for workpiece Y size._".
This only applies to inside center probing so perhaps the message should end with "_...hole size._" or even the whole message should be completely changed to somthing more concise.

A new [Edge build](http://io-engineering.com/downloads) has just been uploaded, please check it out.

Moving forward it would be nice if you could upload the release folder somewhere. IMO a lot easier to relate to than reviewing pull requests, and a lot less work. It will also be easer for other users to review as there si no ned for a VS installation and meddling with code merges and what not.

jschoch, 2020-09-20:
> I am sorry I do not understand your argument.
>
>
> The probe should be above the green point, that is above the corner itself regardless if it is an internal or external corner. The offset is the [...]

this was a typo, i mean to say "XY Clearance" <= "probe distance"

for an external corner if you don't start your probe slightly inside the corner it will never contact because the probe distance will never be enough to trigger the probe assuming "XY Clearance" cannot be greater than the "probe distance"

If you can remove this constraint I can avoid using the offset value and still ensure my probe will not over travel.  

> A new [Edge build](http://io-engineering.com/downloads) has just been uploaded, please check it out.
>
> Moving forward it would be nice if you could upload the release folder somewhere. IMO a lot [...]

happy to do that and i'll check out the new build.