terjeio/ioSender #62: use standard lathe coordinate system

jschoch, 2020-09-20:
Lathe's do not use Z for part diameter they use X.  When offsets for lathe tools can be set in both X and Z the coordinate system should be changed to match the standard lathe setup.

terjeio, 2020-09-20:
>  When offsets for lathe tools can be set in both X and Z the coordinate system should be changed to match the standard lathe setup.

To `G18`, set by a `$N` init command? Or hardcoded and set as default on startup when lathe mode is enabled in the controller?

There are a couple of scenarioes to handle:

1. Legacy grbl. Only one axis can have a tool length offset applied. Default is Z, changing this requires a recompile. Impossible to query which axis is currently compiled in. Z axis tool length offset is of limited use?

2. grblHAL without tool table. Has configuration for lathe mode. This enables G7/G8 support and spindle sync commands if the selected driver supports that. Supports tool offsets for all axes. Will have to touch off tool on every tool change or keep a tool table in the sender in order to avoid that.

3. grblHAL with tool table. Same as 2 above but allows storing tool offsets for a predefined number of tools in the controller. With indexable tools this allows tool change without touch off as I understand it. I want to try this out on my mini lathe.

IMO all scenarioes should be supported going forward.

jschoch, 2020-09-20:
I don't think folks are running legacy grbl for lathes but I may be wrong.  It would be interesting to see if more than a handful are doing it.  I also don't think bart has any plans in the near future to address spindle sync.