terjeio/ioSender #64: product target todo list

jschoch, 2020-09-20:
If you can share what you want to get finished for the first production release I can let you know where I might be able to help.

terjeio, 2020-09-20:
I want probing in a state that is usable and not too confusing.

The rest of the sender is IMO in a production ready state perhaps except for the camera view. The camera view should be refined a bit but not important.

And of course testing, testing, testing...

The testing part is why I want to invite final stage beta testers to help out. Likely these will be legacy grbl users or perhaps also users of Barts ESP32 port if the sender works with that.

terjeio, 2020-09-25:
I am wondering if some kind of "cheap" probe protection should be added or if the sender should wait for a state change to `Idle` before sending the next motion in a sequence. This as I am looking at improving error handling in the probing program class.
Adding a substate to the RUN state in the realtime message could be one way to achieve this, eg. when a probing motion is executing it could be sent as `<RUN:1|...` instead of `<Run|...`. If a `|Pn:P` message is received when not probing then the probing sequence should be then be aborted. "cheap" as it is no replacement for real protection, it could take a relatively long time for the sender to get and act upon the message - but a better than nothing.

The sender is already written to handle substates for all states, not only `Hold` as in the original specification. I have added a setting flag to the `$10` setting in grblHAL in order to get the alarm code as a substate to the `Alarm` state, it could be done the same way for the `Run` state - this to not break existing senders.

What do you think?

---

I added the  `Alarm` substate so that senders can figure out the reason for an alarm at startup, this as some are harmless and can be cleared, some are not and will prevent startup. Unlike legacy grbl where the controller is always reset on connect 32-bit controllers does not always allow that, the connection could be over a network or native USB where a reset would kill the connection...

terjeio, 2020-09-25:
It works. I had to use substate 2 instead of 1 as I had already assigned that to _hold pending_ used for feed hold issued during spindle synced motion.

Good run:
```
G38.3F100Y10
<Run|WPos:-18.955,-0.531,6.650|Bf:33,1023|FS:690,0|WCO:5.770,1.868,4.855|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0>
[GC:G38.3 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F100 S0.]
<Run|WPos:-18.955,-2.831,6.650|Bf:33,1023|FS:638,0>
[GC:G38.3 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F100 S0.]
<Run|WPos:-18.955,-4.778,6.650|Bf:34,1023|FS:108,0>
<Run|WPos:-18.955,-5.000,5.965|Bf:34,1023|FS:500,0>
<Run|WPos:-18.955,-5.000,4.299|Bf:34,1023|FS:500,0>
<Run|WPos:-18.955,-5.000,2.634|Bf:34,1023|FS:337,0>
<Run|WPos:-18.955,-5.000,1.650|Bf:35,1023|FS:0,0>
<Run:2|WPos:-18.955,-4.713,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-4.381,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-4.046,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-3.713,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-3.381,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-3.026,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-2.693,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-2.358,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-2.026,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-1.693,1.650|Bf:34,1023|FS:100,0>
<Run:2|WPos:-18.955,-1.338,1.650|Bf:34,1023|FS:100,0>
<Run|WPos:-18.955,-1.006,1.650|Bf:34,1023|FS:0,0>
[PRB:-13.185,0.748,6.505:1]
ok
```

Probe crash:

```
G53G1F800Z8.841
ok
PM:ok
G38.3F100Y10
<Run|WPos:-18.955,-2.645,8.986|Bf:33,1023|FS:630,0|WCO:5.770,8.125,4.855|WCS:G54|Ov:100,100,100|A:|Sc:|H:0|T:0|TLR:0>
[GC:G38.3 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F100 S0.]
<Run|WPos:-18.955,-4.978,8.986|Bf:33,1023|FS:668,0>
[GC:G38.3 G54 G17 G21 G91 G94 G49 G98 G50 M5 M9 T0 F100 S0.]
<Run|WPos:-18.955,-6.995,8.986|Bf:34,1023|FS:78,0>
<Run|WPos:-18.955,-7.258,8.468|Bf:34,1023|FS:500,0>
<Run|WPos:-18.955,-7.258,6.812|Bf:34,1023|FS:500,0>
<Run|WPos:-18.955,-7.258,5.039|Bf:34,1023|FS:367,0|Pn:P>
PM:fail
Probing cancelled/failed
ok

```

jschoch, 2020-09-25:
this is way better than not detecting it at all!  I've asked Bart to consider a solution for this and opened a ticket but I'm not sure when he will get to it as they are picking apart the state machine and have other priorities.

terjeio, 2020-09-25:
Hard to add - not:

```
        case STATE_CYCLE:
            hal.stream.write_all("Run");
            if(sys_probing_state == Probing_Active)
                probing = true;
            else if (probing)
                probing = probe_state.triggered;
            if(sys.flags.feed_hold_pending)
                hal.stream.write_all(":1");
            else if(probing)
                hal.stream.write_all(":2");
            break;
```
I have not added setting flag and check for that yet, a couple of lines more...