terjeio/ioSender #65: enhancement: probe configurator

jschoch, 2020-09-20:
Beyond the first production release i'd like to discuss how to drive the probing UI with the attributes of the selected probe.  This likely requires a probe configuration wizard of some kind.  This would be a continuation of the discussion in the pull request.

terjeio, 2020-09-20:
Do you think a wizard is required? It is not enough to have some input fields to declare the probe properties? A drop down to select the type and then some fields related to that? E.g. XY offsets for a 3D probe and if there is some kind of probe protection available.

jschoch, 2020-09-20:
it doesn't have to be a full blown wizard, it should make it obvious which fields are specific to the probe setup and type.  If that is simply enabling fields for editing i guess that works ok but one of my goals with the long thread on that pull request was to hide some of the complexity.  Seems simpler to me to put that behind a probe configuration screen.

terjeio, 2020-09-20:
It is simple to hide/collapse fields in WPF so relatively easy to make an input dialog that dynamically changes, for ecample on a selection in a drop box.
If you want to the code behind the dynamic field generation used in the _Settings: Grbl_ tab can be used as well.
And there are some derived controls such as the NumericTextBox and NumericField that provides masked numeric input, the latter with a label and unit attached. I use the NumericTextBox and NumericField extensively across the application to get a consistent look and feel and I suggest you do the same for numeric input.