terjeio/ioSender #66: console enhancements

jschoch, 2020-09-20:
I understand you are already working on a 2nd window for console output.  Here are some things I'd like to help out with.

1. if I run $$ in the MDI window it would be nice if I didn't have to turn verbose off to see it.
2.  it would be really nice to set a regex filter if you are hunting for a value.  Maybe this is the best way to deal with #1.  GROK is a great way to manage layering regex's
3.  My centroid acorn software uses multiple windows and I hate it.  Would  you be open to me prototyping having the console show up in one of the side tabs where "goto" and "Macros" live configurable via the Settings: App?

terjeio, 2020-09-20:
> I understand you are already working on a 2nd window for console output.

I have just added the current console control to a floating window. I have added a filter checkbox for getting rid of realtime messages.

> if I run $$ in the MDI window it would be nice if I didn't have to turn verbose off to see it.

The filter checkbox mentioned above will help? Somehow only showing realtime reports on changes may help to but hard to get right without of overhead? 

> it would be really nice to set a regex filter if you are hunting for a value. Maybe this is the best way to deal with #1. GROK is a great way to manage layering regex's

GROK is a third-party library?

> My centroid acorn software uses multiple windows and I hate it. Would you be open to me prototyping having the console show up in one of the side tabs where "goto" and "Macros" live configurable via the Settings: App?

Give it a try, but I guess that would cover a lot of the senders real estate?

jschoch, 2020-09-20:
> > if I run $$ in the MDI window it would be nice if I didn't have to turn verbose off to see it.
> 
> The filter checkbox mentioned above will help? Somehow only showing realtime reports on changes may help to but hard to get right without of overhead?
>

yes this will help
 
> > it would be really nice to set a regex filter if you are hunting for a value. Maybe this is the best way to deal with #1. GROK is a great way to manage layering regex's
> 
> GROK is a third-party library?

here is a C# implementation:  https://github.com/Marusyk/grok.net

> 
> > My centroid acorn software uses multiple windows and I hate it. Would you be open to me prototyping having the console show up in one of the side tabs where "goto" and "Macros" live configurable via the Settings: App?
> 
> Give it a try, but I guess that would cover a lot of the senders real estate?

I'll give it a try, what resolution is your laptop so I know what minimum display sizes you need?

terjeio, 2020-09-20:
> I'll give it a try, what resolution is your laptop so I know what minimum display sizes you need?

If we can keep the default size as is I would be happy. An enhancement could be to store the latest size somewhere \(in the registry?\) and restore to that. It is possible to hide/show parts of the UI depending on the size, I do that already with the _Program limits_ box when more than 3 axes are displayed in the DRO.

calabr, 2021-02-01:
For troubleshooting resone, it will be good to see commands, sent to the device.

terjeio, 2021-02-01:
> For troubleshooting resone, it will be good to see commands, sent to the device.

I may add that as an option (default off or perhaps on if a GRbl controller is detected) as enabling this severly limits the max speed attainable for laser engraving when executing lots of short movements at a high feedrate. See [this discussion](https://github.com/terjeio/grblHAL/issues/142).

MDI commands will be added to the console in the next build.