terjeio/ioSender #68: One last try before release - change the name.

phil-barrett, 2020-10-01:
I have written about Grbl-GCode-Sender in a number of places (forums, my own documentation, here...) and find the name not very memorable and, honestly, in some ways unfriendly.  While it does describe exactly what it does (good), it is long and makes it hard to write flowing descriptions (bad). I wind up abbreviating it to GCS. Which is fine but would be better not to have to.  Other senders with shorter names, like bCNC, Candle and Grbl Panel, don't get abbreviated and keep their identity when people talk/write about them.  Universal GCode Sender always gets abbreviated.  Candle is a very well named GCode sender - easy to remember and implies light/warmth (it would be even nicer if it lived up to the name!)

So, my suggestions:
- name it after the creator: Terje's Sender, or TSender. For such great work, he deserves recognition.
- abbreviate the official name to GCS and use that always. Have Grbl-GCode-Sender as the tag line.
- use a Norwegian connection. Like Norse or Viking. Again, homage to the creator. Viking is good because it implies robustness, strength and power.

Personally, I think TSender is a fine name.

S2000Stefan, 2020-10-02:
> and find the name not very memorable

I have to agree with phil-barrett, if I want to write something about the GCode Sender I always have to check what the exact name is.

> name it after the creator: Terje's Sender, or TSender. For such great work, he deserves recognition.

My suggestion would be Tio Control or shorter T-Control.
Also think that a relationship with the creator should be established. 
I would also like to have a connection to the original programme tiled in, whereby the GRBL GCode Sender has become much more comprehensive and easier to use.
TIO Panel Control, T-Panel Control?
All just suggestions and food for thought. :)

> use a Norwegian connection. Like Norse or Viking. Again, homage to the creator. Viking is good because it implies robustness, strength and power.

I would also like it very much, unfortunately I have no idea in this direction yet.

> Added simple probe protect to parts of probing sequence.
Requires grblHAL based controller with build 20200927 or later and Run substatus enabled in Status report options.

BTW, where can I find the grblHal version from 20200927 for testing?

terjeio, 2020-10-02:
My comment that I mailed yesterday has been lost in cyberspace...

I was thinking of including the icon in the name:

[image]

Another option: I am living near the sea at an inlet or sound, the current can be quite strong here so my homeplace  is named "Straumen" -> "io Streamer"?

Viking or Norse - nope, IMO better to choose a name from viking mythology if going in that direction. "Balder"? I don't know...

Not easy this.

> BTW, where can I find the grblHal version from 20200927 for testing?

Currently only on my computer - waiting for Phil to commit some driver code for his card and for me to finish testing of the latest refactoring I have done. 

BTW I have put up initial roadmaps for both the sender and grblHAL in the Wikis.

phil-barrett, 2020-10-02:
Norse mythology is good.  Personally, I like Ragnar though not a god. Loki is not good because of mischief.  Odin or Thor are good though perhaps over used. Baldr? I dunno. I'm a huge Viking fan. I mean how can you not admire a culture that sailed across the North Atlantic in open long boats with only the crudest of navigation aids to discover North America 500 years before Columbus. 

But Io Sender is ok by me.  I like the icon treatment.  The only downside is some may think it just refers to Input/Output.

And, yes, I owe you the file for testing.

S2000Stefan, 2020-10-03:
> I was thinking of including the icon in the name:
[image]

I think this is a very good idea. :+1: 
IO Sender sounds very good.
Towards vikings is really not that easy but with phil-barrett we seem to have an expert in this direction. :)

terjeio, 2020-10-03:
If going for Io Sender should it be written as "ioSender"?

S2000Stefan, 2020-10-03:
> If going for Io Sender should it be written as "ioSender"?

Well, I like it very much.
But that is just my opinion.

terjeio, 2021-04-29:
Done.

closed 2021-04-29 by terjeio