terjeio/ioSender #73: Are there any plans to add Chinese language to the software in the future?

lianzaozi, 2020-12-05:
Are there any plans to add Chinese language to the software in the future?

terjeio, 2020-12-05:
There are no plans for any specific languages, the first step will be to make the software translatable. Then someone fluent in a given language, and knowing the technical terminology, should provide a translation.

Making the software translatable is on the [roadmap](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Roadmap): _UI localization_. First step is to get the software come out of beta to a "production" version, my aim is to get this done before the end of the year.

lianzaozi, 2020-12-06:
Does the reason for this problem have anything to do with the processing speed in GCode?

lianzaozi, 2020-12-06:
I can help in the work of adding Chinese language to the software.

lianzaozi, 2020-12-06:
I have not found the hardware for testing grblHAL firmware. Can you recommend the relevant hardware?

terjeio, 2020-12-06:
> I can help in the work of adding Chinese language to the software.

Great, leave this issue open so I can get back to you when ready.

> I have not found the hardware for testing grblHAL firmware. Can you recommend the relevant hardware?

The top one is IMO the [Teensy 4.1 controller](https://github.com/phil-barrett/grbl-teensy-4). [Re-Arm with RAMPS](https://github.com/terjeio/grblHAL/tree/master/drivers/LPC1769) or [Bigtreetech SKR v1.4](https://github.com/terjeio/grblHAL/tree/test/drivers/LPC1769) (currently only in the test branch) are other options. However it is hard to reccommend without knowing what kind of machine you want to control, if you want a breakout board or not. 

The [driver overview table](https://github.com/terjeio/grblHAL/tree/master/drivers) has a row for _Board_ that may guide you in the right direction. Note I do _not_ recommend STM32F103 based redpills/bluepills as these has little room for adding plugins and may be left stuck at using a specific version of grblHAL in the (near?) future.

lianzaozi, 2020-12-07:
I am very happy that I can contribute to this software.
I don't know how to buy these control boards? Can I buy it on eBay?

lianzaozi, 2020-12-07:
This is a motion control board I found in China, I don't know whether this control board can be used for testing grblHAL, I have uploaded the relevant information of this motion control board, please help me to confirm whether this control board can be used for testing grblHAL firmware.
[Motion control board information（Motion control board information）.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/5652491/Motion.control.board.information.Motion.control.board.information.zip)

terjeio, 2020-12-08:
> I don't know how to buy these control boards? Can I buy it on eBay?

The Teensy 4.1 breakout board is only available from [Tindie](https://www.tindie.com/products/philba/grblhal-breakout-board-unkit-for-teensy-41/), others from eBay, alibaba...

I have uploaded the relevant information of this motion control board, please help me to confirm whether this control board can be used for testing grblHAL firmware.

It is a STM32F407 based board so the [STM32F4xx driver](https://github.com/terjeio/grblHAL/tree/master/drivers/STM32F4xx) should work but may require some changes to code (setting up the clock tree). A new mapping file has to be made for it though. I see that it has Ethernet capability - this is planned for this driver.

Is this board available for purchase and if so from where?

lianzaozi, 2020-12-08:
https://item.taobao.com/item.htm?id=553323670176&ali_refid=a3_420434_1006:1122648409:N:kDOEsTbtOPkSR2n6hub%2FVQ%3D%3D:26cd7ffff82a98d0bda8e95ca703f332&ali_trackid=1_26cd7ffff82a98d0bda8e95ca703f332&spm=a230r.1.1957635.7

This is the purchase URL, I am not sure if your network can open this website correctly.

www.ing10bbs.com
This is a technical exchange forum

lianzaozi, 2020-12-08:
After communicating with the merchant just now, I learned that customers in the U.S. purchase support mailing to the U.S. Some countries do not support direct mailing and require transit.

terjeio, 2020-12-08:
Does he want grblHAL added as an optional firmware? This could open up for worldwide sales. I have no idea about the potential volume though.

It seems to me to be a very well made board, is the customer feedback good?

lianzaozi, 2020-12-09:
The vendor who sells the control board would like you to adapt the grblHAL firmware for his control board. I found another same control board. This control board has several different CPU options (STM32F407IGT6 industrial control board F429 F767 development board Cortex-M4 core PLC industrial control board H743)

terjeio, 2020-12-09:
> The vendor who sells the control board would like you to adapt the grblHAL firmware for his control board.

Is he willing to donate a board to me for test/validation or do I have to buy one?

> I found another same control board. 

Adding driver support for boards with processors in the same family should not be too much work as usually an existing driver can be updated with a driver map and a bit of tweaking - at least for core functionality. However, writing a new driver could be a lot more work - F767 and H743 would likely require that. And I need hardware for testing if I do not already have a develepment board with the actual processor on it. If I have one then final testing _could_ be done by someone that owns the target board depending on which peripherals support is required for.