terjeio/ioSender #74: Can you add and cancel the grid and change the color of the grid?

lianzaozi, 2020-12-05:
Can you add and cancel the grid and change the color of the grid?

As shown in the figure below: the track display color is the same as the grid display color. The grid blocks the track, which makes it hard for me to see.

[image]

lianzaozi, 2020-12-05:
I know that the candle software allows me to change the color of the display, and allows me to decide what to display/not to display.

Candle software screenshot
[image]

terjeio, 2020-12-05:
The 3D-viewer will get more [attention later](https://github.com/terjeio/Grbl-GCode-Sender/wiki/Roadmap) when the first production release is out, this can be a part of that. Thanks for the feedback.

terjeio, 2021-05-15:
Took some time to get to the first production release... I have now played a bit with the viewer, added color selection and made the first attempt at highlighting path to be executed (a bit ahead of the "tool" due to the planner buffer in the controller).

[image]

The changes can be tested by downloading the latest edge version (33p5) [from here](http://www.io-engineering.com/downloads).

closed 2021-07-26 by terjeio