terjeio/ioSender #75: Operating hours evaluation (odometers)

S2000Stefan, 2020-12-12:
Not an issue but hopefully a small question.

Is it possible to set up a tap on which you can see the evaluation of the odometer in the IoSender? Some information is recorded by the odometer that could be displayed very nicely, such as travel distances and times. 
For example, to be able to grease the linear guides after so many hours of operation or to carry out other maintenance work. This could be indicated, for example, by a pop-up window or a green, yellow or red LED. Of course, you would then have to be able to enter the time or distance when the service should be carried out. I think that the pure operating hours would be sufficient.
That would be very close to an industrial machine. :-)

It would also be advantageous if the user of a rotational axis (a-axis) could be informed of the existence of a rotational axis by setting a check mark, since the recording of a rotational axis is carried out per 1 degree and one could therefore divide the result by 360 in order to display the rotations, since otherwise the number would increase immeasurably.

terjeio, 2020-12-12:
> Is it possible to set up a tap on which you can see the evaluation of the odometer in the IoSender? Some information is recorded by the odometer that could be displayed very nicely, such as travel distances and times.

It is, I have already added a option value so the sender can easily find out if the data is available. Coding a tab and associated functionality is a bit more work though.

> one could therefore divide the result by 360 in order to display the rotations

Not sure how to handle this, you mean an additional display for calculated number of revolutions? Or changing the current DRO value?

S2000Stefan, 2020-12-12:
> Not sure how to handle this, you mean an additional display for calculated number of revolutions? Or changing the current DRO value?

The DRO display is OK. If you use the A-axis as the axis of rotation, for example, G0 A360 is displayed in degree, in this case 360,000. However, the three digits after the decimal point are a little too precise for an axis of rotation, I think one digit would suffice. However, if someone wants to use the a-axis as a linear axis, he would certainly like to have the three digits after the comma displayed. ;-)
How could you tell the program that there is a rotation axis to automatically reduce the decimal places?

I have just looked at the whole thing again. The odometer records 1.0 at 1000 degrees. If you were to divide this 1000 degrees by 360, you would have 2.7 revolutions, which would increase the number, which is exactly the opposite of what I wanted to achieve.
I probably had a wrong assumption in my head that the odometer would also record 1000.0 at 1000 degrees. That's how you can make a mistake. I think the recording is fine as it is. My suggestion would make everything worse.