terjeio/ioSender #76: ioSender in a Windows VM on Linux

phil-barrett, 2020-12-16:
Not a problem report.  I have a customer that reports ioSender works well on Linux inside a Windows VM.  I think this is very good news for Linux users.

benirnd, 2021-03-10:
Hi Phil,

I am that one customer who reported this. To add some more info, I managed to run it both ways:  

- forwarding the serial port to the Windows VM
- using ethernet
 
It works in both modes but I prefer the ethernet setup since the spindle noise does not reach the host computer anymore like it does when using the serial over the USB cable.

benishor, 2021-03-10:
Hi Phil,

I am that one customer who reported this. To add some more info, I managed to run it both ways:  

- forwarding the serial port to the Windows VM
- using ethernet
 
It works in both modes but I prefer the ethernet setup since the spindle noise does not reach the host computer anymore like it does when using the serial over the USB cable.

phil-barrett, 2021-03-10:
Yes, ethernet is far superior because of that.  And, not that hard for the average CNC system builder.