terjeio/ioSender #77: Sender Beta 8 crashes when loading code, and "error:33" while running

ajtalbot1, 2020-12-26:
Running Sender Beta 8.
G-Code generated with CamBam 1.0.7364.41821

G-Code is very simple.  Drill 4 holes.  Post processor is set to GRBL in CamBam.   When I try to load my G-Code it crashes/closes Sender.  When I strip out all the header, and only have the main g-code, it runs, but quickly hits an "error:33".

Code, and pic of the error attached.

[image]
[20mm thick spoiler board, 4 corners, 0.25in end mill, M4 socket heads.zip](https://github.com/terjeio/Grbl-GCode-Sender/files/5743392/20mm.thick.spoiler.board.4.corners.0.25in.end.mill.M4.socket.heads.zip)

terjeio, 2020-12-26:
Cam-Bam issue? I get the same error with vanilla grbl and grblHAL (Mega2560) and both ioSender and GrblPanel.
If not Cam-Bam then there is a bug in the controller. The offending arc moves has to be checked manually find out.

Which controller are you using? vanilla grbl or grblHAL based? Other?

The file attached does not crash the sender for me, does it for you?

phil-barrett, 2020-12-26:
He is using it on the Teensy 4.1 Breakout Board.

terjeio, 2020-12-26:
> By the way, is there a list of ioSender error messages available?

Yes, in the file _error_codes_en_US.csv_ in the sender folder - they are _not_ hardcoded in the sender source code. The error code is "translated" when received and the corresponding text shown in the status line - `error:33` -> `Motion command target is invalid.` For arcs (G2/G3) this means the IJK parameters are not within tolerance.

ajtalbot1, 2020-12-26:
When I open the attach g-code, my sender closes.  Not sure if this is a
true crash, but it is a very unexpected closure.  I have to re-launch
sender.  The only way I can get it to stay open is remove the header from
the g-code.  Sender works fine with the code generated from MeshCAM.

I am running the sender on my Windows 10 laptop. USB connected to the
Teensy, but plan is to use Ethernet when I get everything production worthy.

Yes, I am running GRBLHal on the Teensy 4.1 Breakout Board. I will try
executing the failed arc command, manually tonight, and report back.
-Adam

ajtalbot1, 2020-12-27:
More info.

Sender (beta-8) crashes/closes with a single line of G-Code. T1 M6

As for the Arc.  G2X1.41Y137.45I2.82J0.0 F100  And I get back the error
"Motion command target is invalid".  Screenshot attached.  I get an
error:33 when using a putty console.

If this helps, here are all my settings:
%
; grblHAL
; 1.1f(IMXRT1062).20200603
; [OPT:VNMSL,35,1024,3,0]
; [NEWOPT:ES,TC]
; [DRIVER VERSION:200606]
; [DRIVER OPTIONS:USB.2 IOPORTS]
;
$0=10
$1=25
$2=0
$3=5
$4=7
$5=0
$6=0
$10=510
$11=0.010
$12=0.002
$13=0
$14=0
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$28=0.100
$29=0
$30=24000.
$31=0.
$32=0
$33=5000.000
$34=0.000
$35=0.000
$36=100.000
$37=0
$39=1
$40=0
$41=0
$42=2
$43=1
$44=0
$45=0
$46=0
$56=5.000
$57=100.000
$58=-5.000
$59=500.000
$60=0
$61=1
$62=0
$63=3
$64=0
$65=0
$100=400.000
$101=400.000
$102=400.000
$110=3000.000
$111=3000.000
$112=1000.000
$120=50.000
$121=50.000
$122=50.000
$130=200.000
$131=200.000
$132=200.000
%

phil-barrett, 2020-12-27:
Man you are on a tear!  Thanks for wringing it out.

terjeio, 2020-12-27:
> 1.1f(IMXRT1062).20200603

This is an old version, could be the reason for the sender quitting? Upgrading the controller or downgrading the sender is the easiest option for fixing this? Or if you do not need the tool change action strip M6 from the gcode by enabling that in _Settings: App_.
This as I am not too keen to retrieve this version to debug the sender, but can do so if you insist.

Error 33 is due to Cam-Bam outputting bad arcs - grbl calculates the arc radius correctly and it does not match the one provided (by IJK values). There is a 0,0117 mm divergence.

Code is here if you want to check yourself, I have already verified that the input parameters are correct.

https://github.com/terjeio/grblHAL/blob/2569bb233f63f109d2217c0a56b624182af6d9ad/grbl/gcode.c#L1793

phil-barrett, 2020-12-28:
I posted [a question about arcs](https://www.cnczone.com/forums/cambam/414286-cambam-arcs-grbl.html#post2422670) in the CamBam section of CNCZone.  One response was to 

> Make sure the arc center mode is set the same in GRBL settings and in CamBam postprocessor settings.

Anyway, hope that helps.

ajtalbot1, 2020-12-28:
New firmware loaded.  Many thanks to Phil for cooking it.

T1 M6 loads!!!

[VER:1.1f(IMXRT1062).20200922:]
[OPT:VNMSL,35,1024,4,0]
[NEWOPT:FLASH,ES,ETH,TC]
[DRIVER VERSION:200923]
[DRIVER OPTIONS:USB.2]
[BOARD:T41U5XBB]
[IP:0.0.0.0]

I will test the advised CamBam settings in the morning.

terjeio, 2020-12-28:
> Make sure the arc center mode is set the same in GRBL settings and in CamBam postprocessor settings.

Grbl does not support absolute mode for center format arcs, only incremental mode - and this is what is emitted by the postprocessor. The error is flagged at the end of a helix and for the first arc used to flatten the bottom of the hole. There is a rather large error in the radius of that arc. From Mach3 docs:

>It is an error if:
¨ when the arc is projected on the selected plane, the distance from the current point
to the center differs from the distance from the end point to the center by more
than 0.0002 inch (if inches are being used) or 0.002 millimetre (if millimetres are
being used). ..."

`$12` is used by grbl to set arc tolerance and it has 0.002 mm as the default value.

The second line is the arc that fails:

```
G2X-2.83Y135.0Z-4.5I1.41J2.45
G2X1.41Y137.45I2.82J0.0
```

I am not so sure it is a good idea to increase `$12` to ignore the error.

ajtalbot1, 2020-12-30:
Agreed it's not a good idea to change $12.  But that was the path I took
and it still failed.  Moved it from 0.002 to 0.01.

As best as I can tell, this is a CamBam bug.  Not an issue with GRBL.  I
will raise an issue with their support.

closed 2021-02-25 by terjeio