terjeio/ioSender #78: Sender beta 8  starting issue / chrashes

einencool, 2020-12-30:
Good morning @terjeio 
I have some huge problems with beta 8 and the teensy 4.1 board.
Very often on program start, it doesn't find the board and when I press unlock & reset, the program chrashes

like you can see in the picture, in the upper bar is standing "keine Rückmeldung", means, it doesn't communicate anymore and after that, the program closes.
On next start it seems ok, but yesterday I got many of these problems.

Sometimes I got error:50

[image]

terjeio, 2020-12-30:
Good morning @einencool.

Error 50/unable to communicate could be due to a loose connection/noise on the Halt input. Please check that, perhaps by temporarily shorting it to ground.

einencool, 2020-12-30:
One thing I forgot to say :-)
I also updated grblhal to the latest branch with auto squaring.

When I power up the machine, I have no power on the motors and so the e-stop is always high.

Before these changes I had to press the "reset" button in the software.

Since these changes I also have to reset and then also press the unlock button.

Normally I had copied all my settings from the configs, but this time there seems something different.

einencool, 2020-12-30:
Well, I don't get it, it says, that it i waiting for the controller, but it recognizes the endstop and so on...
I also flashed it once again, and started the pc several times, and it is shown in the manager correctly, I don't know whats the reason, and now I can't connect to the machine any more...

einencool, 2020-12-30:
Oh interessting,
I opened the "beta 7.3" from October, and I could connect to the machine.
After closing, I'm able to connect with your new beta 8 again...

einencool, 2020-12-30:
Sometimes I got the problem while rerunning the same GCode, that the Sender Beta 8 goes „crazy“.
The Spindle was spinning up, and then the Code says wait for 5 seconds.
But then nothing happens and the sender shows GCode, that would be several lines under the G4P5 Code.

I tried it some times, but after changing to Beta 7.3 everything was running fine again.

One question for Beta 7.3:
Does it also support the autosquaring option?
I‘ve seen the $-Settings added, but there where no text for this.

terjeio, 2020-12-30:
I have made some changes to better handle odd startup scenarioes. A new [edge version](http://io-engineering.com/downloads/) is just uploaded with those.

> Sometimes I got the problem while rerunning the same GCode, that the Sender Beta 8 goes „crazy“.
The Spindle was spinning up, and then the Code says wait for 5 seconds.
But then nothing happens and the sender shows GCode, that would be several lines under the G4P5 Code.

Can you upload the file so I can check what the problem is?

> Does it also support the autosquaring option?

No and yes - the settings texts are missing but otherwise it should work.

einencool, 2020-12-30:
[beides.nc.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5755770/beides.nc.txt)

This was the Code to mill the surface area and since there were some edges I'll had to give it 0.2mm every time, 

I'll give the sender tomorrow a little try because I have to clean up the working area, because I messed around today while milling the aluminium surface *g*

I don't know what the problem was, but today I was not able to connect to the board with the beta 8. I also restartet the PC, reconnected everything, but nothing works. Then I started the beta 7.3 and it works directly. After that, I could close the Beta 7.3 and start the beta 8, until I got the problem, that the Program starts making these problems, and then I switched completely to the Beta 7.3 and could finish my work without any problems.

terjeio, 2020-12-31:
beides.nc.txt runs fine for me, tested with Teensy 4.1 and ESP32 drivers and the latest edge version. "Dry runs" only - no motors or spindle attached.

einencool, 2020-12-31:
Like I said, a couple of times the file works, but then something happens...
And since I had many failures with this beta, I stopped working with this version.
With the 7.3 everything works as expected.
I hope I can make some little tests without milling today.

einencool, 2020-12-31:
Well, I'm not able to work with your Beta 8.2,
this is the message I get, when I start the software: (I tested it several times...)
[image]

And this happens, when I start Beta 7.3.
[image]

And that happens after I close Beta 7.3 and then start Beta 8.2 again
[image]

Note:
I don't unplug or change anything else.

Wish you a nice start into the new year :-)

einencool, 2020-12-31:
Oh, I've seen, that I'm not able to home the machine, the button is greyed out and I can't click it.

S2000Stefan, 2021-01-01:
Good morning @einencool,

I also ran your test programme this morning on my test computer on the desktop. 

> Sometimes I got the problem while rerunning the same GCode, that the Sender Beta 8 goes „crazy“.
The Spindle was spinning up, and then the Code says wait for 5 seconds.
But then nothing happens and the sender shows GCode, that would be several lines under the G4P5 Code.

For me, the nc programme also always stops at the point mentioned (G4P5) and ioSender goes into hold status. After restarting the programme, it goes to Alarm2, whereupon I deactivated the softlimits. 
After that, your nc programme runs without any problems, even from the start. 
Then I set the limits of the x-y axis to much larger values and reactivated the soft limits, the nc test programme then also runs through without problems. 
I think something is wrong with your limits.
It is strange, however, that the ioSender goes on hold first. 
Everything with the new grblHAL version and Beta 8.2.

einencool, 2021-01-01:
First of all a happy new year to everyone.

Good morning @S2000Stefan ,
I have set the soft limits a few mm wider, and I also deactivated them, but that doesn't have some influence.
I'll get my laptop in a few moments and make also a copy of the $-settings and attach the changed files for my grblhal config. Maybe there are some points I have changed which are wrong, because most of it I copied from the last versions.

terjeio, 2021-01-01:
Happy New Year all!

I am currently working on fixing this. I have made som changes after the recent discussion about protocol extensions and these are not yet correctly handled.

Alarm 10 is a grblHAL extension and is from E-stop input active. This input is not generally activated, IIRC only MSP432 and iMXRT1062 (Teensy 4) drivers has it as default. The e-stop alarm is handled different from othe alarms in that is not possible to issue a soft reset to clear it unless the input is deasserted (cleared) first. If a soft reset is allowed clear the alarm while the input is asserted this would result in the E-stop becoming inactive. Not good from a safety perspective.

The Reset input "inherited" from Grbl is different in that it is, per default, possible to ignore by issuing a soft reset. This means that it is, IMO, unusable as an E-stop input (unless it is configured to behave differently). The fuctionality for this is kept "as-is" for backwards comaptibility.

For the iMXRT1062 driver the input pin can be [configured](https://github.com/terjeio/grblHAL/blob/9cfceba724c826d205c569580e61f1b34cee2397/drivers/IMXRT1062/grblHAL_Teensy4/src/my_machine.h#L61-L62) to behave as Reset or E-stop, and the default is dependent on the `COMPATIBILITY_LEVEL` setting. When <= 1 the it acts as E-stop, otherwise as Reset.

I am not entirely sure why the @einencool machine is booting in alarm 10 state but this may explain it:

> When I power up the machine, I have no power on the motors and so the e-stop is always high.

From this I believe the controller starts in alarm 10 state and when the motors are powered the E-stop pin is deasserted.
This means that it will be ok to soft reset and unlock the controller - and that is not what the latest sender does, it warns about the alarm and exits. I am now working on fixing this - and should not be to long before I am able to upload a new version.

It can be noted that I have found a small bug in the controller related to this too. When a soft reset is received when in alarm 10 state _and_ the E-stop input is still asserted every other soft reset does not seem do anything. I'll fix this soon as well.

einencool, 2021-01-01:
Good Morning,
so, here are some configs from my machine.
Here is an old Settings-list from an older Version, there I can see, that $10=510 and now it is $10=511

[GRBL Daten 06.2020.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5758795/GRBL.Daten.06.2020.txt)
[GRBLHal-Config-01012021.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5758796/GRBLHal-Config-01012021.txt)

And here are the files for the Config in GRBLHal, which I updated to my machine:
- Autosquaring Y-Axis
- 2 Endstops on each Axis
- Parking enabled (not yet tested)
- Door Switch enabled (also not tested)
- Encoder for G1-Speeds
- EEPRom enabled
- Some Changes for Acc / Steps / Limits / Spindle

SRC-Folder:
[my_machine.h.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5758800/my_machine.h.txt)
[T41U5XBB_map.h.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5758801/T41U5XBB_map.h.txt)

GRBL-Folder:
[config.h.txt](https://github.com/terjeio/Grbl-GCode-Sender/files/5758803/config.h.txt)

Maybe you can find an issue, or maybe I do have to do a Reset of all settings, because the changes between the Versions are to big.
Thank you in advance for your great support :-)

terjeio, 2021-01-01:
New [edge version](http://io-engineering.com/downloads/) just uploaded. The startup handling is now a lot more advanced and should better handle the different scenarioes where the controller may or may not be able to communicate properly.

I have not yet been able to reproduce the G4P5 issue, so currently I have no idea if this is at the sender or controller side. I'll create a controller build based on your configs and see what happens.

The $10 difference likely due to the controller reverting to default settings when you updated.

S2000Stefan, 2021-01-01:
Happy New Year all, too.

> I have not yet been able to reproduce the G4P5 issue, so currently I have no idea if this is at the sender or controller side. I'll create a controller build based on your configs and see what happens.

I would say that this is not really a problem. As I said if I home the machine and start an nc programme with the soft limits ($20) turned on and the coordinates are outside the range of the limits, ioSender first goes into Hold status and then when I press Cycle start again, it goes into Alarm2. 
But this only happens the first time I start the controller.
With einencool, it seems that no hold is triggered, for whatever reason. The issue can (G4P5) be adjusted so nicely.
As he described above.

But this has nothing at all to do with the error when starting with the ioSender. At least that's how I would conclude it. ;-)

I only noticed it directly when I started his nc-test programme and had the same error directly. My only problem was that my travel paths were not sufficient for his "big programme". 
I hope I was able to explain it a little better now.

einencool, 2021-01-01:
I made a little test with the new Beta 8.3 and it seems to work.
While connecting without power on the motors, I got a message to remove this issue. After that it connects to the board without a problem.
Then I unpowered the motors and it raises the alarm and when I powered them up again. I only had to reset the connection like in the old versions.

Then I only make some moves by hand with the machine and turned the power on and off several times and it seems to work.

One question for interest, why did you change the starting routine in this way

Tomorrow I'll mill some little parts, and I hope that then everything is fine again :-)
Thanks for your hard work.

terjeio, 2021-01-01:
> One question for interest, why did you change the starting routine in this way

The main reason is that the sender needs a lot of information from the controller on startup, even more so now that detailed information about available settings can be fetched from it. Since the controller can be in a state where only real time status requests are responded to this means the sender cannot get this information unless the controller is brought to a state that allows full communication.

Legacy grbl controllers were typically hard reset on connect typically clearing any alarms and bringing it to a state where communication is possible. Many grblHAL controllers simply cannot be reset in the same way, either because they do have the neccessary signal lines available internally or because a hard reset will close the connection. A soft reset on connect _could_ be done by the sender, but this will not solve all issues at hand - E-stop or Door state will not be cleared by that. And a soft reset might not be desirable if a MPG/DRO is connected directly to the controller .

If you try to start a pre beta 8 ioSender when E-stop is active or the controller is in Hold or Door state it will not be able to configure itself properly.

For example when E-stop is active pressing the _Reset_ button will not do anything before E-stop is cleared. This can be confusing and IMO it is better to tell the user that E-stop is active and interactively tell them how to fix this or just exit the sender:

[image]

If the user does nothing this will be the response if no action is taken, this will be repeated until E-stop is cleared or the user exits the sender:

[image]

Hopefully this will lead to less support issues in the future. Note that I am still working on this, I am not entirely happy with the current code.

einencool, 2021-01-01:
Ah Ok, that makes sense :-)
Thank you for your detailed information, I didn't think about these issues.

It is easier to understand for me, when I have some information why some things change :-)

And now have a great first day of this year 🥇

S2000Stefan, 2021-01-01:
One thing I noticed in the new beta 8.3, in the Goto tab both G28 and G30 are greyed out although they are set? 
But they work when entered manually via the MDI.

terjeio, 2021-01-01:
@S2000Stefan : it is yet another safety feature. From the LinuxCNC documentation:

>Warning
Only use G28 when your machine is homed to a repeatable position and the desired G28 position has been stored with
G28.1.

The same applies to G30. When the machine is homed then the buttons are enabled. Is this a bad idea?

einencool, 2021-01-01:
One other question for the soft limits.
In beta 7.3 it was the case, that the soft limits for the "max" position is working very well, but when driving in the "min" direction, the machine runs into the end stops. I've a pull out of 1.5mm fromm the endstop, what else do I have to configure, that the machine stops the 1.5mm before the end stops?
Is there a way?

terjeio, 2021-01-01:
> Is there a way?

Soft limits are not handled by the sender so I do not understand this should in any way be related to the sender version.

The regular soft limits check "inherited" from grbl does not use the pulloff distances in the calculations and it does not check that homing has been performed either. I have a TODO comment in the source about at least adding homed check. 

I have added a separate soft limit for jogging that allows jogging within machine space (minus pulloff distance) when homed even if a jogging command exceeds this. This can be enabled with `$40=1`.

Perhaps I should update the regular soft limit routine to match (but of course error out if travel exceed limits instead of limit the motion as I do for jogging)? I do not believe this should cause any problems since I believe the original implementation was rather simple due to memory constraints in the 8-bit Atmel processors...

einencool, 2021-01-01:
I have enabled $40=1 and i always home the machine on startup

i only mean the Softlimits for Jogging Motions, and there the machine runs always in the endstops in the homing direction. If this could be handled to the pulloff distances, everything would be fine for this :-)

and sorry, i don’t know which part is responsible for this ;-)

S2000Stefan, 2021-01-01:
@ terjeio, of course the machine must be homed beforehand.
I had been running various nc programmes all day today on a trial basis. And of course the milling machine must have been homed beforehand, but apparently I pressed reset at some point in the afternoon and lost the homing. 
I must not have remembered, a quick look at the home button would have given me certainty.  ;-)

I always drive my machine to a predefined place (G28) after work, and that's when I noticed the greyed-out buttons.

Would it be correct to lock G28 and G30 via the MDI if the machine has lost homing?

terjeio, 2021-01-01:
> i only mean the Softlimits for Jogging Motions

Ok, then I have to check my jog limit routine again. However, it is working correctly in my test machine. I just increased the pulloff distance to 3mm and this is respected, a bit strange this is not working for you. Maybe you can check with a larger pulloff distance to see if this helps?

> Would it be correct to lock G28 and G30 via the MDI if the machine has lost homing?

Perhaps, but in general I do not want to add checks in the sender. Such checks should be performed in the controller, but for G28 and G30 there is only a warning about not to issue these commands when not homed. I'll think about it.

einencool, 2021-01-01:
I’ll try 3mm tomorrow, but especially the two Endstops on the Y-Axis are highly precise and work within 0,002mm so i don’t think, that any acceleration or something like this should make a difference. 
But when it will work, you can maybe tell me, where there is the point in the config to change it for my machine. 
Because now, I’m standing in home position and can raise the z-Axis directly into the endstop.

and that’s why I always thought, that the Softlimits are disabled, but in the max direction they work. 
and for the max range i had to set the value + pulloff distance for example:
1,5mm pulloff
Machine is homed and axis is set to 0,0mm
I move 497mm till the max endstop hits, 
means with 498mm i have 0,5mm safety for manual jogging.

but I’ll try the min direction tomorrow :-)

terjeio, 2021-01-02:
> I’ll try 3mm tomorrow

Maybe try even more, 10-20mm, to see if soft limits are active at all.

> i don’t think, that any acceleration or something like this should make a difference

It does not make a difference as the target is changed to be within machine space. Or rather should be if there is a bug.

> But when it will work, you can maybe tell me, where there is the point in the config to change it for my machine.

If you mean the pulloff distance then this can be set at run time with `$27` (in the _Homing_ group of settings). In config.h by `DEFAULT_HOMING_PULLOFF`.

> but I’ll try the min direction tomorrow

By min direction you mean the homing direction?

einencool, 2021-01-02:
Very interesting, it seems, that there was a bug in the older 7.3 Version, because now everything is working as expected.

At first I set it to 15mm and it works.
Then with 3mm everything was ok
and even with my normal 1.5mm it works now. 

So I make my tests today, and when in every case, I'll give a little report and then in best case I'll close this issue 👍 
Thank you

einencool, 2021-01-02:
One point I have noticed while I made some tests today.
I have found my own issue for raising Alarm:1 several times. It's a problem with my new Y-Endstops, which are so sensitive, that when I place my part into the vise and give it a "little punch with the Hammer", then the end stops were triggered. But it was only shown as Alarm :1 and I didn't notice it first.

Is there a way, that this message shows, that one of the endstops was triggered?

Until now it seems good and stable.
In the next hour I'll make some tests with probing to see, if it has still the problem on the right side.

einencool, 2021-01-02:
Probing seems also to be ok, but one thing happens, what I tested, because I remember, that you said, when in Probing Tab, it is not possible to start anything else than probing.

But I pushed the hardware "Start-Button" on my machine while in Probing tab, because I thought, that the Probing routine will start, but sadly this doesn't happen, and I'm absolutely happy, that my 3D-Finder is not mounted in the Spindle collet, because now I would have a wireless Sensor....

When I pushed the button, the Spindle turns on, and the machine wanted to start my loaded CNC-Programm. 
ThenI pushed the E-Stop Button, and wanted to Reset the Software, but I'm not able to, because the GRBL Tag is greyed out and I don't have any access to the Reset buttons... (also the marked STOP Button in the software doesn't work)

[image]

Also one point for the Jogging limits, I don't know if that matches, but when I home the machine, everything works ok with the Softlimits. But when it raises one time the E-Stop or maybe one of the Endstops (while hammering ) then the Softlimits won't work and it will drive directly into the Endstop.

terjeio, 2021-01-02:
> Is there a way, that this message shows, that one of the endstops was triggered?

Yes, alarm 1 is hard limits triggered.

> But I pushed the hardware "Start-Button" on my machine while in Probing tab, because I thought, that the Probing routine will start, but sadly this doesn't happen, and I'm absolutely happy, that my 3D-Finder is not mounted in the Spindle collet, because now I would have a wireless Sensor....

This is not how it should work, I have to check/fix this.

> Then I pushed the E-Stop Button, and wanted to Reset the Software, but I'm not able to, because the GRBL Tag is greyed out and I don't have any access to the Reset buttons... (also the marked STOP Button in the software doesn't work)

Same for this, needs a fix.

> Also one point for the Jogging limits, I don't know if that matches, but when I home the machine, everything works ok with the Softlimits. But when it raises one time the E-Stop or maybe one of the [...]

This is correct behaviour as homed status is lost on e-stop or hard limits triggered, this since it is likely that position is lost due to the abrupt halt. Softlimits for jogging (and the standard soft limits as well) can only be applied in a meaningful way for homed axes. However, it could be argued that when e-stop is triggered homed status should not be lost if the machine was not moving when it happened. But then e-stop is normally hit when it is moving?
You could try to enable _Homing on startup required_ for `$22` - _Homing cycle_ and _Strict mode_ for `$21` - _Hard limits enable_ if you want to ensure that the machine cannot be run unless homed.

einencool, 2021-01-02:
Well, the Homing at the beginning is my normal routine, but it would be great, if there could be an option, that the Softlimits will still work, maybe with a $-Config or in the Sender itself.

In my case, the Motors are Closed Loop, and when I have a real crash, it makes absolut sense, to Home the machine. 
But it would be great, if this could be left to the choice of the user (just my thoughts).

The Hardlimit Endstops behind the Softlimit will still work, and so its just a double safety. :-)

But the rest seems to work good.
Thank you :-)

terjeio, 2021-01-02:
I have relaxed clearing of homed status and added a new flag to $22 for never clearing it on a soft reset to the [test branch](https://github.com/terjeio/grblHAL/tree/test). I hope this does not leads to any bad side effects...

I believe have fixed the sender cycle start problem, will update the edge version tomorrow after a bit more testing.

einencool, 2021-01-02:
Great, sadly my holidays are over, but I hope i can try it sometime next week.
Thank you so much for your fantastic work.

terjeio, 2021-01-04:
New [edge version](http://io-engineering.com/downloads/) uploaded, I have made some further changes to the startup sequence and also message handling. I have blocked the cycle start input from starting a loaded program when in the probing tab, it will start the probing action instead - hope this is okay.

einencool, 2021-01-05:
>I have blocked the cycle start input from starting a loaded program when in the probing tab, it will start the probing action instead - hope this is okay.

Yes, that sounds really good.
Maybe on Thursday I‘m able to make some short tests, but I don‘t know it exactly. If not, it will be at the weekend.

Was your change successful for the alarm that was triggered with standing machine, or didn‘t you change something in that direction?
I‘m searching for a little bit stronger springs for the sensors, so that they won‘t be so extremely sensible

terjeio, 2021-01-06:
> Was your change successful for the alarm that was triggered with standing machine, or didn‘t you change something in that direction?

No changes for this as limit switches are handled in the controller. Disabling limit switches when not in motion is likely a bad idea?

einencool, 2021-01-06:
> > Was your change successful for the alarm that was triggered with standing machine, or didn‘t you change something in that direction?
> 
> No changes for this as limit switches are handled in the controller. Disabling limit switches when not in motion is likely a bad idea?

No no, it sounds very good. :-)
I forget that you already did it :-)

I'll try it at the weekend, this week I have to make some modifications to my 3d printer. And the machine is not at my home, so I'll have to wait 2 days more ;-)

einencool, 2021-01-08:
Hello @terjeio 
Now I've build your new Test-branch from yesterday and your new Beta 8.6

When I give a punch with the Hammer, Alarm1 will still raise.
The Problem now is, that I have to RESET AND UNLOCK the machine. 
I think in the old version, it was only a RESET necessary.
[image]

The Homed status seems to be still available, it doen't run into the Endstop after the Alarm1

The Probing start with the hardware button seems to work fine.
It starts the probing routine.

terjeio, 2021-01-08:
> When I give a punch with the Hammer, Alarm1 will still raise.

I have not changed anything related to this. The only option would be to change the controller to disable hard limits when not in motion. If I am ever going to do that then a setting is required as this should not be the default behaviour. Such a change may have unintended sideeffects?

> The Problem now is, that I have to RESET AND UNLOCK the machine.
I think in the old version, it was only a RESET necessary.

I changed that because the behaviour was not consistent with the messages coming from grbl - "reset and unlock to continue". Earlier I unlocked automatically on a reset, messages then flashed by and were hard to catch. If a big problem I could add a setting for how to handle resets.

einencool, 2021-01-08:
Ah ok, I had thought, while the machine is idle, no Alarms will raise from the Endstops, but then that was my fault.

For clearing the Alarms, it would be great, if you can make a setting to only „Reset“ the Alarms.

And one thing, I mentioned a few weeks/months ago, would it be also possible to add a setting, that „Starting“ the GCode or Probing will prompt a message for confirmation like starting the Macros?

As I mentioned, I‘ve only to push one button on my controller pad, and the machine will start, it would only be a safety. Or can I add a macro for starting the GCode and Probing? Is it „ALT + R“?

einencool, 2021-01-09:
Good Morning @terjeio 
Today I wanted to make a test piece.
I've measured the left and the front corner seperately, because It has a clamp in the corner.

Then I normally start a macro, but the macro doesn't move the machine before the offset will be applied.
Normally the machine should slowly move to X0Y0 and then set the offset, but with the new version, this doen't happen.

This is my macro:
G1 x0y0 f300
G10 P1 L20 X0.4 Y68.63

Can you please check it, and maybe add the offset directly to the probing tab :-)

with best regards
Chris

terjeio, 2021-01-10:
> Can you please check it, and maybe add the offset directly to the probing tab :-)

Works for me, adding offsets are still on the todo list - part of the probe profile drop-down planned.

[image]

> ALL Macros with a Move command and also a Command from the console didn't work

> Also after several tests, sometimes it happens, that the software doesn't recognize any move commands.
It answers with "OK" but nothing happens...

What is reported from the real-time report requests when this happens?
Check _Verbose_ in the _Console tab_ to see them.

einencool, 2021-01-10:
> > Can you please check it, and maybe add the offset directly to the probing tab :-)
> 
> Works for me, adding offsets are still on the todo list - part of the probe profile drop-down planned.
> 
> [image]
> 

That was the same picture, I got, but the machine doesn‘t move to X0Y0 before.
My father was standing beside me, and we talked a little bit, and it took me several times, to recognize, that only a restart works.
After that, it works like it should.

> What is reported from the real-time report requests when this happens?
> Check _Verbose_ in the _Console tab_ to see them.

It was only the „OK“ command, the other lines keep the same.
But I‘m not at the machine the next days, but when I‘m there, I‘ll make a new trial.

terjeio, 2021-01-11:
> It was only the „OK“ command, the other lines keep the same.

The other lines kept coming? Can you copy one of these lines and post here when this problem arise?

einencool, 2021-01-11:
Yes, they come in lets say one second Intervalls.

One point I have also seen, maybe it‘s correct, but I don‘t know exactly.
I Set my Workpiece height with my 3D-Finder (Tool No. 1) .

Then I made my Reference Tool length measurement in the probing Tab with the checkbox enabled (also Tool No. 1).
After that, on the first screen, the red light for „TLO ref‘d“ was lid.
Then I tried made a test if my workflow was correct, so I milled in the air, to see if everything works.

Since everything was correct, I changed again to tool No. 1 but as I now had new offsets for the ednmills, I wanted to clear the Reference measurement.
So I clicked on „Clear“ in the Probing tab, but on the first page, the red light for „TLO ref‘d“ was still on.
I was not able to clear the light, so I pressed the „RESET“ Button, that works and the „TLO ref‘d“ light was off.

Then I probed the workpiece with the 3D-Finder, and then the problems began to start.
Maybe this could help you.
I hope on Thursday I‘m at the machine to try to get the Error again.

einencool, 2021-01-14:
Yesterday evening I tried to get the issue again.
I tried to follow the exact work flow I had done, but sadly it was not possible to produce this error again.

So I'll have an eye on this, but for know I'll have to close this issue, because I don't know what happened.

I also took my father with me to the machine and we tried to do the exact same things, but it didn't happen again.

If there will be another problem again,
I'll reopen the issue.
Thank you for your great support

closed 2021-01-14 by einencool