terjeio/ioSender #79: Support of hand wheel?

mattbue, 2020-12-30:
Hi, is there any support of a hand wheel?
If not, are there any plans of integrating support of it?

Thanks

terjeio, 2020-12-30:
> Hi, is there any support of a hand wheel?

No.

> If not, are there any plans of integrating support of it?

No, not in the sender per now. Some are using game controllers with the sender as it supports keyboard jogging and has shortcuts for many commands. I have no idea about what it would take to add handwheel support as I do not know if there is a common protocol or many different ones. So "a handwheel" could fast become which one(s) to support.

There is some support at the controller side though. The iMXRT1062 driver (Teensy 4.1) has pins that can be used for quadrature input from a basic (or raw) handwheel or encoder - however code support for jogging is very basic if working at all. What is working is using this for overrides, typically with a low PPR encoder. 

I have [made a MPG/DRO](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack) for the controller side that uses a secondary serial input for communication that can be used in standalone mode (no PC connected). Many drivers has support for a secondary serial input, and quite a few also for an I2C based keypad.

mattbue, 2020-12-31:
Thanks for you information.
In the code i have seen by clicking the jog buttons they call the `Button_Click `method and inside basically their content is used as command. So the commands used therefore are `X-, X+, Y-, Y+, Z-, Z+`.
I have connected a wheel (100 pulses/rev) to an Arduino which sends exactely these commands on the serial port.
Executing the `JogCommand ` method with the commands coming from the wheel is ending with an error since the command is coming from another thread.
Unfortunatelly I'm not really familiar with thread safe calls.

terjeio, 2020-12-31:
> I have connected a wheel (100 pulses/rev) to an Arduino which sends exactely these commands on the serial port.

So you simulate keypresses directly with the handwheel, not sure that is a good ide as keyboard jogging relies on a key up event to terminate the jog (unless you send short moves only - step mode).

> Unfortunatelly I'm not really familiar with thread safe calls.

I am struggling with this too - I believe you have to use the dispatcher to overcome that. How it is done in SerialStream.cs (in CNC Core) might be of help?

Since you use an Arduino using the [keypad plugin](https://github.com/terjeio/grblHAL/tree/master/plugins) and connecting directly to the controller could be an option for you. If it is a 5V Arduino then maybe not, depending on the controllers I2C interface.

mattbue, 2021-01-07:
I had another idea how to connect a handwheel. The first tries are looking really good.
Before showing i want to build it up a little bit more.

What i already can say:
The solution I'm working on would be completely independet from GCode Sender and grblHAL ;-)

jameskolme, 2021-04-02:
First and most important thing, thank you to @terjeio for making this sender program. I have used GRBL panel for many years, but it is not compatible with windows 10 anymore, this results communication issues, somehow many other programs suffer same error, but not this one (and layout is similar). I have done my part of removing communication issues to that extend, that I have removed USB port completely from my CNC machine, genuine RS-232 is the way. 

PS. would it be possible to change Feed rate and distance buttons to / * and + - buttons, this way simple and cheap numpad can be used as pendant, without much confusion? And possibly add 100mm distance, longer machine would benefit this. 

But why I write here to hand wheel topic, well because hand wheel would be really nice to have. 

If not that familiar with CNC handwheels, these are generally divided to 100 tics/revolution and jogging with handweel is done by moving exact step distance for every tick, typically 1mm/0.1mm/0,01mm for every tick etc. It seems handwheel would be possible to use with this program. 

It would require arduino Leonardo and some lines of code, this simply mimics key pressing from user wheel input.  There is of course problem without feedback, would be no way of knowing what feed or distance is selected, everything else should be doable.  This can be avoided if every distance and feed are placed under shortcut keys, so pendant can overrule every time jogging starts. This way feedback is not necessary and pendant can even have separate jog buttons and maybe even relative XYZ coordinate display. I think this would work.

terjeio, 2021-04-29:
> PS. would it be possible to change Feed rate and distance buttons to / * and + - buttons, this way simple and cheap numpad can be used as pendant, without much confusion?

These buttons (in the numeric keypad) does not have distinct scancodes so cannot be distinguished from the "normal" keys. Due to that I have added numeric keypad 2 and 8 (NumLock mode) for changing the feed rate (in addition to 4 and 6 for changing the distance).

> And possibly add 100mm distance, longer machine would benefit this.

Use keyboard jogging for longer distances, safe if your controller is grblHAL. Use with \<shift\> modifier for faster speed. Distance and speed is settable in the _Settings: App_ tab:

[image]