terjeio/ioSender #80: Restore last window size on startup, handled not correctly

einencool, 2021-01-16:
Good Morning @terjeio 
Today I checked the Box "Restore last window size on startup in the settings.

Theoretically it works, but the problem ist, that the upper left corner will still be the starting point.
For example I start the programm, the upper left corner ist maybe 300 pixel from the left desktop side and 300 pixel under the top area.
When I maximize the window and restart the programm, the upper left corner is still at 300x300 pixel, but the rest from the window is not shown...

[image]

terjeio, 2021-01-16:
Ok, I'll have to save the position as well then.

einencool, 2021-01-23:
Good Morning @terjeio 
I downloaded your newest Version from today, it seems to save the position.
But one question.
Most time I use it in "full Screen" or Maximize Mode (don't know how to say :-) )
Is there the possibility that the software will restart in this mode?
But now it's better than before.

terjeio, 2021-01-30:
Good morning @einencool

The latest [edge version](http://io-engineering.com/downloads) will now handle this correctly. It has also some changes to _Tool length offset_ probing that I hope is ok, please be careful when using this and report any issues!

einencool, 2021-01-30:
Hello @terjeio 
Today I made a little test, and it seems much better now.
Sadly the „maximized“ Position is not stored, but it‘s much better than before.

Since I didn‘t make a GCode with Tool Changes, I haven‘t tested the changed TLO part.

One question, somewhere you mentioned, that it would be able to see which Limit Pins raised an alarm.
Because I modified my „sensitive“ Endstops with stiffer springs, but I thing I got a noise Problem somewhere.
While the Spindle is running up, it happens two times, that Alarm:1 raises and the machine stops before any movements...

terjeio, 2021-01-31:
> Sadly the „maximized“ Position is not stored, but it‘s much better than before.

Do you have two monitors? Otherwhise I do not understand. Position is not saved, only size - and I set size to -1 when the window is maximized to indicate that it is to be maximized on startup.

> One question, somewhere you mentioned, that it would be able to see which Limit Pins raised an alarm.

I cant remember that, only that it was mentioned that when two switches was used per axis it would be nice to see both in the "LED" status? If you have a problem with intermittent triggering of switches then this will not help as you might not even see a "LED" blink if the trigger pulse is shorter that the polling rate (200ms).

What could (or rather should?) be done is to log the reason for an alarm beeing triggered so that this could be reported separately via a new $-command.

einencool, 2021-01-31:
> > Sadly the „maximized“ Position is not stored, but it‘s much better than before.
> 
> Do you have two monitors? Otherwhise I do not understand. Position is not saved, only size - and I set size to -1 when the window is maximized to indicate that it is to be maximized on startup.

No, I only use a Laptop with the built in screen. 
I started the software several times, and it shows up in the last position, that‘s ok so far :-)
But the software is not „maximized“.
Then I clicked on „maximize“ and waited a few seconds.
After that I closed the Program and restarted it, but it has the same position and size as before and I have to click on maximize again.
That‘s all :-)
I can live with that, that‘s not the Problem :-)

> I cant remember that, only that it was mentioned that when two switches was used per axis it would be nice to see both in the "LED" status? If you have a problem with intermittent triggering of [...]

Ah ok, I thought somewhere I read about it, that I can see in the Console which endstop was triggered to see where the alarm raises...
This option would be great, because I think it‘s a noise issue, because it shows up, while the spindle is starting. But the funny thing is, that it happens two times and I started the machine around 8 times....
So I have to check the wires if they are now lying to close to each other...

terjeio, 2021-01-31:
What is saved in for window size in your _App.config_ (in the sender folder) after you exit the sender with the window maximized?

It should be like this:
```
 <WindowWidth>-1</WindowWidth>
 <WindowHeight>-1</WindowHeight>
```

einencool, 2021-01-31:
Ahh Ok, that makes sense, didn't know what you mean before :-)

I don't know why, but the Config was not updated from the sender.
I always copy the "App.config" and the "ProbingProfiles.xml" to the newest Sender Folder, but the last saved date from the "App.Config" was the 16.01.2021 and not from yesterday...

So I enter the "-1" manually and test it in the next days, because I don't have the machine next to me and the Folder are synced to my PC at Home.

Thank you and for this I'll close this issue, because I think then it is solved :-)
Have a nice sunday evening 
Greets
Chris

closed 2021-01-31 by einencool

terjeio, 2021-01-31:
> What could (or rather should?) be done is to log the reason for an alarm beeing triggered so that this could be reported separately via a new $-command.

I am going to add a new command, `$LEV` (for Last EVents) that reports the inputs that was active when the core was last informed about a change via interrupt. This to aid finding which pin(s) caused problems due to interference. The output will contain the same letters for inputs as used for the `|Pn:` element in the real-time report.

I'll have to figure out how to best handle dual input limit switches (min & max) as this will require changes to both the core and _all_ drivers. I am likely to report max limit switches by using the corresponding lowercase letters when this is ready.

einencool, 2021-01-31:
That sounds interesting.

But one point.
Does it matter if it‘s the max or the min Endstop, that is triggered? Do some Boards have inputs for both sides? 
On the Teensy there is min & max together on one input.
The only thing is that the Y1 OR Y2 Input could matter :-)

terjeio, 2021-02-01:
> Does it matter if it‘s the max or the min Endstop, that is triggered? Do some Boards have inputs for both sides?

Not that I am aware of. But who knows what will become available at some point?
So I have decided to add core support for up to four limit switches per axis for limits and one or two for homing if separate from the limit switches. Now I need to retest auto-squaring again to be sure I have not broken anything...